FastLED/FastLED #1348: APA104 Malfunction? wrong connections?

serlancelot, 2022-01-06:
Hello, I have a problem with this led ... which is that I can't get it to work and it does something weird ... I'll tell you:

I connect everything according to the technical sheet, I use the flashing program that is in the fastled library, and in it I modify the program to select the APA104 line and cancel the NeoPixel I compile and load.

What does the led do? If I connect it only to the power supply, it remains green and if I connect the data, it continues without doing anything.

If, on the other hand, I load the color palette example, I configure it for the APA104, compile it and load it.

What does the led do? If I connect it only to the power supply it stays with the green color and if I connect the data it seems that it does nothing and in seconds it begins to turn blue and it desolders the cables.

Say they are in 1.2mm package. I don't know what it could be, any ideas?

Thank you very much in advance. All the best!

serlancelot, 2022-01-06:
The color change is because it heats up, if I blow when it starts to change it goes down to green again and this happens when I connect the data line if I leave it disconnected it remains green ...
The data line is connected to a 330 ohm resistor (according to the technical sheet between 100-1k and as a normal value 300ohm) but this value changes according to the number of LEDs. I have put a 1k one on it and it no longer does the color change by heating, but it still does not work ...
What else could I look at?

yesyesuk, 2022-01-06:
Can you provide a diagram or photo showing how you connected everything up?
The fact that it heats up to the point where wires are desoldering suggests that you wired it wrong.

serlancelot, 2022-01-06:
To take a bad picture .... because the cables are very very fine, but I will tell you how it is connected!

All according to the technical sheet (I attach the link). I connect the power supply of the LED to the GND and 5v of the arduino and the data cable with the resistor I connect it to digital pin 6 or 5 or 3, it depends on the example of the library ... The data output pin is not connect because I am only testing with a led.

http://neon-world.com/uploads/soft/20211118/1637228322.pdf 

Thanks!
If you really need the photo I try to do it ...

serlancelot, 2022-01-26:
Hello, well, everything has been solved... the problem was that the data sheet corresponding to the led was the old version and they didn't change it on the manufacturer's website.... so they work fine now! What a silly head-banging because of an image....
Greetings

closed 2022-01-26 by serlancelot