FastLED/FastLED #1349: ESP32-C3 output stability issue (flickering, artifacts + guru meditation after few seconds)

DmytroKorniienko, 2022-01-09:
Problem description: same code works as expected on ESP32 & ESP32-S2 but on ESP32-C3 produce unexpected pixels flickering and other artifacts. Our project made for PIO and at that moment require newest SDK and some hacks to get working target ESP32-C3 on current VSCode + PIO version : https://github.com/DmytroKorniienko/FireLamp_JeeUI/tree/dev

target:
```
[env:esp32c3] ; 4mb without OTA
framework = arduino

platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
platform_packages =
    ;framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.1
    framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git

; platform = espressif32
; platform_packages =
;     toolchain-riscv-esp@file://d:\toolchains\riscv32-esp-elf ; https://github.com/espressif/crosstool-NG - toolchains releases
;     framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git
;     platformio/tool-esptoolpy @ ~1.30100

board = esp32dev ; board hack
;board = esp32-c3-devkitm-1
board_build.mcu = esp32c3
board_build.partitions = noota.csv
board_build.f_cpu = 160000000L
board_build.f_flash = 80000000L
; board_build.flash_mode = qio
board_build.arduino.ldscript = esp32c3_out.ld
build_type = debug
build_flags =
    ${extra32.build_flags}
    -DLED_BUILTIN=3 ; 3,4,5
    -DLAMP_DEBUG=1
    -DEMBUI_DEBUG
    ;-DFASTLED_ALLOW_INTERRUPTS=1
    ;-DFASTLED_INTERRUPT_RETRY_COUNT=3
    ;-DINTERRUPT_THRESHOLD=0
    ; Logging level: 1 = error, 2 = warning, 3 = info, 4 = debug, 5 = verbose
    ;-DCORE_DEBUG_LEVEL=4
    ;-DCONFIG_ARDUHAL_LOG_COLORS=1
lib_deps =
    ;${env32.lib_deps}
    https://github.com/plerup/espsoftwareserial.git
    https://github.com/DmytroKorniienko/DFRobotDFPlayerMini
    https://github.com/DmytroKorniienko/EmbUI.git#dev
    https://github.com/FastLED/FastLED.git
monitor_filters = esp32_exception_decoder, default
extra_scripts = replace_fs.py
monitor_speed = 460800
upload_speed = 921600
; monitor_port = COM11
; upload_port = COM11
; monitor_port = COM17
; upload_port = COM17
monitor_rts = 0
monitor_dtr = 0
; monitor_rts = 1
; monitor_dtr = 1
```

but I think same issue may be reproducible on any other environment with ESP32-C3 and exchange over WiFi via web-sockets / UDP

Let me know if you need any additional information/testing at any ESP32/ESP32-xx or ESP8266 controllers.

Related discussion: https://www.reddit.com/r/FastLED/comments/qfbz79/esp32_s3_lets_see_if_i_can_make_it_work_with/hm75n6o/

sethtroisi, 2022-02-17:
I am having similar flickering, one note is it only appears for me when I have enabled (and used?) WiFi on the esp32.
If I comment out my WiFi code then it doesn't flicker.

jpswensen, 2022-03-06:
I just thought I would add on my recent experiences. I tested with both the lastest esp-idf 4.4 and the latest from Github for FASTLED. I am using the esp32-c3-devkitm-1 board and trying to control the built-in RGB led. 

The strange behavior is that it seems as if only the very first call to FastLED.show() is actually taking effect. I am using the basic blink example (have tried in both ArduinoIDE and VSCode+PIO and have the same result). Whatever I set the initial color to works great, but every subsequent call to FastLED.show() doesn't change the color.

When I get home, I am going to try with and external strip, but don't have one with me now.

planevina, 2022-05-28:
I have the same problem.
same code, work good with esp32-s2
flickering, artifacts ...and many problem with esp32-c3

s-light, 2022-06-21:
i also encounter this problem:
if i enable wifi i get kind of random flickering / colors.
if wifi is disabled it is working fine.

i have created two test-sketches:
minimal 
```c++
#include <FastLED.h>

#define NUM_LEDS 2
#define DATA_PIN 8
CRGB board_pixels[NUM_LEDS];

#include <WiFi.h>
IPAddress apIP(192, 168, 178, 1);
IPAddress netMsk(255, 255, 255, 0);

void setup() {
    Serial.begin(115200);
    Serial.println("FastLED_Wifi.ino");
    Serial.printf("ESP32:\n");
    Serial.printf("  model = %s Rev %d\n", ESP.getChipModel(), ESP.getChipRevision());
    Serial.printf("  cores: %d\n", ESP.getChipCores());
    Serial.printf("  mac: %d\n", ESP.getEfuseMac());
    Serial.println();

    // GRB ordering is assumed
    FastLED.addLeds<NEOPIXEL, DATA_PIN>(board_pixels, NUM_LEDS);

    blink_rgb();
    delay(1000);

    WiFi.mode(WIFI_OFF);
    // WiFi.mode(WIFI_AP);
}

void loop() {
    blink_rgb();
}

void blink_rgb() {
    Serial.println("Blink RGB_");
    board_pixels[0] = CRGB( 50, 0, 0);
    board_pixels[1] = CRGB(100, 0, 0);
    FastLED.show();
    delay(500);
    board_pixels[0] = CRGB(0,  50, 0);
    board_pixels[1] = CRGB(0, 100, 0);
    FastLED.show();
    delay(500);
    board_pixels[0] = CRGB(0, 0,  50);
    board_pixels[1] = CRGB(0, 0, 100);
    FastLED.show();
    delay(500);
    board_pixels[0] = CRGB(1, 1, 1);
    board_pixels[1] = CRGB(1, 1, 1);
    FastLED.show();
    delay(500);
}

```

with button to toggle wifi on / off
```c++
#include <FastLED.h>

#define NUM_LEDS 2
#define DATA_PIN 8
CRGB board_pixels[NUM_LEDS];

#include <WiFi.h>
IPAddress apIP(192, 168, 178, 1);
IPAddress netMsk(255, 255, 255, 0);

// boot button
uint8_t buttonPin = 9;

bool buttonPushedLast = false;

void setup() {
    Serial.begin(115200);
    Serial.println("FastLED_Wifi.ino");
    Serial.printf("ESP32:\n");
    Serial.printf("  model = %s Rev %d\n", ESP.getChipModel(), ESP.getChipRevision());
    Serial.printf("  cores: %d\n", ESP.getChipCores());
    Serial.printf("  mac: %d\n", ESP.getEfuseMac());
    Serial.println();

    // GRB ordering is assumed
    FastLED.addLeds<NEOPIXEL, DATA_PIN>(board_pixels, NUM_LEDS);
    FastLED.delay(1);

    WiFi.mode(WIFI_OFF);

    pinMode(buttonPin, INPUT_PULLUP);
}

void loop() {
    blink_update();
    bool buttonPushed = !digitalRead(buttonPin);
    if (buttonPushed != buttonPushedLast) {
        // simple debounce
        delay(30);
        // check for button released
        if (!buttonPushed && buttonPushedLast) {
            toggleWifi();
        }
        buttonPushedLast = buttonPushed;
    }
}

void toggleWifi() {
    if (WiFi.getMode() == WIFI_OFF) {
        board_pixels[0] = CRGB(0, 150, 150);
        board_pixels[1] = CRGB(0, 150, 150);
        FastLED.delay(1);
        FastLED.show();
        Serial.println("WiFi mode: WiFi_AP");
        WiFi.mode(WIFI_AP);
        WiFi.softAP("FastLED_Wifi.ino");
        delay(3000);
    } else {
        Serial.println("WiFi mode: WiFi_OFF");
        WiFi.mode(WIFI_OFF);
        board_pixels[0] = CRGB(100, 100, 0);
        board_pixels[1] = CRGB(100, 100, 0);
        FastLED.delay(1);
        FastLED.show();
        delay(3000);
    }
}

const uint8_t colors_count = 4;
CRGB colors[colors_count][NUM_LEDS] = {
    {
        CRGB( 30, 0, 0),
        CRGB(100, 0, 0)
    },
    {
        CRGB(0,  30, 0),
        CRGB(0, 100, 0)
    },
    {
        CRGB(0, 0,  30),
        CRGB(0, 0, 100)
    },
    {
        CRGB(1, 1, 1),
        CRGB(1, 1, 1)
    },
};
uint8_t currentColor = 0;

void blink_rgb_next() {
    board_pixels[0] = colors[currentColor][0];
    board_pixels[1] = colors[currentColor][1];
    FastLED.delay(1);
    FastLED.show();
    currentColor = currentColor + 1;
    if (currentColor >= colors_count) {
        currentColor = 0;
    }
}

uint32_t timestampStart = 0;
uint32_t duration = 500;

void blink_update() {
    if (millis() - timestampStart > duration) {
        timestampStart = millis();
        Serial.println("color next..");
        blink_rgb_next();
    }

}
```

StormYudi, 2022-07-24:
Same problem here.

lpatiny, 2022-08-27:
I also encountered the same problem with a Lolin ESP32-c3 and tried to find another way to let it work.

I had success using the following libraries:
```
	adafruit/Adafruit NeoPixel@^1.10.5
	me-no-dev/AsyncTCP @ ^1.1.1
	ottowinter/AsyncMqttClient-esphome@^0.8.6
```

I can download 300kb image (of a cat ;) repeatedly without any flickering of the addressable leds (WS2812). Of course I will have to rewrite a lot of my code but I'm working on it.

kitecraft, 2022-09-30:
Same problem as well. ESP32-C3    FastLED flicker when WiFi enabled.

scottlawsonbc, 2022-12-28:
Also having the same issue.

TechOverflow, 2023-02-27:
Can confirm the above, didn't want to create a new issue.
ESP32-C3-WROOM-02
Arduino IDE and FastLED latest versions.
Whenever WiFi is on, there is severe flickering. When I switch to Adafruit Neopixel library, the problem goes away.

rexshamhk, 2023-03-15:
> Can confirm the above, didn't want to create a new issue. ESP32-C3-WROOM-02 Arduino IDE and FastLED latest versions. Whenever WiFi is on, there is severe flickering. When I switch to Adafruit Neopixel library, the problem goes away.

ESP32-S3-WROOM-1 also having the same issue.
Actually FastLED, Adafruit's Neopixel, NeoPixel Bus libraries are having the same issues.
If I use RMT / bitbanging method to drive LEDs, it always fails to light up leds correctly if I use wifi.
If I am not using wifi, it works well.

Monacrylic, 2023-04-23:
I'm using an M5Stack C3 Mate. Exact same problem with both libraries, FASTLED as well as Adafruit Neopixel. Both work without WiFi.

TechOverflow, 2023-04-23:
I have been using Adafruit successfully ever since. My lights have 60 and 180 LEDs and do not flicker. They are on 24/7 and send updates via wifi every 10 minutes.

Monacrylic, 2023-04-23:
> I have been using Adafruit successfully ever since. My lights have 60 and 180 LEDs and do not flicker. They are on 24/7 and send updates via wifi every 10 minutes.

My bad, just checked again, I was using a PIN involved in serial connection. Changed the data pin, Adafruit Neopixel works perfectly. Thanks!

azuresp, 2023-05-16:
I'm seeing the same problem with an Adafruit Feather ESP32-S3. I observe seeing flakiness when I turn on either the WiFi or the battery voltage sensor.

11CaptainKirk, 2023-07-13:
If you are seeing flicker only with WiFi/BLE on (especially with both on), make sure you are using a level shifter. Technically 3.3V is already out of spec for most LEDs, but they seem to work fine most of the time. Turning on WiFi/BLE likely causes very slight voltage droop on the GPIOs which sends it just out of the range of working. Make sure the chip you use is fast enough, though; I used an sn74hct buffer/driver.

greycoding, 2023-07-15:
I have an **Adafruit QT Py ESP32 C3** and can confirm that with WiFi enabled LEDs flicker when I use FastLED.
When I use the NeoPixel library it does not flicker.

I tested the behavior with three scenarios:

1. LED Strip DIN connected directly to the MCU.
2. LED Strip DIN connected to the MCU via a 470 Ohm resistor.
3. LED Strip DIN connected via a SN74HCT125N.

In all three scenarios the LEDs flickered with the FastLED library when I turned on WIFI. On the other hand, when I use the NeoPixel Library, the LEDs do not flicker.

Strangely enough, the LEDs with the FastLED library also flickered when I enabled WiFi only once at the beginning and disabled it right after.

void turn_WIFI_OFF()
{
  WiFi.mode(WIFI_MODE_NULL);
  WiFi.mode(WIFI_OFF);
  esp_wifi_stop();
  Serial.println("WIFI OFF");
  delay(2000);
}

toybuilder, 2023-11-26:
This is only tangentially related, but I had a project on ESP32 (non-C3) that was using Arduino ESP support with FastLED for a Bluetooth (not WiFi) project.  The project worked for years and then immediately failed when I made some changes to adjust pixel locations.

After a bit of digging around, I discovered that the particular version of Arduino ESP mattered -- the newer version that I used since the original deployment resulted in some kind of memory corruption that would cause the project to occasionally crash and often resulted in weird output that could be described as flickering and artifacts.

In my case, I had to back-rev all the way back to esp32 1.0.6 to get it to work reliably.

If some people are having good success with ESP32-C3 and others are not, it might be worth comparing the ESP BSP versions.

old-Jerry, 2023-12-26:
When I ported the awtirx2.0 project to xiao eps32-c3, the display flickered abnormally when I turned on the wifi, and I also failed to turn off the wifi after connecting it
Do you have an explanation for the solution?

jekader, 2024-01-16:
I'm hitting the same issue while porting PixelIt to ESP32-C3, to me it looks liks like the timings are off. Looking at previous reports suggesting this is linked to WiFi and given that the C3 is a single-core MCU, there might be hardware interrupts messing things up and causing flickering.

I've seen a patch addressing this for the 8266 which the C3 is a replacement for:
https://github.com/maddogjt/FastLED/commit/ae4c696d53981bd24cc6932593204b906b82d786

Maybe we can come up with something similar for C3 to see if it solves the issue, especially given that other LED libraries don't seem to be affected?

bradrblack, 2024-01-20:
Having same issue here with ESP32-C3 "super mini" boards and various 8x8 WS2812B matrices.  Activated pixels seem to randomly turn to max bright white and flicker. Would love to see a fix.  For another project, I moved from FastLED to Adafruit_NeoPixel/NeoMatrix and the problem was solved (so I don't think it is strictly a HW issue) but would prefer not to re-port multiple projects.

bradrblack, 2024-01-21:
It seems my issue with flickering was fixed with https://github.com/FastLED/FastLED/pull/1515 which has been merged into master but not part of the 3.6.0 release.

jekader, 2024-01-21:
Wow, @bradrblack thanks for the info! I can confirm that FastLED from master works properly on my esp32c3 supermini!

L3-N0X, 2024-01-26:
I can also confirm that the flickering is away with FastLED from master on my esp32-C3 supermini.
But however I'm using painlessmesh and execute everything which normally would be in loop() in another function which is called by the task sheduler. That's necessary to keep the mesh stable. But now, it seems like everytime the painlessmesh has to do something and adds tasks to the task sheduler, the Frame-Rate of my LEDs drops rapidly and the LEDs won't continue with their pattern for a second. Is this a hardware limitation, an error with FastLED or just a incompatibility with FastLED and painlessmesh on ESP32-C3 (Because of CPU Architecture?)

Code works perfectly on a Wemos D1 mini (ESP8266) and ESP-01 without those Framerate drops.

kokospalme, 2025-04-23:
[#1515](https://github.com/FastLED/FastLED/pull/1515) did Not help in my case. I already tried using Adafruirts Neopixel library: FastLED did the math, and then I copied about 230px to the neopixel array and send it out to ´´´PIN 10´´´.
´´´gpio_matrix_out´´´ was already at 0x80. I use FastLED within a library which get it's data through setter and getter.
Does anyone still facing this issue?