FastLED/FastLED #1354: Can't use on-board LED on Adafruit QT Py SAMD21

thnikk, 2022-01-21:
The built-in neopixel on the QT Py SAMD21 (pin 11) is not working with FastLED. When compiling, I get this error:

```
In file included from .pio/libdeps/2k-touch/FastLED/src/FastLED.h:50,
                 from src/main.cpp:5:
.pio/libdeps/2k-touch/FastLED/src/fastpin.h: In instantiation of 'class FastPin<11>':
.pio/libdeps/2k-touch/FastLED/src/fastpin.h:242:29:   required from 'class FastPinBB<11>'
.pio/libdeps/2k-touch/FastLED/src/platforms/arm/d21/clockless_arm_d21.h:10:54:   required from 'class ClocklessController<11, 12, 30, 18, GRB, 0, false, 50>'
.pio/libdeps/2k-touch/FastLED/src/chipsets.h:578:7:   required from 'class WS2812Controller800Khz<11, GRB>'
Compiling .pio/build/2k-touch/lib496/Adafruit_TinyUSB_Arduino/arduino/ports/nrf/Adafruit_TinyUSB_nrf.cpp.o
.pio/libdeps/2k-touch/FastLED/src/FastLED.h:94:34:   required from 'class NEOPIXEL<11>'
.pio/libdeps/2k-touch/FastLED/src/FastLED.h:316:28:   required from 'static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with CHIPSET = NEOPIXEL; unsigned char DATA_PIN = 11]'
src/main.cpp:94:52:   required from here
.pio/libdeps/2k-touch/FastLED/src/fastpin.h:210:24: error: static assertion failed: Invalid pin specified
  210 |  static_assert(validpin(), "Invalid pin specified");
```

I'm using platformio, but the same thing happens with the RGBSetDemo using the Arduino IDE.

marmilicious, 2022-01-21:
There was another Adafruit board, I think the Bluefruit, and to specify the pin for the onboard pixels, instead of 8 or 11, etc, it was **PIN_NEOPIXEL**.  You might give that a try.

thnikk, 2022-01-21:
I actually tried it right before posting this issue and it returns the same error. I checked the variant.h file for the board since I know there are usually pin definitions in there and it's defined as pin 11.

Spegs21, 2022-02-07:
@thnikk The pin definition is missing in fastpin_arm_d21.h:
https://github.com/FastLED/FastLED/blob/4d73cddfe4bd2b370ee882b6f68769bf7f8309f4/src/platforms/arm/d21/fastpin_arm_d21.h#L236

You need to add `_FL_DEEPIN(11,18,0);` in the `#elif` block and change `#define MAX_PIN 10` to `#define MAX_PIN 11`.

Sheepybloke2-0, 2022-03-12:
This issue that I found might also be relevant here: https://github.com/FastLED/FastLED/issues/1363#issuecomment-1065957796

Brannan-repo, 2024-03-08:
> _FL_DEEPIN

Sorry to a stickler, but this took me a while to figure out you have a typo in your code, it should be _fl_deFpin, not 2 E's

closed 2026-09-01 by zackees