FastLED/FastLED #1359: RP2040 Support

gsexton, 2022-01-26:
Is there any plan for implementing Raspberry Pi RP2040 support? I built a project that uses Fast LED with an Adafruit ItsyBitsy RP2040, and got to the end and found that it's unsupported. The project I've built is pretty cool:

https://www.instructables.com/Animated-Led-Star/

The project author wrote about being resource constrained, so I thought this would be better. I didn't realize FastLED had platform dependencies :(  Since the rp2040 has floating point, I don't think implementing the math libraries would be pretty easy. I would be willing to kick in some hardware for a committer to do the port.

If there's no plan, or it's quite a ways out, what would be a good board to use? Preferably one that would be a little on the higher end for CPU/Memory?

Thanks in advance!

George

gsexton, 2022-01-28:
I found the pull request for rp2040 support, and it worked for me, so I'm closing this. It would be great if you could merge that pr.

closed 2022-01-28 by gsexton

chatelao, 2022-04-18:
What ID had the RP2040 pull request.

nmaas87, 2022-05-10:
https://github.com/FastLED/FastLED/pull/1261