FastLED/FastLED #1363: FastLED not functional on Trinket M0 with WS2812B

Spegs21, 2022-02-02:
I've been trying to get the blink sketch running on a Trinket M0 but have so far been unsuccessful. Output from pin 1 is buffered to through a 1g126 to 5V. The same setup works just fine with the Neopixel library.

I have a picoprobe SWD hooked up to it and it's not stuck or in hardfault. I'm not sure what to look at to debug this issue.

Spegs21, 2022-02-09:
Forcing software pins in the config allows it to function. I checked the pin with a scope in hardware mode and there is no activity.

JesseMFrey, 2022-02-25:
@Spegs21 which config are you referring to? I'm having a similar problem with on the QtPy SAMD21. Works with Neopixel library, but not FastLED. weird part is it worked before, can't figure out what changed

Spegs21, 2022-02-25:
@JesseMFrey In the fastled_config.h file:
https://github.com/FastLED/FastLED/blob/4d73cddfe4bd2b370ee882b6f68769bf7f8309f4/src/fastled_config.h#L12

Spegs21, 2022-02-25:
It's probably just all SAMD21 based boards. @samguyer can you comment on why this might be happening?

samguyer, 2022-02-25:
@Spegs21  I'm not an expert in the ARM support for FastLED. I wonder if it's just a pin naming problem? But @JesseMFrey , you are saying that it *used* to work and now it doesn't?

JesseMFrey, 2022-02-25:
@samguyer yes, it used to work. Sadly I was not paying enough attention to know which versions of things I was using before that worked.

samguyer, 2022-02-25:
@JesseMFrey That's strange. Looking at the history, I don't see anything substantial committed in some time. Do you know which specific ARM platform was affected? The D21?

JesseMFrey, 2022-02-25:
@samguyer presumably, I've only used it with the QtPy SAMD21

Spegs21, 2022-02-25:
Seeing as these are both Adafruit boards I wonder if it's related to a Adafruit SAMD Core change.

Sheepybloke2-0, 2022-03-12:
Oh! I just solved this issue with my Gemma M0! This issue is with the fact that this [chunk of the pin definitions **](https://github.com/FastLED/FastLED/blob/4d73cddfe4bd2b370ee882b6f68769bf7f8309f4/src/platforms/arm/d21/fastpin_arm_d21.h#L130-L148) is** before the rest of the board definitions! When the Arduino library is compiling, it spits out two defines, one for the board and one for the core (e.g. "ADAFRUIT_GEMMA_M0" and "ARDUINO_SAMD_ZERO"). But, because the default pin definition for "ARDUINO_SAMD_ZERO" is before the platform definition, that is the set of pins that is chosen. 

A temporary solution is putting your platform above the "ARDUINO_SAMD_ZERO" core definition, but maybe that definition should be placed last in the config since it's the most generic? Otherwise, I'm not sure if it is proper for both of those defines to be specified for the compiler, so one could be removed or an Adafruit configuration updated to only include one of the defines. Ether way, this seems to be the issue at hand.

Spegs21, 2022-03-14:
@Sheepybloke2-0 This is a really interesting find. My understanding of those pin definitions is that they are supposed to be for specific boards but the ARDUINO_SAMD_ZERO block you linked is really weird. I'd expect it to the the Arduino Zero (SAMD21) but it has way too many pins. Maybe it's checking the wrong define for that block and it is supposed to be referring to a different board.

I'll play around with compiling for different boards to look at what all is defined.

lemuroid, 2022-06-03:
This issue was killing me the prior week i.e.  FastLED on a trinket M0.   I wasted much time before properly googling the solution found here.   Code that has worked well in the past suddenly stopped working on the exact same hardware.  thanks for solving this one.

CsVance82, 2022-09-15:
> Oh! I just solved this issue with my Gemma M0! This issue is with the fact that this [chunk of the pin definitions [...]

This worked.

billpugh, 2022-10-30:
I did some exploration; using FastLED on a Trinket M0 works through version Adafruit SAMD 1.7.7. 

In Adafruit SAMD 1.7.8, I get a verification error when uploading to the trinket M0.

The bug described above is present in versions 1.7.9 and 1.7.10. 

I've posted this in the Adafruit forums, which is their recommended way to get their attention. 
https://forums.adafruit.com/viewtopic.php?t=195647

lemuroid, 2022-10-31:
I believe the work-around is to edit fastled_config.h   to uncomment this line (line 11 methinks) 
#define FASTLED_FORCE_SOFTWARE_PINS//uncomment for trinket m0
you may have to re-comment those lines for other boars.   (at least I do)

 
 

https://forums.adafruit.com/viewtopic.php?t=195647

hathach, 2022-11-02:
this should be fixed in Adafruit core release 1.7.11

BrookeDot, 2025-09-14:
I know this is an older issue but it seems the problem cropped up again for me in 3.10.2.  Earlier versions including 3.10.0 and 3.10.1 compile.

There were quite a few changes between 3.10.1 and 3.10.2: https://github.com/FastLED/FastLED/compare/3.10.1...3.10.2 so hoping someone more familiar with the changes can take a look. 

I am using a Trinket M0 with board version [1.7.16](https://github.com/adafruit/ArduinoCore-samd/releases/tag/1.7.16).

```
FastLED/src/platforms/compile_test.cpp: In function 'void fl::compile_tests()':
FastLED/src/platforms/compile_test.cpp:48:47: error: static assertion failed: u32 must be exactly uint32_t
   48 |     static_assert(fl::is_same<u32, uint32_t>::value, "u32 must be exactly uint32_t");
      |                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
FastLED/src/platforms/compile_test.cpp:51:46: error: static assertion failed: i32 must be exactly int32_t
   51 |     static_assert(fl::is_same<i32, int32_t>::value, "i32 must be exactly int32_t");
      |                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
FastLED/src/platforms/shared/active_strip_data/active_strip_data.cpp: In member function 'fl::string fl::ActiveStripData::infoJsonString()':
FastLED/src/platforms/shared/active_strip_data/active_strip_data.cpp:81:22: warning: structured bindings only available with '-std=c++17' or '-std=gnu++17'
   81 |     for (const auto &[stripIndex, stripData] : mStripMap) {
      |                      ^
FastLED/src/platforms/shared/active_strip_data/active_strip_data.cpp: In member function 'fl::string fl::ActiveStripData::infoJsonStringNew()':
FastLED/src/platforms/shared/active_strip_data/active_strip_data.cpp:116:22: warning: structured bindings only available with '-std=c++17' or '-std=gnu++17'
  116 |     for (const auto &[stripIndex, stripData] : mStripMap) {
      |                      ^
exit status 1

Compilation error: exit status 1
```

Thanks!

zackees, 2025-09-16:
This is a great idea. This issue now has contradictory information and the issue has to be extracted out.

It would be great to have a new issue that's clean.

zackees, 2025-09-16:
Here is a report on the current state of this bug according to chat gpt:

**Summary of the GitHub issue FastLED/FastLED #1363**

---

### Original Problem (2022)

* **Boards affected:** Adafruit Trinket M0 (and other SAMD21-based boards such as Gemma M0 and QtPy SAMD21).
* **Symptoms:** FastLED “blink” sketch would not drive WS2812B LEDs, while the Adafruit NeoPixel library worked.
* **Early finding:** Forcing software pins (`#define FASTLED_FORCE_SOFTWARE_PINS` in `fastled_config.h`) made it work, but with hardware-driven pins there was no output.

---

### Root Cause Identified

* **Pin-definition conflict in Adafruit’s SAMD core.**
  When compiling, both a board-specific macro (e.g. `ADAFRUIT_GEMMA_M0`) and the generic `ARDUINO_SAMD_ZERO` macro were defined.
* The FastLED configuration chose the generic `ARDUINO_SAMD_ZERO` block first, giving the wrong pin map.
* **Workarounds / fixes**:

  * Temporarily reorder or edit the pin-definition blocks, or
  * Uncomment `FASTLED_FORCE_SOFTWARE_PINS`.

---

### Upstream Resolution

* Adafruit corrected the macro definitions in their core (fixed in **Adafruit SAMD core 1.7.11**).
* FastLED also merged a commit (#1424) to handle Adafruit D21-based boards better.
* After those changes the original 2022 problem was considered fixed.

---

### New Regression (Sept 2025)

* **BrookeDot** reports that with **FastLED 3.10.2** (using Trinket M0, Adafruit SAMD core 1.7.16) the sketch no longer compiles.

* **Compiler errors:**

  ```
  static assertion failed: u32 must be exactly uint32_t
  static assertion failed: i32 must be exactly int32_t
  ```

  plus C++17 “structured bindings” warnings.

* Earlier FastLED versions **3.10.0 and 3.10.1 compile fine**, so something introduced in 3.10.2 broke the build for this board.

* Another user (**lemuroid**) confirmed the same failure and reverted to an older FastLED revision.

---

### Current Status & Maintainer Guidance

* The **original 2022 hardware/pin-definition bug is resolved upstream**.
* The **latest compilation failure is a new and different issue**, introduced between FastLED 3.10.1 and 3.10.2.
* Maintainer **@zackees** notes the thread now mixes the old fixed bug with this new regression and asks for **“a new issue that’s clean”** so the 3.10.2 compilation problem can be tracked separately.

---

**In short:**
This GitHub thread originally tracked a SAMD21 pin-definition bug that Adafruit and FastLED already fixed. In September 2025 a different problem appeared: FastLED 3.10.2 no longer compiles on the Trinket M0 due to `uint32_t`/`int32_t` type assertions. The maintainer has requested that anyone investigating the 3.10.2 regression open a **new GitHub issue** dedicated to that fresh problem.