FastLED/FastLED #1364: How to use the same effect (Comet) multiple instants with Different Speed and Different Color.

samanthasiriwardena1990, 2022-02-13:
Hope you all doing fine these days. I am trying to replicate these futuristic blue/purple lights that move on kind of that maze shape using the FastLED library. timestamp 0:00 - 0:10 seconds of the video. I was thinking of the Comet effect but I have no idea how to use it multiple times in a single LED strip in the same segment. I want to use it multiple times, with different speeds, and at least use two different colors. and with some randomiztions. and I don't like to use the delay() inside my code. so I'm asking you all good friendly people in this forum could you please point me in the right direction.
video link.
[https://www.youtube.com/watch?v=f_b2ixiEaiM](url)

Thank you.

marmilicious, 2022-02-13:
Hello @samanthasiriwardena1990 
We use github "issues" here mostly just for bugs or improvements in the library itself.  For all general discussions, help, and how-to, please use the FastLED Reddit group here: https://www.reddit.com/r/FastLED

Also, that YouTube link isn't working for me. Please check as update as needed when posing to the Reddit group.

samanthasiriwardena1990, 2022-02-13:
> Hello @samanthasiriwardena1990 We use github "issues" here mostly just for bugs or improvements in the library itself. For all general discussions, help, and how-to, please use the FastLED Reddit group here: https://www.reddit.com/r/FastLED
> 
> Also, that YouTube link isn't working for me. Please check and update as needed when posing to the Reddit group.

thank you marmilicious for your kind reply.
I'm really sorry I posted this in the wrong place.
Please accept my apologies.
I was searching for weeks to do this but could not find a way and I already post it on Reddit but moderators might be in the Deep Sleep so that's why I posted here. and I fixed the youtube link.
thank you all.

marmilicious, 2022-02-13:
I see a new post on Reddit, but did you previously post there?  I don't see any other posts to FastLED by that username.

closed 2024-10-12 by zackees