FastLED/FastLED #1365: Parse error in template argument list    FastLED WS2182B

WhatsThisBoxFor, 2022-02-13:
Getting parse error in template, Arduino Uno board, just randomly did this and only does it on this file, I can go download the DemoReel100 and it is set up the same way and it will upload and display fine. any help?

 here's the code causing the error.

`  FastLED.addLeds<WS2812B, DATA_PIN_B, GRB>(ledsB, NUM_LEDS_STRIP_2);
`

And here's the error code 

```
V1_Dakota_Boys:23:69: error: no matching function for call to 'CFastLED::addLeds<WS2812B, <expression error>, <expression error> >(CRGB [64], int)'
   FastLED.addLeds<LED_TYPE, DATA_PIN_A, GRB>(ledsA, NUM_LEDS_STRIP_1);
                                                                     ^
In file included from C:\Users\tulli\OneDrive\Documents\Arduino\V1_Dakota_Boys\V1_Dakota_Boys.ino:1:0:
C:\Users\tulli\OneDrive\Documents\Arduino\libraries\FastLED\src/FastLED.h:225:130: note: candidate: template<ESPIChipsets CHIPSET, unsigned char DATA_PIN, unsigned char CLOCK_PIN, EOrder RGB_ORDER, long unsigned int SPI_DATA_RATE> CLEDController& CFastLED::addLeds(CRGB*, int, int)
  template<ESPIChipsets CHIPSET,  uint8_t DATA_PIN, uint8_t CLOCK_PIN, EOrder RGB_ORDER, uint32_t SPI_DATA_RATE > CLEDController &addLeds(struct CRGB *data, int nLedsOrOffset, int nLedsIfOffset = 0)
```

Spegs21, 2022-02-25:
It's complaining about DATA_PIN_A / strip 1. Is DATA_PIN_A defined correctly? It may be an invalid pin. Check for invalid pin errors.

WhatsThisBoxFor, 2022-02-27:
> It's complaining about DATA_PIN_A / strip 1. Is DATA_PIN_A defined correctly? It may be an invalid pin. Check for invalid pin errors.

That was Exactly it! Somehow, I deleted the PIN number after the DATA_PIN_A definition. Thank You!

closed 2022-02-27 by WhatsThisBoxFor