FastLED/FastLED #1374: Bug with WS2812 and ESP32/8266 (solved with delay between .shows)

dgm3333, 2022-03-10:
For years I've had problems with running WS2812 strips from ESP32/8266 but had no problems running from a nano or other arduino.
I'd always assumed it was the voltage difference but despite trying numerous highspeed stepups I'd never resolved it and had given up. But I yet again wanted a wireless driver so tried again today.
Testing with my oscilloscope I discovered that the outputs with identical test code from a nano and an ESP are vastly different - with the ESP there is essentially no delay between updates with the ESP but there is a significant on with the nano. Adding this delay in code totally resolved the problem!!! Even the very basic demo example below was previously failing and now works.

The only compiler difference I can see is a pragma message that the ESP is defaulting to bitbanged due to lack of hardware SPI (whereas nano doesn't). 

I'm using Arduino 1.8.15 and FastLED 3.5.0

```
#include <FastLED.h>

#define LED_PIN     33
#define COLOR_ORDER GRB                                       
#define LED_TYPE WS2812     
#define NUM_LEDS    300
#define MAX_BRIGHT 255

struct CRGB leds[NUM_LEDS]; 

void setup() {
  FastLED.addLeds<LED_TYPE, LED_PIN, COLOR_ORDER>(leds, NUM_LEDS);
  FastLED.setBrightness(MAX_BRIGHT);         
}

void loop () {
    uint8_t thisSpeed = 10;
    uint8_t deltaHue= 10;
    uint8_t thisHue = beat8(thisSpeed,255); 
    fill_rainbow(leds, NUM_LEDS, thisHue, deltaHue);            
    FastLED.show();
    delayMicroseconds(75);    // THIS IS THE SECRET SAUCE !!!
}
```

samguyer, 2022-03-28:
@dgm3333 There *should* be an enforced minimum delay in the driver itself (a delay of 50 microseconds, which is the latch time), but I think there was a bug in it. I'll check.

CeriniA, 2022-05-06:
Hello! i have the same problem when NUM_LEDS is greater than 75. i am working with ESP32 and WS2812b.