FastLED/FastLED #1380: How to compile for ESP32-S3 using ESP-IDF?

niliha, 2022-04-01:
Hi, I'm trying to include FastLED (master) as an IDF component in ESP-IDF v4.4 but the build fails.
I'm using  Arduino core (master) as in https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html.

Using the `CmakeLists.txt` from https://github.com/FastLED/FastLED/pull/1352, this is my build log:
[build.txt](https://github.com/FastLED/FastLED/files/8397781/build.txt)

When targeting ESP32 the build succeeds.

Thank you, any help is appreciated!

solhuebner, 2022-04-10:
@niliha I will check when I am back home

solhuebner, 2022-04-13:
Hi @niliha, I went through the exercise of updating my project to the master version of FASTLED and https://github.com/espressif/arduino-esp32/releases/download/2.0.3-RC1/esp32-2.0.3-RC1.zip and even though some compile warnings are normal it compiled ok for me. Then I checked your build file and saw that you are using an esp32-s3 but your toolchain is still esp-2021r2-8.4.0 and should be patch3 according to the release notes / maybe also try the rc1 instead of head and make sure that you are using ESP-IDF 4.4 / I am using VS Code with the Espressif IDF extension and that installs patch2 if you are selecting the 4.4 release / which IDE do you use?

niliha, 2022-04-18:
Hey @solhuebner, thank you very much for looking into it!
Glad to see, that you can compile it.

While I also freshly installed ESP-IDF 4.4 through the VS Code extension, your observation appears very promising.
I will report back to you this week.

niliha, 2022-04-25:
I'm sorry, it took me a while!

So indeed something was wrong with my IDF version.

I'm now using
* [ESP-IDF Release v4.4.1](https://github.com/espressif/esp-idf/releases/tag/v4.4.1)
* [ESP32 Arduino 2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) 
* [FastLED master](https://github.com/FastLED/FastLED/commit/08388047ac3e0ebcaafe71563ae2a55421d70c02) with your [CMakeLists.txt](https://github.com/FastLED/FastLED/blob/489baaba551d7e557b4011d8bef3c79da2096ecb/CMakeLists.txt)

Unfortunately it still doesn't compile for ESP32-S3 (while it does for ESP32) ([build.txt](https://github.com/FastLED/FastLED/files/8558490/build.txt)).
I created a [minimal reproducible example](https://github.com/niliha/esp32-s3-fastled-idf-example) which might help.

Thank you!

solhuebner, 2023-03-23:
https://github.com/FastLED/FastLED/pull/1352#issuecomment-1479370439

closed 2023-11-25 by niliha