FastLED/FastLED #1383: FastLED nto working on ItsyBitsy nRF52840

shurik179, 2022-04-15:
I am trying to get FastLED working with Adafruit's ItsyBitsy nRF52840 Express. I know support is considered experimental but so far, it is not working at all :(

Here are details: 
  - FastLED version: 3.5
  - connections: I am using APA102 (dotstar) strip, connecting Data to ItsyBitsy SPI MOSI pin, and Clock, to SPI SCK pin
  - Code: FastLED Blink example with minor changes (attached)
  - Setup: 
```
#define DATA_PIN 24
#define CLOCK_PIN 25
CRGB leds[NUM_LEDS];

void setup() { 
     FastLED.addLeds<DOTSTAR, DATA_PIN, CLOCK_PIN, BGR>(leds, NUM_LEDS);  // BGR ordering is typical
}  
```
 Pin numbers for MOSI and CLK pin are taken from https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/variants/itsybitsy_nrf52840_express/variant.h#L101 

- result: LEDs do not light up at all or light up randomly

I checked the connections and power (5v) using multimeter - they are OK

The sketch file is attached
[FastLED-nrf52840.ino.txt](https://github.com/FastLED/FastLED/files/8496657/FastLED-nrf52840.ino.txt)

shurik179, 2022-04-15:
Just for the record, the pin numbers I used above match what is in FastLED repository in 
arm/nrf52/fastpin_arm_nrf52_variants.h : 
https://github.com/FastLED/FastLED/blob/master/src/platforms/arm/nrf52/fastpin_arm_nrf52_variants.h#L390
so it should work.. but doesn't

josephreisinger, 2022-08-02:
Not sure an additional report is helpful, but confirming the same results for me on an APA102 strip, connected to the same pins. I also tried connecting to a few other combinations, e.g. 
```
#define DATA_PIN 13
#define CLOCK_PIN 12
```
thinking maybe it was the hardware SPI but same effect (random colors / very blinky).

zackees, 2022-08-05:
Can you try disabling hardare spi and see if it fixes it?
```
#define FASTLED_FORCE_SOFTWARE_SPI
#include <FastLED.h>
```

josephreisinger, 2022-08-08:
@zackees confirming that clears up the issue on my end, both using `APA102` and `LPD8806` controllers with the ItsyBitsy nRF52840.

closed 2024-09-19 by zackees