FastLED/FastLED #1386: How to call/ change LED colour by input(text)

lc2047, 2022-04-28:
Hi 

How to change the LED color by text input?
In FastLED, there have serval methods.
1. leds[i] = CRGB::Red
2. leds[i] = CRGB(0,0,255);
3. leds[i] = 0xFF0000;
4. leds[i].setRGB(0,0,255);

However, I prefer using the predefined colors list of FastLED. The list has well defined in "pixeltypes.h". 
If duplicate predefined colors list in my code, seem it is not good for memory efficiency. Also, have a risk(not align) in FastLed update.
I am a beginner in programming and don't know how to convert color name("Red") to RGB data(e.g. 0xFF0000). 

Please help and share your comments with me.

Sincerely

LC

marmilicious, 2022-04-28:
Hello LC. We use github "issues" here mostly just for bugs or improvements in the library itself.  For all general discussions, help, and how-to, please use the FastLED Reddit group here: https://www.reddit.com/r/FastLED

lc2047, 2022-04-29:
@marmilicious 

Thanks & note.

lc2047, 2022-04-29:
move to correct area

closed 2022-04-29 by lc2047