FastLED/FastLED #1387: ESP32 Harware SPI is not documneted but is recommended by compiler

burner-, 2022-05-08:
https://github.com/FastLED/FastLED/wiki/SPI-Hardware-or-Bit-banging tells that some of hardware support harware SPI. 
With esp32 compiler says pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
By googling that I find tickets after tickes and frustrated people mashing their heads to wall by trying use that HW SPI. Anyway looks that there is no documentation that and most of them had find mostly workarounds. So I recommend to documents following things:
- What defines user need to set to get it work
- What pins are supported for that purpose.

braddo99, 2022-09-12:
I also don't yet have this working, but at least to get closer, one add a couple of defines prior to including fastled.h. This is in the comments of the pull request that added hardware SPI support for ESP32:
https://github.com/FastLED/FastLED/pull/1047

Like so: (but beyond that it's still not working for me, anyone have a link to a working example with hardware SPI on ESP32?)

#define FASTLED_ALL_PINS_HARDWARE_SPI
#define FASTLED_ESP32_SPI_BUS HSPI

#include <FastLED.h>
...

Subcode, 2023-01-09:
I'm looking for an answer on this aswel. Following the advice in tickets does not work.
using it in platformio gives linker errors.

zackees, 2023-01-09:
I'm working with ESP32 and I can say that it's likely you guys are trying to use the ESP32 on either Arduino or Platformio. Both of these programs use the arduino as a component system which includes a copy of the idf toolchain which actually manages configuration, compilation, flashing and monitoring.

The issue here is that the all important sdkconfig file which manages global configuration of esp32 projects (Defines/include paths etc) is hidden from the developer and it's difficult to change. This is a major oversight of the espressif and hopefully it will be solved one day so that configs in a local setting (platformio.ini and whatever the heck arduino ide uses) will be imported into the sdkconfig file.

My advice is to try and figure out how to get access to this sdkconfig file and change it without the arduino/platformio version of the idf.py toolchain resetting it, which it loves to do.

Otherwise, it might be time to try installing the idf.py toolchain, then install the arduino as a component so that you get access to the familiar arduino library functions. Make sure you are using the idf v4.X toolchain. If you go to the 5.X series then the arduino component won't compile as of this writing.

Also, I really like the idf.py toolchain. Its pretty quick to learn and unlocks all the great features of the chip, including PSRAM, which gives you > 4MB heap. Whatever it is that you want to do, the idf.py toolchain will give you what you want. Arduino and platformio is a crap shoot for any particular feature.