FastLED/FastLED #1388: Feature Request / Clarification: Set maximum power consumption per led "strip"

mbariola, 2022-05-08:
Hi,

Is there a way to specify max power consumption per each strip? I have 2 strips who are powered by two different power supplies, with different amperage capabilities. I was able to specify how to control both strips independently, however having a single power cap could lead to either both strips not reaching the amount of brightness they could, or if I set it to the sum of power from both supplies, chances are I might burn one or either. Thanks

zackees, 2022-05-08:
Why not just get a higher amperage supply or supply a custom led controller yourself? The library gives you an escape hatch to add your own custom control classes.

https://github.com/FastLED/FastLED/blob/b5874b588ade1d2639925e4e9719fa7d3c9d9e94/src/chipsets.h

mbariola, 2022-05-09:
I see your point, @zackees. 
My problem is not the available power, which is enough, but the current connections and cable thicknesses I have for one of the strips. I'm reusing hardware from an old prototype and the esp32 and the strip are physically and electrically tightly integrated and rewiring / resoldering would be a pain. So I am actually concerned that too much of a draw by this "frail" part will overheat or wear those thin wires out.

I think that a full 360 degrees coverage of multiple strips needs the possibility to also have constraints issued for each of them. My C++ chops are not on par with the library code, so odds are I won't be adding the functionality myself anytime soon

smurphyUSI, 2022-06-05:
Did anybody get back to you on this question?    I would suggest the Reddit for further questions, as more people frequent those posts than here.

If you didn't get an answer.

FastLED lets you use multiple LED controllers (i.e FastLED.addLeds) at a one time.

You can use 'setBrightness' on each controller to set the maximum power draw of each section independently.
'setBrightness' is actually a scale, so turns down the brightness (hence power use) of any subsequent LED draws.
I personally plug a power meter into the outlet and calibrate my LEDs by drawing all White at max power to find the level I need.

If you check this example, you will find what you probably need in the 'One array, many strips' section.
https://github.com/FastLED/FastLED/wiki/Multiple-Controller-Examples

'addLEDs' has an 'offset and length' method, which lets you use a single array for all of your LEDs, but then lets you specify which sections are controlled by a particular controller.