FastLED/FastLED #1392: ESP8266,turn on wifi, need twice “show()" function to work....

planevina, 2022-05-29:
Hi.
I want to do an "O-Clock" like this  "https://www.instructables.com/O-Clock/"
 I try to use esp32-c3 ,but it's not stable , many problems , so I go back to esp8266.

There is a strange problem in esp8266...

This function runs every second.↓↓↓

void OClockLed::modeSimple(uint8_t h, uint8_t m, uint8_t s, int tint_a, int tint_b)
{
    FastLED.clear();
    leds[h] += CRGB(255, tint_a, tint_b);
    leds[m] += CRGB(tint_b, 255, tint_a);
    leds[s] += CRGB(tint_a, tint_b, 255);
    FastLED.show();
}

========================
after leds show 6-8 times , it stucked 

finally I found that **when I  turn off wifi connection ,  It works good
when I turn on wifi ,It stucked after leds show 6-8 times ....**

But wifi is necessary for this project .
I add a new FastLED.show();   like this :

void OClockLed::modeSimple(uint8_t h, uint8_t m, uint8_t s, int tint_a, int tint_b)
{
    FastLED.clear();
    leds[h] += CRGB(255, tint_a, tint_b);
    leds[m] += CRGB(tint_b, 255, tint_a);
    leds[s] += CRGB(tint_a, tint_b, 255);
    FastLED.show();
    FastLED.show();
}

It's working good ...

So... maybe problem is esp8266 ,not  fastled ?

and ... 
Is there any othertrouble when I call the "show()" function twice in a row?

Thanks.

kostyamat, 2022-05-29:
leds[s] += CRGB(tint_a, tint_b, 255);
delay(1);
FastLED.show();

planevina, 2022-05-30:
> leds[s] += CRGB(tint_a, tint_b, 255); delay(1); FastLED.show();

It stucked, too.
"delay(20)" is same...

planevina, 2022-05-30:
"FastLED.delay(1)"  is working.

prodestrian, 2022-06-25:
I'm experiencing this same problem unfortunately, which is how I came across this Issue.

I was able to consistently and reliably control the first LED in the strip (set it to Green or Red). But I couldn't change it to Blue, or illuminate any of the LEDs after that. It's bizarre. A non-Wifi test worked fine, it was only when I enabled Wifi that the problem started to occur.

I tried using `FastLED.delay(1);` but this didn't seem to help. And yet adding `FastLED.show()` **twice** completely fixed the problem, it's been running stable ever since.

If I get time I'll try to put together a demo sketch which illustrates the problem better (I'm out of time for today). But for now I've got a fully working sketch and FastLED is running smoothly, and I've added a comment pointing to this Issue to explain why we call `show()` twice.

nickion, 2023-04-22:
This appears still to be an issue. I'm using an async network library, web sockets and handling http requests, and I'd not seen problems with some fast running animation code until now. Having added a slow flasher which is doing a show() only on the led transitions, only the first led tends to be updated on the first show. Calling show on every loop() iteration it was fine, but as said by others, doing a double show seems to suffice. I did also notice incorrect colours on some occasions too, suggesting there may be some other consequences from the issue.

MindStudioOfficial, 2023-07-10:
I am also having this Issue. I am using a NodeMCU with the ESP8266 module on it to control my LEDs. My code uses Wifi and the EEPROM. Sometimes only the first LED updates.

danieljharris, 2023-07-26:
Having the exact same issue with ESP8266-12E

peterples, 2023-08-14:
Same problem, same solution. Is it possible that for some data it is enough to call **show** once and for others it is necessary to call twice? I am displaying an animated GIFs and some frames are displayed correctly and some are displayed incorrectly. Even some GIF will not appear at all if I call the **show** only once.

goddade, 2024-09-07:
Same problem. Only need to call show() twice before connecting to wifi.

Akrobate, 2024-12-29:
Same problem for me with a Wemos D1 ESP8266

davide1a, 2025-09-05:
I guess this was never solved? I've just got my first D1 mini, and having the same issue, after having used the fastLED library just fine on Arduino and esp32s. 

For me, it's the same to start with. I send a mqtt message to turn the LEDs on and off, and on the first try after rebooting, I have to send the message twice (first message only partially lights the first LED). The second send of the message turns all 8 LEDs on. But from then on, it works just fine. I can keep turning them on and off by sending just the one message.

I don't think the wifi is causing any issues, as I have tried turning the LEDs on in the setup before wifi is started, and it does the same thing.

zackees, 2026-04-19:
Quick triage pass — this falls in the WiFi ↔ bit-banged-clockless family of issues on ESP8266. The driver has been reworked since this was filed. Current state in master:

- **IRAM hot path** — `showRGBInternal` is `FL_IRAM`, avoiding iCache misses mid-frame.
- **Per-bit deadline abort** — if an NMI (or WiFi ISR) stretches a bit past `T1+T2+T3`, the frame aborts cleanly and `showPixels` retries. See [`clockless_esp8266.h:100-102`](https://github.com/FastLED/FastLED/blob/master/src/platforms/esp/8266/clockless_esp8266.h#L100-L102).
- **Configurable retry loop** — `FASTLED_INTERRUPT_RETRY_COUNT` controls how many times a dropped frame is retried.

Recommended current recipe when WiFi is active:

```cpp
#define FASTLED_INTERRUPT_RETRY_COUNT 2   // retry on interrupt-stretched frames
#define FASTLED_ALLOW_INTERRUPTS 1        // keep WiFi responsive between pixels
#include <FastLED.h>

void setup() {
    // optional — only if you're OK with reduced WiFi responsiveness:
    // WiFi.setSleep(WIFI_NONE_SLEEP);
    FastLED.addLeds<WS2812, DATA_PIN, GRB>(leds, NUM_LEDS);
    FastLED.setMaxRefreshRate(100);       // gentle upper bound helps with WiFi heavy load
}
```

Hardware checklist that frequently matters more than software for this symptom:
1. **Data-line level shifter** (74HCT245 / 74AHCT125) — ESP8266 is 3.3 V, most WS28xx strips expect 5 V logic. 3.3 V works borderline, glitches under electrical noise.
2. **≥1000 µF cap** on the LED strip's 5 V rail near pin 1.
3. **Short, direct data wire** (< 30 cm) with a ~220 Ω series resistor.
4. **Common ground** between ESP8266 and LED PSU.

If the flicker persists on master with the recipe above **and** the hardware checklist is clean, please share: your Arduino-ESP8266 core version, a 2–3 second scope/logic-analyzer capture of the data line during steady-state show, and whether the flicker correlates with WiFi packet activity. With that I can decide if there's still a driver-side fix needed, or if this one can close as "requires hardware remediation".

zackees, 2026-08-20:
Deferred during the bug burn-down pending a minimal current-master hardware reproduction, ESP8266 core version, wiring/level-shifting details, and logic-analyzer capture that can distinguish driver timing from electrical or WiFi-related causes.