FastLED/FastLED #1397: FastLED not C++17 compliant

JakeSays, 2022-06-07:
FastLED generates a ton of `register` keyword usage warnings when compiled with -std=c++17 (or greater). C++17 does not support the `register` storage class specifier.

smurphyUSI, 2022-06-11:
What compiler and for what CPU architecture are you targeting?  (I'm guessing you are using Visual Studio as an IDE for this).
It maybe practical to add build flags to check for this scenario.

It's worth pointing out that 'register' is current set by the standard to throw the warning, because the C++ committee doesn't know how it should work on modern home and server CPUs.
Once you enable any compiler optimizations for nearly any common CPU it becomes impossible for a coder to know if the 'register' specifier will do anything useful, because the compiler will probably choose something better anyways.
So, the only smart thing for the committee to do was to force people to stop using 'register' (at least for now).

This doesn't apply to older CPUs or the type of AVR chips FastLED is usually run on.
A smart compiler will automatically take care of the registers for you on these platforms, but 'speed-demon' programmers may want to do some very specific register optimizations on these platforms (and likely compile with something like gcc-avr).

luebbe, 2022-06-18:
How would I be able to turn these warnings off on PlatformIO just for FastLed?

JakeSays, 2022-08-17:
@smurphyUSI Sorry for the late reply. I just returned to the project using FastLED and remembered this issue.

I'm using the standard ESP32 toolchain from ESP-IDF, and my IDE is qtcreator on linux.

I found a simple enough solution:
add `#pragma GCC diagnostic ignored "-Wregister"` to fastpin.h. I'll submit a PR for this soon as I get a chance, if you're interested.

macsimski, 2023-03-12:
well, it pops up in a lot of files in this library. should i include the directive in all of them?

zackees, 2023-03-13:
FYI

https://github.com/FastLED/FastLED/pull/1460

closed 2024-08-08 by zackees