FastLED/FastLED #1402: Include Header Structure

kwautztretschke, 2022-06-29:
Is there a reason for the way the cpp files include their respective headers?
For example, the noise.cpp file doesn't include the noise.h file, but instead _every single_ file includes FastLED.h, resulting in you always having to include the _whole_ library.

I want to use the noise functions for a different project, but cannot simply include the pixeltypes.h and noise.h headers and be done, since all other headers are automatically included.

Is there a reason behind this, or is this just bad coding practice?

kwautztretschke, 2022-06-29:
It's impossible to isolate which cpp file needs which function prototypes (for example, do the 800 lines of code in noise.cpp depend on any of the math8.h functions?).

EVERYTHING depends on EVERYTHING.

zackees, 2022-06-30:
Yes, it's good practice to just include `fastled.h` and let the linker do dead code removal. Think of it more like a library instead of a collection of orthogonal modules. Restructuring the header files would likely affect stability and break existing users.

smurphyUSI, 2022-06-30:
The include strategy is neither messy, bad coding practice, nor uncommon.

As for the noise functions, I took a quick look at the code and it appears to be a forklift of Ken Perlin's noise estimate.
I just did a check for any other 'fast' Perlin noise gits and found one here, but there are more:
https://github.com/Auburn/FastNoiseLite

Additionally, FastLED doesn't have a very deep build-tree, so you can always just copy and paste the code sections you want of our FastLED and into your project.  You may only have to go one or two levels deeper for missing code.    If you share, you should, of course, credit that code section to FastLED.

kwautztretschke, 2022-06-30:
having every file depend on every other file instead of having a tree structure violates the need-to-know principle.
A cpp source file should only include its own corresponding header, and if need be some implementation-specific libraries.
The header file is intended for including dependencies inside the project.
If you draw out the header dependencies in a graph (or a class diagram, or in UML), then ideally a tree structure should be evident, and cross edges should be avoided. Working in tree structures is a very very common pattern applicable in many aspects of software engineering, and I've never seen this "everyone knows everyone" approach regarding headers anywhere else.

Especially on embedded, just hauling an entire library into the binary because "the linker will take care of it" isn't always the best idea.
I highly doubt that doing a proper include structure would break anything for anyone, because if so, then I'd argue the library could be considered more "broken" in it's current state.

Copy pasting code sections is a very bad way of "solving" this problem and goes against the very philosophy of using libraries.
This results in so much code duplication (with different versions of the same code each time!) over many projects, and is the reason why we invented shared libraries in the first place.
But maybe that's just not the way you do things anymore. Maybe, in the age of StackOverflow, copying and pasting random code snippets is just how you code nowadays.

alienozi, 2022-06-30:
I mus agree with @kwautztretschke, the structure of the header files are simply convoluted. Leaving the entire work to the compiler is just closing your eyes and hoping that the dumb computer will be smart enough to do a perfect job. Since we are talking about an embedded environment, every resource is important, so both the compiler _and_ the code must be optimized.
[image]

smurphyUSI, 2022-06-30:
Why the hell are you talking about 'need-to-know priciples and UML standards about programming languages which predate both of those by over two decades?
Did it occur to you that there are other, completely valid, methods of solving coding problems that aren't the way a perticular private coporation tells you is vaild?

Don't mistake header inclusion as code-dependency.  Even the most basic of linker will not screw this up, and this is not an optimization.  And, it IS the responsibility of the developer to include the correct target code either, which can be in the code, or the make, or even the headers.  The compiler and linker cannot, and will not, do this on your behalf.

And, why are you complaining about snipping code.   People have been snipping code forever.
I also gave you a suggested libraries for noise functions, so you wouldn't have to snip code.
Instead, you decide to complain about an LED library not cleaning your floors for you.
Get a hold of yourself.

zackees, 2022-07-01:
The person who made FastLED Danial Garcia died in tragic accident:

https://sweetlaw.com/daniel-garcia-and-yulia-krashennaya-killed-in-boat-fire-accident-near-channel-islands-santa-barbara-ca/

Asking for large refactors in this library is probably out of the question. We are just happy that the remaining contributors are continuing to provide direction and are responding to pull requests as new leds and micro processors become available. One day we may not be so lucky.

JakeSays, 2022-09-18:
I agree that the headers in fastled are a complete mess, regardless of what others may say in this issue. Including fastLed.h everywhere is a good practice solely because it's the only way to get fastLed to work.

I wrote a simulator to help with development of a device I was working on, but to get it to work *and* play nice with other code I ended up fixing most of the headers.

I'd be willing to contribute the changes if there is interest. If not then I'll just fork fastled and continue from there.

zackees, 2024-09-19:
I've been modularizing header includes so that something simple doesn't include the whole world. The big accomplishment was getting crgb.h without including fastled.h

Now that all the headers that noise.h depends on also don't depend on fastled.h, I have corrected the issue and this should now work as expected.

closed 2024-09-19 by zackees