FastLED/FastLED #1407: WS2818 is not working

kokospalme, 2022-07-11:
Hello,
I have a custom Board with an WS2818B chip, but FastLED does not work with it. I already tried seveal configurations (WS2812, WS813, WS2811 ...) but none of them worked with the WS2818. Furthermore the datasheet of the WS2818 says, for example the low-level-time has to be 580nS - 1 uS, but the WS2813 has a low-level-time of 220nS-240nS. So the timings are sligthly different. I wondered why others got their WS2818 ICs working with FastLED and the WS2813 configuration...

Can i change the timings within the fastLED library or something? 
Thank you!

zackees, 2022-07-11:
Yes, you have create your own LED controller class and pass that into the addLeds() function override.

https://github.com/FastLED/FastLED/blob/b5874b588ade1d2639925e4e9719fa7d3c9d9e94/src/chipsets.h#L577

kokospalme, 2022-07-12:
according to this post: https://github.com/FastLED/FastLED/issues/1311
I changed the timings:
`// WS2818B - 300ns 600ns 600ns
template <uint8_t DATA_PIN, EOrder RGB_ORDER = RGB>
class WS2818BController : public ClocklessController<DATA_PIN, C_NS(300), C_NS(300), C_NS(600), RGB_ORDER, 4> {};
/*
 * T1		== T0H (220 … 380  nS)	—> T1 = 370
 * T2+T3	== T1L (580 … 1000 nS)	—> T2 = 600 - 300 = 300 ,	T2+T3= 300+600 = 900
 * T1+T2	== T1H (580 … 1000 nS	—> T3 = 600 ,				T1+T2 =370+600 = 900
 * T3		== T1L (580 … 1000 nS)
 *
 * RES =  > 280 uS??
 *
 */`

But nothing is happeing... Are the variables T1, T2 and T3 correct?

zackees, 2022-07-12:
I don't know if your timings are correct, you could debug further by directly subclassing `CLEDController`

kokospalme, 2022-07-12:
I don't see, where I can change the reset timing. At WS2812B it's >50uS. The WS2818B should have >280uS. Is there a chance of also changing the reset timing?

smurphyUSI, 2022-07-13:
The WS2818 is an improved variation of the WS2811, so the timings should be working.    Make sure they are wired correctly, because it's not the usual affair as 4-wire units.

Unfortunately, between revisions of different 'WS28XX' units, and clones, the timings are sometimes incorrectly documented.
I would suggest trying again, running as WS2811 and then WS2812.
Regardless of the variation, the timings should be one of those two.

kokospalme, 2022-07-13:
I could change the timings, even the reset-timing like this:
```template <uint8_t DATA_PIN, EOrder RGB_ORDER = RGB>
class WS2818BController : public ClocklessController<DATA_PIN, C_NS(250), C_NS(380), C_NS(600), RGB_ORDER, 4, false, C_NS(1230)> {};```

kokospalme, 2022-07-13:
Alright, thanks for the quick answers! Maybe it's the optocpoupler's signal that's not good enough or something... I'll post if that caused the error, in case others facing the same problems

zackees, 2024-10-12:
Following up on this, you can use this python script to calculate T1,T2,T3 from the T0H/T0L, T1H, T1L timings from the data sheet. If this fixes you then our timings are wrong.

If you get vastly different numbers than us then our timings are wrong:

//  print("Enter the values of T0H, T0L, T1H, T1L, in nanoseconds: ")
//  T0H = int(input("  T0H: "))
//  T0L = int(input("  T0L: "))
//  T1H = int(input("  T1H: "))
//  T1L = int(input("  T1L: "))
//  
//  duration = max(T0H + T0L, T1H + T1L)
//  
//  print("The max duration of the signal is: ", duration)
//  
//  T1 = T0H
//  T2 = T1H
//  T3 = duration - T0H - T0L
//  
//  print("T1: ", T1)
//  print("T2: ", T2)
//  print("T3: ", T3)

closed 2026-08-20 by zackees