FastLED/FastLED #1408: FastLED giving white output when using HSV

forbenefitthehuman, 2022-07-13:
I need to do a lot of fading and brightening in a project, so I cut over to FastLED so I could use HSV values, but it only gives a white output ( assuming equal  RGB ) , the  V variable seems to work ( no output when set to zero ) but the other two seem to have no effect.

```
#include <FastLED.h>
CRGB strip[27];

void setup() {
  FastLED.addLeds<NEOPIXEL, 23 >(strip,27);
}

void loop() {
 
  // strip[0] = CRGB::Red; 
  strip[0] = CHSV(20, 128, 255);
  FastLED.show();
  delay(30);
}
```

The line commented out works fine
The HSV line gives white.

Using ESP32 and WS2812 strip, worked fine under Adafruit Neopixel Library

smurphyUSI, 2022-07-13:
All white is often a wiring order problem, or the saturation value set very low.
Setting the both the value and saturation to 255 is a good way to test.

Try this test:

#include <FastLED.h>
CRGB strip[27];

void setup() {
  delay(3000);   // Some boards need a power-up delay (not sure about the ESP32)
  
  FastLED.addLeds<WS2812,23,   GRB   >(strip, 27).setCorrection(TypicalLEDStrip);   // Could also be RGB depending on exact strip

  FastLED.setBrightness(255);  // This is the MAX brightness your setup can handle.  It's perhaps the easiest way to control total power consumption, because if you set this to any value other than 255, it scales down your 'V' values proportionatly

  fill_solid(strip, 27, CHSV(0,255,255));    // This should fill solid red.   If it's green instead, then change the GRB to RGB
  FastLED.show();
}

void loop() {
}

For a more extensive list of RGB order tests, try the example for calibration:
https://github.com/FastLED/FastLED/blob/master/examples/RGBCalibrate/RGBCalibrate.ino

forbenefitthehuman, 2022-07-14:
Thanks for your reply.
Your test code worked, and I've updated my project code, it was the saturation value that was causing the issue.

I didn't expect the saturation to drop the colour so sharply, even at 200 the colour is barely visible.
Now I get my "oil lamps" to turn off without weird flashes of blue or green, or complicated equations.

Thanks again.

forbenefitthehuman, 2022-07-14:
The esp32 does not require the delay running the test or my code