FastLED/FastLED #1410: fastspi.h uses wrong variable for spi1

anthonyrubalcaba, 2022-07-17:
/FastLED/src/fastspi.h >> line 91,92: 

template<uint32_t SPI_SPEED>
class SPIOutput<SPI1_DATA, **SPI_CLOCK**, SPI_SPEED> : public Teensy4HardwareSPIOutput<SPI1_DATA, SPI1_CLOCK, SPI_SPEED, SPI1, 1> {};

The SPI_CLOCK should be SPI1_CLOCK

Corrected:

template<uint32_t SPI_SPEED>
class SPIOutput<SPI1_DATA, **SPI1_CLOCK**, SPI_SPEED> : public Teensy4HardwareSPIOutput<SPI1_DATA, SPI1_CLOCK, SPI_SPEED, SPI1, 1> {};

This bug was discovered by user [AntiDysentery](https://forum.pjrc.com/members/86791-AntiDysentery)
in this post: 
https://forum.pjrc.com/threads/70671-Teensy-4-1-with-FastLed-No-Hardware-SPI

zackees, 2022-07-18:
Do you know how to create a pull request for this bug fix?

Fork -> Create branch -> do changes -> git commit / push and follow the instructions that appear. Your git should now contain a pull request. Please mention this bug URL and github will link them automatically.

ssilverman, 2022-07-22:
I made a pull request a couple of days ago that fixes this: #1413.

zackees, 2022-07-22:
I'd ok it if I had ownership, but I don't.

PR #1413 (2022-11-26): Fix Teensy 4 SPI1_CLOCK
Ref: https://forum.pjrc.com/threads/70671-Teensy-4-1-with-FastLed-No-Hardware-SPI
Companion PR: https://github.com/PaulStoffregen/FastLED/pull/1

Fixes #1410
closed 2022-11-26 by kriegsman