FastLED/FastLED #1412: Using FastLED in a library

slowgergo, 2022-07-18:
### Hello guys!

I'm once again having trouble but this this time it most likely is my programming incompetence. I have a bigger project where I use FastLED to light up some WS2812b leds. The code was getting too long so I wanted to break it up into multiple files.

**What I want to achieve:** To have a library that has functions for lighting up leds which can be accessed by other separate libraries 
(eg: main code calls button handler lib and that lib calls led handler to light up some leds
eg2: main calls animation lib and that calls led handler)

**The problem:** I have written a lib that can light up leds perfectly but when I call it in main AND in button handler the `CRGB leds[54];` and `FastLED.addLeds<WS2812B, 13, GRB>(leds, 54);` obviously gets screwed up because of the second instance of the library being called. 

I have googled everything how to pass the pointer from the `CRGB leds[54];` to somehow having it global I couldn't find a solution. Last promising thing was a singleton class but I have NO idea how to implement that.

My testing code where only the button library's FastLED calls work because it was constructed last:
<details>
  <summary>main.cpp</summary>

```
#include <Arduino.h>
#include <alib.h>
#include <button.h>
#include <FastLED.h>

LedHandler leds;
Button button;

void setup() 
{
  Serial.begin(74880);
  Serial.println("Start");
  leds.leds[4] = CHSV(0,255,255);
  FastLED.show();

  delay(2500);

  Serial.println("Btn lib");
  button.buttonBlink();

  Serial.println("Normal lib");
  leds.blink();
}

void loop() 
{
  leds.rainbow();
  delay(50);
  
}
```
</details>

<details>
  <summary>alib.h</summary>

```

#ifndef ALIB_H
#define ALIB_H
#include <Arduino.h>
#include <FastLED.h>

class LedHandler
{
  public:
    LedHandler();
    void rainbow();
    void blink();
    CRGB leds[54];
    
  private:
    int color;

};

#endif
```
</details>

<details>
  <summary>alib.cpp</summary>

  ```
#include <alib.h>

LedHandler::LedHandler()
{
  FastLED.addLeds<WS2812B, 13, GRB>(leds, 54);
}

void LedHandler::rainbow()
{
  fill_solid(leds, 54, CHSV(color,255,200));
  delay(1);
  FastLED.show();
  color++;
  if (color == 256)
  {
    color = 1;
  }
}
void LedHandler::blink()
{
  leds[8] = CHSV(0,255,255);
  FastLED.show();
  delay(250);
  leds[8] = CHSV(0,255,0);
  FastLED.show();
  delay(250);
}

  ```
  
</details>

<details>
  <summary>button.h</summary>

  ```
#ifndef BUTTON_H
#define BUTTON_H
#include <alib.h>
#include <Arduino.h>

class Button
{
  public:
    Button();
    void buttonBlink();
    LedHandler buttonLeds;
  private:

};

#endif
  ```
  
</details>

<details>
  <summary>button.cpp</summary>

  ```
#include <button.h>

Button::Button()
{

}
void Button::buttonBlink()
{
  buttonLeds.blink();
}
  ```
  
</details>

(alib.cpp is correctly indented on my side)

What should I google that could help me? :D 
Or how should I structure my code differently so I don't run into this problem? Maybe get rid of the separate classes thing so they are just separate .h and .cpp files?

Not having `#include <alib.h>` in `main.cpp` is not a solution because in the end I would hopefully have approx 4 files calling functions from `alib.h`.

Using PlatformIO in VS Code
Esp8266 @ 2.6.3  
FastLED @ 3.4.0 
(having the newest versions I had the register warnings and some flickering with using wifi before)

Thanks for any help and advice!

zackees, 2022-07-18:
You are designing your program like a C++ app. This will give you lots and lots of problems. I've done this myself until I learned to submit to the C style and the result is much simpler code.

The problem here is that you have objects which are being statically initialized. Static initialization of complex types which call into arduino's libraries in their constructors will break. If they don't break on your current hardware, they may break on an update or a different board. Assume that any static initializer calling into arduino results in undefined behavior.

If you want to keep your C++ style object usage then my suggestion is that you create pointers which get set to instances in the setup() function.

if you want to do it more idiomatic way do something like this:

```
// button.h
void button_setup(int pin); // called from setup
int button_update(...); // called from loop()
```

This will ensure everything get's initialized at the right time.

zackees, 2022-07-18:
I want to add that plain-old-data types are fine to statically initialize. For example the pixel types (e.g. blocks of CRGB and CHSV types) in fastled should all be statically initialized and then passed in during the initialization phase.

slowgergo, 2022-07-19:
### Thank you both for fast replies!

> I learned to submit to the C style

How would I go about that with the current scenario? Could you point me to some resources?

So to recap get rid of `FastLED.addLeds<WS2812B, 13, GRB>(leds, 54);` from the constructor and have it in a function which will be called form `main.ccp` setup. Done.

Next I should get a `CRGB` array in main and pass the pointer from it to the lib in the setup right? 
I have this in main.cpp:

```
.....
LedHandler leds;
CRGB buf[54];

void setup()
{
leds.setup(CRGB *buf);
.....

```

What do I put in `alib.h` for the function arguments of `leds.setup()`? I kinda started shooting in the dark trying and got some weird errors.... I have a little understanding about pointers with C but it clearly isn't enough.

> I also don't recommend trying to wrap FastLED itself, if you just want controllers.
Each call to addLeds returns a controller type, which you can save the pointers to.

You mean have `CRGB leds[54];` and `FastLED.addLeds` in main and pass the pointer from `FastLED.addLeds` to some lib I need it in?
Can you please give me an example? 

Thank you both for your help, and sorry for being off topic!

zackees, 2022-07-19:
This is usually my pattern:

defs.h:
```
#define NUM_LEDS 54
...
```

gfx.cpp:
```
#include "defs.h"
#include "FastLED.h"
CRGB gLeds[NUM_LEDS] = {0};  // zero initialize array.

void gfx_setup() {
  FastLED.addLeds(...);
}

void gfx_draw() {
  ...
}
```

gfx.h:
```
void gfx_setup();
void gfx_draw();
```

slowgergo, 2022-07-19:
### Huge Thanks now it works just as I intended!

Out of curiosity what is this phenomenon called? I mean they are .cpp files but without `#ifndef` or  `#pragme once` and the other class. Does this just get linked together as a big file by the compiler/linker and its easier for humans to read?
I don't know proper terminology never really learned coding in school :(

### Thank you again for the help! Have a great day!

closed 2022-07-19 by slowgergo

zackees, 2022-07-19:
Yes, the arduino compiler toolchain looks for all cpp files and compiles them in the root directory and also the src/** directory. Then all the compiled object files are linked together in the executable and dead code is aggressively eliminated.

Glad it all worked out!