FastLED/FastLED #1414: Set Configuration from spiffs

trinityworld, 2022-07-22:
Data pin,
LED type , 
Color order,
Brightness from

read json object from spiffs like :

`{`
 ` "Data pin": 4,`
`  "LED type": "WS2812",`
`  "Color order": "RGB",`
`  "Brightness ": 84`
`}`
How to assign values accordingly?
I tried with templet & class but gives me constant declaration error

my concern is, how to compare and assign values from json dynamically ?

`FastLED.addLeds<WS2812, DATA_PIN, RGB>(leds, NUM_LEDS);`

zackees, 2022-07-22:
DATA pins for FastLED are compile time constants:

https://github.com/FastLED/FastLED/issues/282

So my recommendation is to use something like adafruits neopixel library (since you are using WS2812) or subclass your own pixel type and pass that into one of the overloaded functions of .addLeds<>()

samguyer, 2022-11-25:
@trinityworld Is the hardware configuration really dynamic? Like, can you actually have a situation where a new kind of LED strip is plugged in after the device is running?

Gidrogen, 2023-10-24:
For five years I have been praying to the gods for these improvements in my favorite library.🙏
During this time, I have noticed a large number of similar requests for improvements to Fastled for more powerful microcontrollers.

Should we expect it to be available in the future?🙂