FastLED/FastLED #1415: FastLED + wifi = flickering with esp8266

TypingPenguin44, 2022-07-25:
Hey guys!

I have seen some posts and read some threads about this issue and I wanted to ask once and for all.
I understand this issue is mainly because wifi interrupts causing fastled to flicker the leds.

My specific problem: 
I have ws2812b-mini leds with an esp01f (esp8285) on a custom pcb and when wifi is initialized in the code it starts flickering. Not 1 led the whole 54 leds light up for example green for a split second then turning back to red.
I tried with an esp12f (esp8266) with the same behavior.

### **The twist:** I connected both (esp 8266 and the esp 8285) to a ws2812b led strip (which is cut up led by led and is in a enclosure) and it didn't flicker at all. 

Same code same powersupply (320w lab psu) same everything. I looked up the datasheets of the leds and the timing specs are the same as well. I connected with my phone no flickering, debug showing a device connected etc. It worked as it should without any hiccups.

I even tried with the esp8285's led control pin wired to both the normal ws2812b's data in and the ws2812b-mini's data in together and the normal ones didn't flicker and the mini ones did.

The pcb's grounding is good enough it draws the same amount of amps/led as the ones in the enclosure...

I tried the usual `#define FASTLED_ALLOW_INTERRUPTS 0` and the retry count thing and it didn't help.

ONE major thing is that when I started with this project 2021 march/april this flickering didn't happen at all with the same hardware. Sadly I can't reproduce this with the same old libraries and I am at a loss why it does this.....

Tomorrow I will try with i2s and another library but making that work will be a hackjob wiring wise on the pcb.....

Thanks for any help or insight, have a great day!

lameslime, 2022-09-17:
I'm having the same issue, got worse when I replaced legacy Blynk ecosystem with mqtt and homeassistant. Kinda fixed it by only using `FastLED.show()` when it recieves changes, but that also means still flickering while updating leds and no animations...

samguyer, 2022-11-25:
@TypingPenguin44 Did you make any progress on this problem? The most interesting behavior is that the same pin can correctly drive the regular WS2812s, but not the mini WS2812s. One possibility is that the voltage drop is a little different because the LEDs are different sized, so maybe the signal from the ESP is properly read by the bigger WS32812s, but not the little ones. Have you tried adding a level shifter?

TypingPenguin44, 2022-11-27:
I haven't made any progress with this library. I took apart the pcbs cut traces and stuff and started using NeoPixelBus and finished the project. (Kinda got fed up with all these libs not working as intended without major problems...) Now I cant test the behavior again if something changed with this lib since.

A level shifter wouldn't hurt that's for sure but as I said I have a custom pcb and I have not implemented one because when I was testing the proof of concept I had no problems with the leds flashing.  

For anybody in the future: I decided on NeoPixelBus because of the webserver running beside the leds so the interrupt based FastLED wasn't a good choice from the beginning but I had to modify NeoPixelBus to make it work so no lib is without problems.

I'm closing this thread because I can't test the behavior.

closed 2022-11-27 by TypingPenguin44

samguyer, 2022-11-28:
@TypingPenguin44 Sorry things didn't work out for you with FastLED! Good luck with your project.

Jaibpi96, 2023-12-12:
¡SOLUCIONADO! El problema esta en el core de ESP8266 para arduino. Haciendo downgrade a version 2.7.4 vuelve a funcionar como antes. Voy a probar versiones mas recientes pero 2.7.4 funciona con FastLed sin problemas.

Para quien no sepa cambiar la versión del core : Herramientas/Placa/Gestor de tarjetas y ahi buscamos esp8266 y cambiamos a versión mas antigua.

SEÑORES DE ESPRESSIF ME HABEIS VUELTO LOCO DURANTE DIAS...

suromark, 2023-12-27:
I can confirm that reverting to the 2.7.4 core has fixed the strobing/flickering issue with FastLED and ESP8266 for me too. I built an 8x8 LED matrix decoration years ago and when I changed some code and re-flashed it yesterday, the display went off the rails completely. I managed to somewhat alleviate the issue by disabling all MQTT code, but since I've reverted the core it's now back to stable behavior even with MQTT + Webserver enabled.

I'm using Platformio, and the ini string I've used starts as follows:

[env:d1_mini]
platform = espressif8266@2.6.3
board = d1_mini
framework = arduino
lib_deps = 
    fastled/FastLED @ ^3.6.0
    knolleary/PubSubClient @ ^2.8

mkazimoto, 2024-06-21:
I went back to version 2.7.4 in the Arduino IDE and it worked! 
[image]