FastLED/FastLED #1416: Need help figuring out what to do with an art piece by focalintent

raindrift, 2022-07-27:
Hi, FastLED maintainers. I'm Ian Baker. I was a shopmate of Dan's for many years, at the American Steel warehouse in West Oakland, CA.

We are currently in the process of being evicted from the space by the [real estate speculators](https://brokeassstuart.com/2022/02/21/american-steel-artists-speak-about-billionaires-evicting-them/) who purchased it last year. While moving out, we found a piece of Dan's art. We think it's the piece called "Liquid Eye", a circular wall-hanging from around 2014. It is carefully packed into a custom wood crate, which is about 8'x8', and 4 inches thick. [Here's a post about it.](https://forum.makerforums.info/t/doing-some-major-code-overhauling-on-liquid-eye/65037)

I know this is really unusual content for a GitHub issue. We will also try to contact his family, but it seems like maybe the FastLED community might have a better idea of what is the best thing to do with it. Moving it to our new shop and continuing to store it in a crate seems like the wrong choice. Would one of you who knew him be interested in taking custody of it? We have to move everything out by July 31.

smurphyUSI, 2022-07-27:
Sorry to hear about the evication.

I'll ping @kriegsman here, as it's possible that he or MC March may at least know somebody who will take responsibility for that piece.

zackees, 2022-07-27:
I love that piece. I will host it and give it a good home and hang it up with my other LED art pieces in my house.

kriegsman, 2022-07-27:
First, thank you for reaching out, and for being custodian of that piece of his art for all this time.  Second, that really sucks about being evicted!  Jerks.

On the one hand, Dan's family isn't in the Bay Area, but on the other hand, Dan has a big loving extended 'family' that is all over the place -- including there.  I'll try to see who I can contact, but honestly, I think that anyone who can give this piece a good and loving home, hopefully where it can be seen by more people, would be great to work with.    

For now, let's use this Github issue & thread to connect about this.

raindrift, 2022-07-27:
@kriegsman, Corey got in touch with me after you pointed this thread out to her. It sounds like she has space at home to safely store it, and is in touch with Dan's family and can work with them to figure out the right thing to do with it on a somewhat less urgent timeline. So I think the current plan is to hand it off to her. Thank you so much for your help!

closed 2022-07-27 by raindrift

kriegsman, 2022-07-27:
I'm glad that it's all going to be taken care of, and thank you again for getting in touch and helping Liquid Eye have a good continuation.

I'll just putting some Liquid Eye links here to share for everyone, too:
- Work-in-progress video from Dan: https://www.youtube.com/watch?v=Vkcm8dmTnq4
- Two videos of Liquid Eye at Burning Man: https://www.youtube.com/watch?v=C4d6ePNmQ64 https://www.youtube.com/watch?v=WO6LiyrEQC0
- FB page for Liquid Eye (and other works by Dan) https://www.facebook.com/LiquidEyeArtProject

Liquid Eye was one of Dan's first 'big' pieces, and one of his first that sandwiched LED strips between smoked plastic sheets.  He later made "Tree of Life" and "Flow" using the same technique, as well as the very large "Locus Lux Liquida" https://www.flickr.com/photos/focalintent/albums/72157634554418457  It was also the first piece where Dan did XY mapping of irregularly arranged pixels, and then used that mapping in the code, to great effect.

After taking Liquid Eye to Burning Man, and putting it out in the middle of the playa for people to find at night, he told a story about it.  He said that he was going out to it each night to check on it, and make sure it was still working okay.  And one night he went out to check on it, and found a circle of people kneeling around Liquid Eye, and _worshipping_ it.  He said, "You know, that some's pretty damn good feedback!"