FastLED/FastLED #1417: Is FastLED abandoned?

sblantipodi, 2022-08-04:
I'm using FastLED 3.5.0 on ESP8266 and ESP32.

As soon as I include FastLED I have dozens of warnings, I see no support for newer ESP32-C series and ESP32-S series...

Is there someone working on a new release that will fix this problems soon? 

Thanks
Davide

zackees, 2022-08-05:
The main contributor Dan died in a tragic accident.

https://www.reddit.com/r/FastLED/comments/czd31f/dan_garcia_passed_away/

Is FastLED abandoned? No. But changes are a lot slower now. You might have better luck using a different library or adding support yourself.

samguyer, 2022-11-25:
@sblantipodi We are trying to get back to active development! There should be better support now for some of the ESP32 variants. To some degree, we rely on contributors for this kind of thing -- there are just too many boards for us to keep track of support for all of them.

closed 2022-11-25 by samguyer