FastLED/FastLED #1419: Mystery PR checks fail: "Error: Unknown development platform 'atmelavr'"

ssilverman, 2022-08-06:
My last two PRs can't pass checks because of an "Error: Unknown development platform 'atmelavr'" error.

https://github.com/FastLED/FastLED/pull/1413
https://github.com/FastLED/FastLED/pull/1418

kriegsman, 2022-11-25:
In some places online I'm seeing 'atmelmegaavr' used instead of 'atmelavr'; looking into it, but any additional thoughts or input would be quite welcome, as I'm definitely not a CI expert here!

kriegsman, 2022-11-26:
OK, this looks fixed by https://github.com/FastLED/FastLED/pull/1428, with thanks to @dmadison.

In order to process the PRs that were submitted in the interim, I'll probably have to Close and Re-open them to make the updated tests re-run.

closed 2022-11-26 by kriegsman