FastLED/FastLED #1422: I2S support for the ESP32S2

davorvr, 2022-08-10:
Hi, I would like to use the I2S mode on the ESP32S2 variant, but the code currently doesn't compile. So far, two errors have occurred.

1. The ESP32S2 only has one I2S peripheral, so any references to `&I2S1` are invalid and the code refuses to compile.
2. The `i2s` struct member `clka_en` doesn't exist anymore, and it's been replaced with `clk_sel` in ESP32S2's respective `i2s_struct.h` file due to a restructuring in the `I2S_CLKM_CONF_REG` register: `I2S_CLKA_ENA` bit is replaced with `I2S_CLK_SEL` bits on the ESP32S2, and can be set to `0` or `3` for no clock, `1` for the configurable analog PLL clock and `2` for the 160 MHz clock. Setting `clk_sel = 2` should correspond to the previously used `clka_en = 0` for the original ESP32 variant in order to use the 160 MHz clock instead of the analog PLL.

I have modified the code accordingly ([which can be found here](https://github.com/davorvr/FastLED/tree/esp32s2)), and, although the blink example now compiles successfully, there is no LED output, but there are brief flashes on occasion. I am unable to probe the output with an oscilloscope currently, but might be able to in a few days.

zackees, 2022-08-10:
The missing LED output might be due to incorrect hardware spi implementation on the ESP32 chipset families. You can see if defining `FASTLED_FORCE_SOFTWARE_SPI` fixes the issue for you, like so:

```C++
#define FASTLED_FORCE_SOFTWARE_SPI
#include <FastLED.h>
```

samguyer, 2022-11-25:
@davorvr Thanks for the report (and the modified branch). Did you manage to get any further with it? I might try to update the I2S driver to use more portable APIs from the ESP-IDF.

zeezooX, 2023-04-20:
Have you managed to make it work?

zackees, 2024-10-12:
People have made it work and @shlomozippel gave me some code that's in platforms/esp/32/experimental that probably works. But it's only the C2 that will be fixed so it's not a high priority for me right now as i'm working on the 4.0 release.

shlomozippel, 2024-10-12:
The code I shared is for the LCD peripheral, which was separated from the I2S peripheral in ESP32-S3. It won't work for the S2 unfortunately

zackees, 2024-10-13:
Yves has a special driver for i2s.

It has an adafruit like strip interface: