FastLED/FastLED #1423: Fastled Not compiling in mini core atmega88

anupam-dubey, 2022-08-12:
Usaing minicore for atmega88

In file included from C:\Users\MASTER\Documents\Arduino\libraries\FastLED\src/FastLED.h:48:0,
                 from C:\Users\MASTER\Desktop\mega8_150_2019_vishal\mega8_150_2019_vishal.ino:2:
C:\Users\MASTER\Documents\Arduino\libraries\FastLED\src/fastpin.h: In instantiation of 'class FastPin<13>':
C:\Users\MASTER\Documents\Arduino\libraries\FastLED\src/platforms/avr/clockless_trinket.h:100:49:   required from 'class ClocklessController<13, 6, 8, 6, (EOrder)10, 0, false, 10>'
C:\Users\MASTER\Documents\Arduino\libraries\FastLED\src/chipsets.h:477:7:   required from 'class WS2811Controller800Khz<13, (EOrder)10>'
C:\Users\MASTER\Documents\Arduino\libraries\FastLED\src/FastLED.h:111:52:   required from 'class WS2811<13, (EOrder)10>'
C:\Users\MASTER\Documents\Arduino\libraries\FastLED\src/FastLED.h:302:39:   required from 'static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with CHIPSET = WS2811; unsigned char DATA_PIN = 13; EOrder RGB_ORDER = (EOrder)10]'
C:\Users\MASTER\Desktop\mega8_150_2019_vishal\mega8_150_2019_vishal.ino:51:67:   required from here
C:\Users\MASTER\Documents\Arduino\libraries\FastLED\src/fastpin.h:210:2: error: static assertion failed: Invalid pin specified
  static_assert(validpin(), "Invalid pin specified");
  ^~~~~~~~~~~~~
Using library FastLED at version 3.4.0 in folder: C:\Users\MASTER\Documents\Arduino\libraries\FastLED 
exit status 1

anupam-dubey, 2022-08-12:
Issue resolved by adding || defined(__AVR_ATmega88P__) in plateforms/avr/fastpins_avr.h 
line no.247

closed 2022-08-12 by anupam-dubey

un-clouded, 2023-12-14:
In case it is helpful, this is now line 271 and the appendage should be:

     ... || defined(__AVR_ATmega48__)

(or whichever part you're using in MiniCore, such as ATmega48A, etc.