FastLED/FastLED #1429: `addLeds<>()` and inherited class

Master811129, 2022-09-04:
Hello, In my program i made a class and inherited `CRGB` and add some more functionallity to the class for my LED project.
``` cpp
class LEDHelper:public CRGB{
private:
//int a = 0;
//private variables
public:
//somthing
};
```
the problem is that i can use `FastLED.addLeds<>();` it compiles but leds show wrong colors... is there any way to use addLeds for inherited classes? I think someshow I should give the size of my class to the the FastLED library.

smurphyUSI, 2022-09-05:
This won't work.   Hard stop.

.addLeds<>() , as suggested by the '<>' is a template.
This means that addLeds compiles things like the chipset type, RBG order and pinout directly into static code.

You won't be able to do any run-time checking of your LEDHelper class from withing any functions which use class data passed into addLeds.

I don't know what your goal is, but you could potentially wrap a CRGB object inside a helper class.
You could then do operations with your helper, which pass the changes down to CRGB.

zackees, 2022-09-06:
Yeah, don't subclass CRGB and expect anything to work. FastLED expects a certain byte layout and when you add extra members it messes that up.

JakeSays, 2022-09-07:
@Master811129  Deriving from CRGB works fine as long as the memory layout doesn't change. This means that you can't add data members or virtual methods to your derived class. If you need either of those in your derived then you'll need to use composition instead (which is usually better anyway).

zackees, 2022-09-08:
Correct, but if you are subclassing CRGB then you are probably approaching the problem in the wrong way and should instead use a free function. A beginner may sublcass CRGB and then read that it's always best to add a virtual destructor and then boom, they have a vtable and the sublcass no longer fits in `sizeof(CRGB)` bytes.

Master811129, 2022-09-08:
Thank you all for helping me. Probably defining CRGB in ino file and then give a pointer to my class would be a better idea.

closed 2022-09-08 by Master811129