FastLED/FastLED #1430: Not Compiling for ESP32

av4625, 2022-09-08:
I get the following error when compiling for ESP32:
```
/.../Arduino/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp: In static member function 'static void ESP32RMTController::init(gpio_num_t)':
/.../Arduino/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:111:15: error: variable 'espErr' set but not used [-Werror=unused-but-set-variable]
     esp_err_t espErr = ESP_OK;
               ^~~~~~
/.../Arduino/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp: In member function 'void ESP32RMTController::startOnChannel(int)':
/.../Arduino/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:239:15: error: variable 'espErr' set but not used [-Werror=unused-but-set-variable]
     esp_err_t espErr = ESP_OK;
```

Its strange as it is definitely used in the first function, the `ifdef`s might be confusing it in the second and it is only used in an `else`.

av4625, 2022-09-13:
This commit does fix the issue but obviously not the best solution: https://github.com/FastLED/FastLED/compare/master...av4625:FastLED:master

emilfihlman, 2022-11-27:
Getting this also.

av4625, 2023-08-28:
I understand those flags. But they are the default when compiling with Arduino for ESP32. You would have to dig around in arduino settings to change them and it would change them for everything, not just this one project or library

av4625, 2023-08-30:
Haha! You do make some good points like if a a new version of GCC started throwing a new warning and then lots of things stopped building. 

You mentioned copying platform.txt to platform.ini. I think this is a platform io thing? I haven't used platform io before so could be wrong. 

I am not using platform io here and just building with Arduino IDE. It must use the settings in platform.txt some how?

Akil0101, 2024-03-13:
i am trying to fix this error please help

In file included from c:\Users\king\Documents\Arduino\libraries\FastLED\src/FastLED.h:75,
                 from C:\Users\king\AppData\Local\Temp\.arduinoIDE-unsaved2024213-10864-1xvc100.uoc\DemoReel100\DemoReel100.ino:5:
c:\Users\king\Documents\Arduino\libraries\FastLED\src/fastspi.h:157:23: note: '#pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output'
  157 | #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\king\Documents\Arduino\libraries\FastLED\src\noise.cpp: In function 'void fill_raw_2dnoise8(uint8_t*, int, int, uint8_t, q44, fract8, int, uint16_t, int, uint16_t, int, uint16_t)':
c:\Users\king\Documents\Arduino\libraries\FastLED\src\noise.cpp:623:100: error: ambiguous overload for 'operator*' (operand types are 'q44' {aka 'q<unsigned char, 4, 4>'} and 'int')
  623 |     fill_raw_2dnoise8(pData, width, height, octaves-1, freq44, amplitude, skip+1, x*freq44, freq44 * scalex, y*freq44, freq44 * scaley, time);
      |                                                                                             ~~~~~~ ^ ~~~~~~
      |                                                                                             |        |
      |                                                                                             |        int
      |                                                                                             q44 {aka q<unsigned char, 4, 4>}
In file included from c:\Users\king\Documents\Arduino\libraries\FastLED\src\pixeltypes.h:7,
                 from c:\Users\king\Documents\Arduino\libraries\FastLED\src\controller.h:9,
                 from c:\Users\king\Documents\Arduino\libraries\FastLED\src\FastLED.h:57,
                 from c:\Users\king\Documents\Arduino\libraries\FastLED\src\noise.cpp:6:
c:\Users\king\Documents\Arduino\libraries\FastLED\src\lib8tion.h:909:14: note: candidate: 'uint32_t q<T, F, I>::operator*(uint32_t) [with T = unsigned char; int F = 4; int I = 4; uint32_t = long unsigned int]'
  909 |     uint32_t operator*(uint32_t v) { return (v*i) + ((v*f)>>F); }
      |              ^~~~~~~~
c:\Users\king\Documents\Arduino\libraries\FastLED\src\lib8tion.h:911:14: note: candidate: 'uint16_t q<T, F, I>::operator*(uint16_t) [with T = unsigned char; int F = 4; int I = 4; uint16_t = short unsigned int]'
  911 |     uint16_t operator*(uint16_t v) { return (v*i) + ((v*f)>>F); }
      |              ^~~~~~~~
c:\Users\king\Documents\Arduino\libraries\FastLED\src\lib8tion.h:913:13: note: candidate: 'int32_t q<T, F, I>::operator*(int32_t) [with T = unsigned char; int F = 4; int I = 4; int32_t = long int]'
  913 |     int32_t operator*(int32_t v) { return (v*i) + ((v*f)>>F); }
      |             ^~~~~~~~
c:\Users\king\Documents\Arduino\libraries\FastLED\src\lib8tion.h:915:13: note: candidate: 'int16_t q<T, F, I>::operator*(int16_t) [with T = unsigned char; int F = 4; int I = 4; int16_t = short int]'
  915 |     int16_t operator*(int16_t v) { return (v*i) + ((v*f)>>F); }
      |             ^~~~~~~~
c:\Users\king\Documents\Arduino\libraries\FastLED\src\noise.cpp:623:127: error: ambiguous overload for 'operator*' (operand types are 'q44' {aka 'q<unsigned char, 4, 4>'} and 'int')
  623 |     fill_raw_2dnoise8(pData, width, height, octaves-1, freq44, amplitude, skip+1, x*freq44, freq44 * scalex, y*freq44, freq44 * scaley, time);
      |                                                                                                                        ~~~~~~ ^ ~~~~~~
      |                                                                                                                        |        |
      |                                                                                                                        |        int
      |                                                                                                                        q44 {aka q<unsigned char, 4, 4>}
c:\Users\king\Documents\Arduino\libraries\FastLED\src\lib8tion.h:909:14: note: candidate: 'uint32_t q<T, F, I>::operator*(uint32_t) [with T = unsigned char; int F = 4; int I = 4; uint32_t = long unsigned int]'
  909 |     uint32_t operator*(uint32_t v) { return (v*i) + ((v*f)>>F); }
      |              ^~~~~~~~
c:\Users\king\Documents\Arduino\libraries\FastLED\src\lib8tion.h:911:14: note: candidate: 'uint16_t q<T, F, I>::operator*(uint16_t) [with T = unsigned char; int F = 4; int I = 4; uint16_t = short unsigned int]'
  911 |     uint16_t operator*(uint16_t v) { return (v*i) + ((v*f)>>F); }
      |              ^~~~~~~~
c:\Users\king\Documents\Arduino\libraries\FastLED\src\lib8tion.h:913:13: note: candidate: 'int32_t q<T, F, I>::operator*(int32_t) [with T = unsigned char; int F = 4; int I = 4; int32_t = long int]'
  913 |     int32_t operator*(int32_t v) { return (v*i) + ((v*f)>>F); }
      |             ^~~~~~~~
c:\Users\king\Documents\Arduino\libraries\FastLED\src\lib8tion.h:915:13: note: candidate: 'int16_t q<T, F, I>::operator*(int16_t) [with T = unsigned char; int F = 4; int I = 4; int16_t = short int]'
  915 |     int16_t operator*(int16_t v) { return (v*i) + ((v*f)>>F); }
      |             ^~~~~~~~
[... cut ...]
c:\Users\king\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp: In static member function 'static void ESP32RMTController::doneOnChannel(rmt_channel_t, void*)':
c:\Users\king\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:347:5: error: 'gpio_matrix_out' was not declared in this scope; did you mean 'gpio_iomux_out'?
  347 |     gpio_matrix_out(pController->mPin, 0x100, 0, 0);
      |     ^~~~~~~~~~~~~~~
      |     gpio_iomux_out

exit status 1

Compilation error: exit status 1

zackees, 2024-09-06:
This has been fixed for a while, closing.

closed 2024-09-06 by zackees