FastLED/FastLED #1431: support for WS2813B-B

arifainchtein, 2022-09-11:
Hello,
I have been using the WS2812B but because of cost, i was hoping to use the WS2813B-B.  Is the WS2813B-B cipatible withthe FastLed library? If i  have code for the WS2812 working is there any configuration changes to use the ws2813?

thanks

smurphyUSI, 2022-09-12:
Have you attempted to use the WS2813 setup yet?
FastLED does support WS2813 variations, but sometimes the RGB order is swapped.

Try running this in your code:

#include <FastLED.h>
#define NUM_LEDS 100
#define DATA_PIN 6
// - Change the above to your actual number of LEDs and the data pin of your controller

CRGB leds[NUM_LEDS];

void setup()
{ 
  FastLED.addLeds<WS2813, DATA_PIN>(leds, NUM_LEDS);
}

If the it works, but the color is wrong, try chaning your RGB order.
There is an example script here to check the different options.
You just have to uncomment and run one-at-at time for each variation you think maybe correct:
https://github.com/FastLED/FastLED/tree/master/examples/RGBCalibrate

arifainchtein, 2022-09-12:
Hello,
thanks fr the answer.  I have not yet used the ws2813 but i am sending some pcbs to manufacture and wanted to replace the ws2812-4020 with ws2813 because of cost.  Knowing that the FastLed library is compatible is great because i really like this library.
Will send the pcbto manufacture and will report later.
thanks

marmilicious, 2023-12-08:
@briandilley Have you tried it? My understanding is the only difference between the WS2813 A,B,C,D variants is brightness level, so should work.

https://www.ledstripstudio.com/wiki/ws2813/

briandilley, 2023-12-08:
Yes - it works.

marmilicious, 2023-12-08:
@samguyer Unless @briandilley has anything else at the moment, this issue seems like it can be closed.

closed 2024-09-19 by zackees