FastLED/FastLED #1435: LavaColors_p is missing a value

DavidSchinazi, 2022-09-21:
[LavaColors_p](https://github.com/FastLED/FastLED/blob/08388047ac3e0ebcaafe71563ae2a55421d70c02/src/colorpalettes.cpp#L43) is a TProgmemRGBPalette16. All other TProgmemRGBPalette16s contain 16 colors except LavaColors_p only contains 15:

```
extern const TProgmemRGBPalette16 LavaColors_p FL_PROGMEM =
{
    CRGB::Black,
    CRGB::Maroon,
    CRGB::Black,
    CRGB::Maroon,

    CRGB::DarkRed,
    CRGB::Maroon,
    CRGB::DarkRed,

    CRGB::DarkRed,
    CRGB::DarkRed,
    CRGB::Red,
    CRGB::Orange,

    CRGB::White,
    CRGB::Orange,
    CRGB::Red,
    CRGB::DarkRed
};
```

Since C zero-initializes initializer lists, this means that instead of what was intended, the lava palette actually looks like this:

```
extern const TProgmemRGBPalette16 LavaColors_p FL_PROGMEM =
{
    CRGB::Black,
    CRGB::Maroon,
    CRGB::Black,
    CRGB::Maroon,

    CRGB::DarkRed,
    CRGB::Maroon,
    CRGB::DarkRed,
    CRGB::DarkRed,

    CRGB::DarkRed,
    CRGB::Red,
    CRGB::Orange,
    CRGB::White,

    CRGB::Orange,
    CRGB::Red,
    CRGB::DarkRed,
    CRGB::Black    // <---- Automatically added at the end by the C standard.
};
```

It would be best to add the missing color back. In my own project I added one more DarkRed in position 4:
```
extern const TProgmemRGBPalette16 LavaColors_p FL_PROGMEM =
{
    CRGB::Black,
    CRGB::Maroon,
    CRGB::Black,
    CRGB::Maroon,

    CRGB::DarkRed,
    CRGB::DarkRed,
    CRGB::Maroon,
    CRGB::DarkRed,

    CRGB::DarkRed,  // <---- Added here.
    CRGB::DarkRed,
    CRGB::Red,
    CRGB::Orange,

    CRGB::White,
    CRGB::Orange,
    CRGB::Red,
    CRGB::DarkRed
};
```
and it looks quite good. I'd suggest doing that.

marmilicious, 2022-09-27:
Nice catch!

DavidSchinazi, 2023-10-26:
Looks like this was fixed by eb888d103f8dd7060a2baf2b2fbc652b08e0c4e3, so we can close this issue. Thank you!

closed 2023-10-26 by DavidSchinazi