FastLED/FastLED #1437: ESP32 Crashes when FastLED.show() is called inside an interrupt function. HELP.

Sladerix, 2022-10-07:
Hi everyone... I'm getting an annoying error.

I've an ESP32 Dev Module (38pin) and I attached an interrupt to the onboard button.
i want to push the button, trigger the interrupt that calls the function in which I turn on the LEDs.

### Code
```
#include <Arduino.h>
#include <FastLED.h>

#define BUTTON_BUILTIN 0

#define NUM_LEDS 9
#define DATA_PIN 25

CRGB leds[NUM_LEDS];

void setup() {
  Serial.begin(115200);
  pinMode(BUTTON_BUILTIN, INPUT);
  attachInterrupt(digitalPinToInterrupt(BUTTON_BUILTIN), turnOnLed, RISING); // Testing purposes

  FastLED.addLeds<WS2812B, DATA_PIN, GRB>(leds, NUM_LEDS);  // GRB ordering is typical
  FastLED.setBrightness(255); // Max

  Serial.println("Ready");

  // LED procedure to say "Ready"
  for (auto &&led : leds) {led = CRGB::Black;}
  FastLED.show();
  leds[4] = CRGB::Cyan;
  FastLED.show();
  delay(1000);
  for (auto &&led : leds) {led = CRGB::Black;}
  FastLED.show();
}

void loop() {

}

void turnOnLed() {
  //for (auto &&led : leds) {led = CRGB::Red;}
  leds[0] = CRGB::Red;
  FastLED.show();
}
```

If I try to push the button in order to call the "turnOnLed" function using interrupts, the following happen:

### Serial Monitor Output
```
Ready
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x400888ac  PS      : 0x00060a34  A0      : 0x80087ccf  A1      : 0x3ffbe9a0  
A2      : 0x3ffb8598  A3      : 0x3ffb8074  A4      : 0x00000001  A5      : 0x00000001  
A6      : 0x00060a23  A7      : 0x00000000  A8      : 0x3ffb8074  A9      : 0x3ffb8074  
A10     : 0x00000018  A11     : 0x00000018  A12     : 0x00000001  A13     : 0x00000001  
A14     : 0x00060a21  A15     : 0x00000000  SAR     : 0x00000020  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x40084f1c  LEND    : 0x40084f2b  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x40009f0f  EPC2    : 0x00000000  EPC3    : 0x400ea76a  EPC4    : 0x400888ac

ELF file SHA256: 0000000000000000

Backtrace: 0x400888ac:0x3ffbe9a0 0x40087ccc:0x3ffbe9c0 0x40086303:0x3ffbe9e0 0x40081167:0x3ffbea20 0x400d0e43:0x3ffbea40 0x400d1091:0x3ffbea60 0x400d11d1:0x3ffbeab0 0x400d0e50:0x3ffbeaf0 0x400d0e70:0x3ffbeb10 0x400811cd:0x3ffbeb30 0x40083b59:0x3ffbeb50 0x400d1d3d:0x3ffb1fb0 0x400864c9:0x3ffb1fd0

Core 0 register dump:
PC      : 0x400871ce  PS      : 0x00060034  A0      : 0x80087995  A1      : 0x3ffbe4d0  
A2      : 0x3ffbdf2c  A3      : 0x0000cdcd  A4      : 0xb33fffff  A5      : 0x00000001  
A6      : 0x00060021  A7      : 0x0000abab  A8      : 0x0000abab  A9      : 0x3ffc0418  
A10     : 0x00000003  A11     : 0x00060023  A12     : 0x00060021  A13     : 0x00000000  
A14     : 0x0000002a  A15     : 0x3ffbbbf0  SAR     : 0x0000001f  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x400871ce:0x3ffbe4d0 0x40087992:0x3ffbe500 0x400865b3:0x3ffbe520 0x4008872d:0x3ffbe540 0x40083b62:0x3ffbe550 0x400ea767:0x3ffbc130 0x400e2007:0x3ffbc150 0x4008795e:0x3ffbc170 0x400864c9:0x3ffbc190

Rebooting...
```

### EspExceptionDecoder (Core 1)
```
Decoding stack results
0x400888ac: vListInsert at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/list.c line 188
0x40087ccc: vTaskPlaceOnEventList at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 2901
0x40086303: xQueueGenericReceive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 1590
0x40081167: ESP32RMTController::showPixels() at /Users/antonellobarbone/Documents/Arduino/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp line 203
0x400d0e43: ClocklessController25, 60, 150, 90, (EOrder)66, 0, false, 5>::showPixels(PixelController(EOrder)66, 1, 4294967295u>&) at /Users/antonellobarbone/Documents/Arduino/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.h line 405
0x400d1091: CPixelLEDController(EOrder)66, 1, 4294967295u>::show(CRGB const*, int, CRGB) at /Users/antonellobarbone/Documents/Arduino/libraries/FastLED/src/controller.h line 412
0x400d11d1: CFastLED::show(unsigned char) at /Users/antonellobarbone/Documents/Arduino/libraries/FastLED/src/controller.h line 90
0x400d0e50: CFastLED::show() at /Users/antonellobarbone/Documents/Arduino/libraries/FastLED/src/FastLED.h line 502
0x400d0e70: turnOnLed() at /Users/antonellobarbone/Desktop/test/test.ino line 40
0x400811cd: __onPinInterrupt at /Users/antonellobarbone/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/cores/esp32/esp32-hal-gpio.c line 220
0x400d1d3d: loopTask(void*) at /Users/antonellobarbone/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/cores/esp32/main.cpp line 20
0x400864c9: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

### EspExceptionDecoder (Core 0)
```

Decoding stack results
0x400871ce: vTaskEnterCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/include/freertos/portmacro.h line 353
0x40087992: xTaskIncrementTick at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 2490
0x400865b3: xPortSysTickHandler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 289
0x400ea767: esp_pm_impl_waiti at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/pm_esp32.c line 492
0x400e2007: esp_vApplicationIdleHook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/freertos_hooks.c line 63
0x4008795e: prvIdleTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 3382
0x400864c9: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```  

I think this is a very basic thing to do, why this is happening?

thanks in advance

zackees, 2022-10-07:
Try:

```C
#define FASTLED_FORCE_SOFTWARE_SPI
#include <FastLED.h>
```

Also... don't call `FastLED.show()` in your interrupt routine. Interrupts are supposed to be very fast. You should instead flip a bit and then immediately return from the interrupt routine. Then this bit is checked in the `loop()` which executes the change on the next cycle. But honestly, you should certainly skip the interrupt routine altogether for the button press and just check it on iteration in the loop. The reason is very simple, interrupts and debouncing don't work well. Your button will generate a gazillion little firings microseconds apart as the button transitions from the on <--> off state.

The only time you should really use an interrupt for a button is if your main loop iteration time is large (> 250 ms) or you have put the device into a low power mode and the interrupt has is being used to wake up the device.

Sladerix, 2022-10-12:
Thank you I will try to implement it and il will close the issue

samguyer, 2022-11-25:
Thanks @zackees !
@Sladerix The reason it's crashing is that FastLED show() uses interrupts itself to manage sending the data, so the CPU probably gets into a weird recursive interrupt state when show is called inside the handler.

closed 2022-12-04 by Sladerix