FastLED/FastLED #1438: ESP32: FastLED.show hangs when core 1 is under load

Ezekiel-DA, 2022-10-10:
Hi all,

I am working on a project that uses FastLED to drive a significant number of WS2812B LEDs while performing quite a bit of other work, namely audio and BLE.

For a little bit of context (and, I will admit, a shameless attempt at trying to get more help ASAP before Halloween!), the actual "finished" project is a costume for a kiddo in a wheelchair, as part of a charity project I and several coworkers donate time to via the awesome non profit [Magic Wheelchair](https://www.magicwheelchair.org/).

You can see some shots of the costume in action and the eye-searing glory of close to 1500 LEDs, all on battery power, at an early Halloween parade in Salem, MA, in [this news report](https://whdh.com/news/salem-girl-in-wheelchair-receives-custom-costume-joins-in-annual-parade/).

Now for the issue I am attempting to figure out: as the code stands right now, when mp3 playback starts, FastLED essentially "crashes": FastLED.show() no longer returns, which blocks the main task (using the Arduino framework in Platform.io's ESP32 Platform support).

If I move the call to FastLED.show() to a separate task, that task stops running, while the main loop() task continues to run, predictably.

Hunting down a minimal reproduction for this has been a little tricky:
* without mp3 playback, FastLED behaves normally. I attempted to simulate load on Core 1 with something simpler than mp3 playback but was unsuccessful
* with fewer than 3 FastLED controllers (only 1 or 2 calls to addLeds()), the problem also stops occurring
* the number of LEDs added also seems to play a role

This is the most minimal reproduction I was able to create: https://github.com/Ezekiel-DA/FastLED_audio_hang_repro

It still relies on bringing in a whole external [I2S / mp3 decoding library](https://github.com/schreibfaul1/ESP32-audioI2S), but it is otherwise fairly simple, does not require any of the external hardware that the actual costume uses, etc. You do not need to connect any LEDs or connect anything that would allow actual sound output to see the problem occur.

To reproduce:
* clone [this repo](https://github.com/Ezekiel-DA/FastLED_audio_hang_repro) and open the project with Platform.io
* run the Upload Filesystem Image command in Platform.io to upload the sample mp3 file to flash
* upload the actual code
* observe that, once music playback starts (automatically after a set number of loops through loop()), looping through the task that runs FastLED.show() stops

Thanks in advance for any help!

Ezekiel-DA, 2022-10-10:
More information:

I found a "fix" / workaround: increasing the priority of the task dedicated to calling `FastLED.show()` to above the priority of the mp3 decoding task fixes the issue.

But: I still believe there is a hang bug in FastLED. My expertise as a software engineer is all desktop / web and very little embedded, so apologies if I am incorrect, but here is my logic:
* `loop()`, the Arduino style main task, does not hang, despite being priority 1, i.e. lower than the mp3 decoding and lower than I had the FastLED task set to in the repro code
* Adding simple serial print debugging to the repro code that traces entry and return from FastLED, I get this:

```
before FastLED show
Loop #100001
after FastLED show
before FastLED show
after FastLED show
<snip: a whole bunch more before / after, obviously>
FastLED Task Loop #101   // <== emitted by my FastLED.show() task
before FastLED show
after FastLED show
before FastLED show
after FastLED show
<... more of the same predictable output>
before FastLED show
after FastLED show
Music starts; FastLED.show() task will stop responding.
Loop #300001  // <== emitted by loop()
before FastLED show
after FastLED show
before FastLED show  // <== note that we will never hit the after trace!
Loop #350001 // main loop, at priority 1, is still getting ticks, while the show() loop at priority 3 isn't 
Loop #400001
Loop #450001
Loop #500001
<... this, endlessly, even when mp3 decoding ends>
```

zackees, 2022-10-10:
try
```C++
#define NO_HARDWARE_PIN_SUPPORT
#include <FastLED.h>
```

Ezekiel-DA, 2022-10-11:
> ```c++
> #define NO_HARDWARE_PIN_SUPPORT
> ```

Thanks for the suggestion! Unfortunately this makes no difference in either the full firmware or the minimal reproduction I posted. As far as I can tell from a quick look through the FastLED, this flag does not appear to be in use for the ESP32 platform? I can only find references to it for AVR, ARM, etc.

Ezekiel-DA, 2022-10-12:
Update: crossposted this to the reddit community [here](https://www.reddit.com/r/FastLED/comments/y1nf7a/esp32_fastled_issue_on_large_costume_project/) for more input, and found what appears to be a better workaround:

```
#define FASTLED_RMT_MAX_CHANNELS 1
```

This obviously kills most of the advantage of using the RMT driver (as far as I understand), but it does seem to fully solve my issue right now.

If anyone with expertise with the ESP32's RMT APIs wants to take a crack at this, I am more than willing to try to provide more information or any support in investigating.

zackees, 2022-10-13:
A debugging suggestion is to look at FastLED.addLeds<WS2812B, 12, GRB>(strip1Leds, 3) and see that the template argument is being used to select a particular CLED class for WS2812, which is then used to call another function with the constructed pixel controller.

You could create a dummy class for the pixel controller that does nothing, and then pass that dummy class into FastLED.addLeds().

If the program crashes, then it's something in the framekwork of FastLED. If it does crash, then it's something inside the pixel controller. It's probably something to do with the pixel controller in my opinion and in this scenario, you could then copy the real Cpixel controller class and then start commenting functions out until it works.

samguyer, 2022-11-25:
@Ezekiel-DA Apologies for getting to this issue so far past your deadline! How did it go in the end? Is there more to do here?

In general, I suspect that when the CPU is overloaded, some of the interrupts drop on the floor. As a result, what probably happens is that the semaphore that signals show() is done never gets triggered. What I should probably do is set a reasonable timeout.

TriviumGames, 2022-11-27:
I believe I have been seeing this ... for many months.  I'm using this in an escape room with Olimex ESP32-POE's.  

I have 15 LEDs in 3 strips (i.e., 3 calls to addLeds)

My loop() I think is pretty normal:
- Poll some sensors as fast as possible, send a UDP packet if stuff changes.
- Poll a UDP socket for some relatively infrequent network messages to update some CRGB values
- If it's been more than 30ms since the last time I did it, call FastLED.show()
- I guess I'm also polling for OTA updates.

Notably I am not doing anything funny with scheduling my own tasks.  Because I'm polling as fast as I can I guess I would say the CPU is fully busy? IDK.   It does seem to be the case that the network stuff uses some tasks to run things on other cores. At least, after loop() hangs, the device still responds to ICMP ping requests.

Things I have noticed:
- With no application code changes, this went from never happening, to rarely happening (~ once a week) to happening much more often (~once every couple hours).  I believe it is toolchain "upgrades" that affected this frequency. 
- With the latest everything, this is happening an average of once every 2.5 hours, which was enough to make me dig in and add enough logging to even narrow this down to FastLED.  A little under a quarter of the time it gets stuck, the ESP32 watchdog timer notices and reboots the device for me.

I will try some of the workarounds and suggested experiments here on Monday (weekends are our busy time), and even if I find something that works around this, I'm happy to do anything I can to help find a proper fix for this.

Ezekiel-DA, 2022-11-29:
@samguyer no worries!

As we had discussed in the reddit thread and here, I found a workaround for this particular project by setting `#define FASTLED_RMT_MAX_CHANNELS 1`, which seems to "fix" the problem (or at least make it so infrequent that I haven't seen it again).

That being said, I'm sure we'll do another Magic Wheelchair build at some point next year, and I'd like to stay with this ESP32 + FastLED combo. Knowing my teammates, they'll want even more hardware and software features, so I would definitely be interested in a real fix that allows multiple RMT channels to be used, for better performance with many channels.

Is there anything I can do to help diagnose this?

samguyer, 2022-11-30:
> Well I switched it over to use i2s, and so far it seems to be working 100%
> (I am assuming the several hour multi-block power outage this afternoon was
> not related to this change 😺 ). I'd be happy to help troubleshoot RMT
> stuff, though. I'll see if I can get this reproducing reliably on a spare
> ESP32 that's *not* part of anything customer-facing.

I'm planning to set aside some time soon to dig into the performance of the
RMT driver. The most helpful thing is to have examples of sketches that
cause a failure.

TriviumGames, 2022-11-30:
Great. I'll try to pare my code down to make a test case that doesn't require it to be plugged in to an escape room to do anything.

dotorg, 2022-11-30:
Just to add another raised hand on this issue -- after quite a bit of time trying to diagnose a very infrequent hang on two of my devices, I finally tracked it down to FastLED.show() not returning. Sometimes after ten minutes, sometimes after a few days of calling it every 50ms.   As a datapoint that may, or may not, be useful -- this is on an ESP32S2, not a full ESP32. (Primarily, the difference is being single core, but software-based USB stack on them also means generally never getting any WDT output, etc)

So I don't have anything specific to help, but if it helps narrow it down, it seems to be happening on the S2 as well.

samguyer, 2022-11-30:
OK, that's useful information. I just ordered myself a few of these newer ESP32 variations, so I can try things out myself. I suspect that the code is getting stuck at the semaphore "take" at the end, which is waiting for the interrupt handler to do a "give" telling it that all the data is sent. If that doesn't happen for some reason, the "take" will just sit there. BUT I think I can set a timeout, so maybe that's a stopgap until I figure out why it's failing.

Ezekiel-DA, 2022-12-01:
Just in case this got missed and might be helpful: there is a [minimal-ish reproduction case](https://github.com/Ezekiel-DA/FastLED_audio_hang_repro) linked in the original issue above.

It's not quite as minimal as I'd like because I couldn't generate the right load on Core 1 without the MP3 decoder, but it should compile, run and reproduce the issue with no external code or additional hardware needed.

For me, this test cases consistently reproduces the hang in `FastLED.show()` on a standard ESP32 dev board.

dotorg, 2022-12-01:
That example is pretty much spot-on what my use case is -- two strips being updated in a FreeRTOS task. I'm not hammering the CPU in the main loop task, but it periodically does HTTPS links and has periods of fairly busy parsing of the data that comes back. Anecdotally, all or most of my hangs happened during that process (which led me to believe it was something in my code deadlocking until I isolated the hang to the FastLED call).  

Now, interestingly (and maybe relevant when investingating) -- only the LED update task hangs. The main loop continues to run (which is handy, I can remote reboot them via a web interface) and because of it, the WDT doesn't restart everything. My LED loop is running priority 2, vs the loop task's priority 1. Given how FreeRTOS works, that means FastLED is hanging or deadlocking in a way that is blocked waiting on something the scheduler knows about, not caught in an actual loop. (FreeRTOS on the ESP32 will starve low priority threads if high priority ones do not yield their slice back to the scheduler via vTaskDelay.)

That may help narrow down where the contention point is. 

In my particular case, switching to a single RMT channel only added about a half millisecond to my update timing, which was already only running at about 30% duty cycle, so that wasn't a big deal.  If I may make a suggestion, though -- either the docs should be updated to make clear there's an issue with the RMT interface on ESP32, or the channel count should default to 1 until there isn't. For anything but the most trivial of projects, this kind of non-deterministic hang in the middle of some complex set of logic (complex enough to keep the CPU load up) is incredibly hard to diagnose.

samguyer, 2022-12-01:
@dotorg Thank you. That's very useful information. One option for the number of channels is to base it on the actual number of strips. So, if you only have one strip attached, it will use all of the RMT memory for that one channel. (The memory architecture is very weird: the buffer memory is mapped to sequential addresses, so channels can use more memory by just overrunning the memory assigned to the next channel!)

samguyer, 2022-12-01:
@Ezekiel-DA Your organization sounds amazing. I signed myself up to be a volunteer. :-)

Ezekiel-DA, 2022-12-01:
> Now, interestingly (and maybe relevant when investingating) -- only the LED update task hangs. 

Same here, this is exactly the behavior my real app and my sample reproduce. In doing some basic debugging with printf, I can fairly confidently say that `FastLED.show()` is entered and never returns, as discussed above and in the linked reddit thread.

> either the docs should be updated to make clear there's an issue with the RMT interface on ESP32, or the channel count should default to 1 until there isn't. For anything but the most trivial of [...]

That's my feeling on this as well. As it stands, I would love to use FastLED on future Magic Wheelchair builds like the one that led to finding this issue, but I will have to plan around not being able to take advantage of multi-channel RMT, which will limit how many WS2812 LEDs I can address while maintaining a reasonable framerate. (I could upgrade to some of the clocked LEDs instead, but they're more expensive, which is actually significant when you're using 1500 of them!)

Ezekiel-DA, 2022-12-01:
> @Ezekiel-DA Your organization sounds amazing. I signed myself up to be a volunteer. :-)

@samguyer That's awesome! If you're interested in a consulting role on our team's inevitable next build (probably next summer once everyone recovers from this one), I believe we're geographically close, actually!

samguyer, 2022-12-01:
@Ezekiel-DA OK, you can find my contact information on the form I filled out. I'm in the Boston area. I couldn't tell from the web site where the organization is based -- Oregon, maybe?

For those of you looking to drive large numbers of LEDs of the same type, the I2S driver might be a better bet. The main advantage, from a software systems standpoint, is that it generates a single interrupt to refill the buffers for *all* of the parallel channels, whereas the RMT has one interrupt for each channel. That's probably why it helps to reduce the number of RMT channels.

dotorg, 2022-12-01:
Unfortunately, at least for me, I2S isn't an option because it's not working on the S2.

samguyer, 2022-12-01:
@dotorg I thought we had a PR for S2 support for I2S, but maybe I'm mistaken. Check the latest head.

Looking at the code, I'm using portMAX_DELAY as the timeout for the synchronization, which means that it will wait essentially indefinitely for the show() operation to complete. I could set that to a lower value, but then I have to choose a number.

dotorg, 2022-12-08:
https://github.com/FastLED/FastLED/issues/1422

I saw that one, and had actually made the same change myself, with the same results -- code compiles, no output. I can't find anything suggesting I2S is working on the S2.  I was going to dig into it, but the time I've had to look at FastLED has been trying to figure out a weird dithering problem causing color stripes on a matrix. If I get that figured out, I may take a look at the I2S code, as well.

davidlmorris, 2023-10-27:
I've been running into a similar problem but I have a workaround if you are prepared to get down and dirty and run a task to show the LEDs rather than directly in the loop.  I'm using Esp32 (the org), FastLED  (rmt driver), 4 x longish WS2812B LEDs strings ranging from 256 to 470.  

FastLED.Show() will very, very occasionally not return.  However, this doesn't seem to be reproducible in any reliable way.  I've had run times lasting as little as 8 minutes and one as long as 17 hours.

I've added code that places FastLED.Show() in a separate task and wraps it so it is called not more often than the framerate of the longest string, put some guards around it to prevent a call while it is active, and yet it would still occasionally hang.  

Looking at the (rather clever) clockless_rmt_esp32.cpp driver code I couldn't see anything obviously wrong, but noted the semaphore 'gTX_sem', so I added the following function:

`void IRAM_ATTR GiveGTX_sem()
    {
    if (gTX_sem != NULL)
        {
        xSemaphoreGive(gTX_sem);
        }
    }
`

And called this if FastLED.Show() took longer than one second to respond (checking from the loop next time around).   This seemed to do the trick and the system would keep running until the next hang usually many, many hours later.  I've logged some of these restarts if this would be of any interest?  (They are actually quite rare).

@samguyer  Would it be possible to add this function to the library (and  `extern void IRAM_ATTR GiveGTX_sem(); ` in clockless_rmt_esp32.h)?

This way we have a workaround until a complete solution is found (especially, as I suspect it is just one of those interrupt collisions "in the lap of the god's" thingys that can't really be controlled).

Additional information, I've played around with various settings to no (noticeable) effect:
This is my 'optimal' selection which is basically the default except for FASTLED_RMT_MAX_CHANNELS, but again that setting didn't seem to make much difference (I couldn't even see a speed change). 

`// #define FASTLED_RMT_MAX_CHANNELS 4      // Only four outputs so would 4 be better than the default 8?
// #define FASTLED_RMT_BUILTIN_DRIVER false
// #define FASTLED_ESP32_FLASH_LOCK
// #define FASTLED_RMT_SERIAL_DEBUG
// #define FASTLED_ALLOW_INTERRUPTS 1
// #define INTERRUPT_THRESHOLD 0
// #define FASTLED_INTERRUPT_RETRY_COUNT 2 // default is effectively 2
// #define FASTLED_FORCE_SOFTWARE_SPI      //The default is undefined as shown.`

One thing I haven't tried is changing 'portMAX_DELAY' to some value (like  pdMS_TO_TICKS(1000)) when xSemaphoreTake(gTX_sem, portMAX_DELAY) is called which might also be worthwhile (however, I am uncertain of the implications).

cor3kl, 2023-11-12:
Hello, I was searching the internet because I ran into the same problem of fastLED "freezing".

You seem to be right @davidlmorris. 
As @samguyer stated already here:
> In general, I suspect that when the CPU is overloaded, some of the interrupts drop on the floor. As a result, what probably happens is that the semaphore that signals show() is done never gets triggered. What I should probably do is set a reasonable timeout.

the reason is high load on cpu. 
I can reproduce this on my ESP32-S3 by using heavy tasks and giving the task in which fastLed.show() is called low priority.

Wouldn't it be a possible solution to change the blocking semaphore from infinite to specific period of time?
The longest time show() should take would be ~30µS * NUM_LEDS in case of SoftwareSPI and WS2812b.
(Maybe add some safetytime)

So in this case I'd sugget to change the lines in the rmt_controller with `SemaphoreTake(gTX_sem, portMAX_DELAY);` 
to `xSemaphoreTake(gTX_sem, maxWaitTime);`
with `const TickType_t maxWaitTime = pdMS_TO_TICKS( (NUM_LEDS * 30 / 1000) +100 ); `

100ms safetytime? 
Normally the semaphore should be released way before reaching the time.
Perhaps there is a better way to determine the required number of ticks.

davidlmorris, 2023-11-12:
@ cor3kl It is a good idea, but one of the things I meant by 'uncertain of the implications', is that while WS2812B LEDs have a fixed output frequency reflected in your formulas, this isn't true for every LED type, and this could make it complicated to get a good value.  

Clearly, this was an issue at some point because this seems to appear in _most_ of the drivers and doesn't seem to change despite all the different MCUs and LED chips:

`        virtual uint16_t getMaxRefreshRate() const
            {
            return 400;
            }`

(Surely, the WS2812B driver at least could return a calculated value based on the frequency and gap?)

cor3kl, 2023-11-13:
> One thing I haven't tried is changing 'portMAX_DELAY' to some value (like pdMS_TO_TICKS(1000)) when xSemaphoreTake(gTX_sem, portMAX_DELAY) is called which might also be worthwhile (however, I am uncertain of the implications).

Sorry, I somehow did not read your very last sentence.

Semaphore is released at four points in clockless_rmt_driver.
We should find out, where exactly fastLed.show() hangs and if it is always at the same point.

       

```

else
{
  portBASE_TYPE HPTaskAwoken = 0;'
  xSemaphoreGiveFromISR(gTX_sem, &HPTaskAwoken);
  if (HPTaskAwoken == pdTRUE) portYIELD_FROM_ISR();
}

```

what if exactly after the line portBASE_TYPE HPTaskAwoken = 0;
a task with higher priority starts and portYIELD_FROM_ISR() is called?
would this lead to a deadlock?

davidlmorris, 2023-11-14:
I thought that is what the test for HPTaskAwoken is trying to avoid, but we are looking at rare (though not impossibly rare) events.  So try setting FASTLED_RMT_BUILTIN_DRIVER to true (I think in my tests it didn't make any difference, but I would need to go back and check now).  But to be honest I'm a tad out of my depth here and anything I would suggest is mere speculation.

I was rather hoping for some feedback from @samguyer.

samguyer, 2023-11-14:
Hi! I mostly took the framework for this code from examples in the FreeRTOS documentation. I'm not sure how important it is -- or how often it actually gets triggered. But you can see the examples here:
https://www.freertos.org/a00124.html

samguyer, 2023-11-14:
@davidlmorris I'd be interested to know more about what happened when you used the built-in RMT driver. I provided that option for people who have applications that need to use the RMT for other things (like the intended purpose of sending/receiving IR signals).

davidlmorris, 2023-11-14:
@samguyer My recollection is that it didn't make any difference.  And as it turns out my current settings (I was last actually trying to force the error!) have that set.  I'll set it back to the settings below and I'll run another long long test overnight tonight my time and see if/when it fails in about 24 hours.

// default settings
#define FASTLED_RMT_MAX_CHANNELS 4      // Only four outputs (pins) so would 4 be better the default 8?
// #define FASTLED_RMT_BUILTIN_DRIVER false
#define FASTLED_ESP32_FLASH_LOCK
// #define FASTLED_RMT_SERIAL_DEBUG
#define FASTLED_ALLOW_INTERRUPTS 1
#define INTERRUPT_THRESHOLD 0
#define FASTLED_INTERRUPT_RETRY_COUNT 2 // default is effectively 2
#define FASTLED_FORCE_SOFTWARE_SPI  
// #define FASTLED_RMT_SERIAL_DEBUG 

// Current debug settings
#define FASTLED_RMT_BUILTIN_DRIVER false
#define FASTLED_RMT_SERIAL_DEBUG  1
#define FASTLED_RMT_SERIAL_DEBUG  1

In the meantime would it be possible to have 
`void IRAM_ATTR GiveGTX_sem() { if (gTX_sem != NULL) { xSemaphoreGive(gTX_sem); } } `
added to clockless_rmt_esp32.cpp, and  
`extern void IRAM_ATTR GiveGTX_sem();`  
in clockless_rmt_esp32.h)?

This way my work around code (which I can publish in full here if you would like) will keep working in future updates of FastLED.

I'll let you know the results of the test.

davidlmorris, 2023-11-14:
Corrected to 
'#define FASTLED_RMT_BUILTIN_DRIVER true'.

samguyer, 2023-11-14:
Hmmm. I'd like to try to understand why the semaphore is not being released properly before we add something like the GiveGTX_sem() function. Did anyone try changing the timeout to something much shorter than portmax_delay?

samguyer, 2023-11-14:
The call that should be changed is the one at the end of ESP32RMTController::showPixels (line 204 or so). That's the one that blocks until all of the data is sent.

davidlmorris, 2023-11-14:
I've not managed to get the Esp32 code debugging/decoding thing working in my system, so I can't tell where it jams, but that was the spot I suspected (hence my added call, which seems to fix it).  Changing portMAX_DELAY to one second would disguise what is going on, though as I said before, I think this is some rare Esp32-related interrupt issue (they have had them before).  If I get a chance I'll try that (I even think it will work), but first I'll test FASTLED_RMT_BUILTIN_DRIVER.  The problem with this sort of issue is that getting an error is so 'hard to force'/rare that it makes it all but impossible to track down.   I had to wait 17 hours for the last one!

davidlmorris, 2023-11-15:
Some more clues, while doing some other debugging but with FASTLED_RMT_BUILTIN_DRIVER set to true I got this:

'14:46:52:020 -> 
14:46:52:020 -> FastLED.Show() has jammed after 6.31 minutes since boot!
14:46:52:069 -> Current displayPatternNumberAll: 11.
14:46:52:119 -> Forcing semaphore reset try.
14:46:52:169 -> '

And then a few minutes later...

14:50:31:267 -> 
14:50:31:267 -> FastLED.Show() has jammed after 9.96 minutes since boot!
14:50:31:267 -> Current displayPatternNumberAll: 11.
14:50:31:267 -> Forcing semaphore reset try.
14:50:31:310 -> 

This made me wonder if it wasn't something time-of-day related, like RF or power supply glitches.  The MCU is drawing only 21 milliamps through a USB wall wart supplying ~4.9 volts, and so I am quite convinced that the power supply is (at least should be) good enough.

Interestingly, and perhaps more concerning, the speed difference was only pretty minor 1% or 2% in favor of 'FASTLED_RMT_BUILTIN_DRIVER=false' depending on what was going on with the pattern formatting code.  (However, one pattern had lots of math up front and little display code, so less was going on, and that one was about 100% faster in favor of 'FASTLED_RMT_BUILTIN_DRIVER=true'!... though I should check this as it seems like a rather large outlier).

The other issue was that there was some minor stuttering of a scrolling text display on a matrix with 'FASTLED_RMT_BUILTIN_DRIVER=false' where 'FASTLED_RMT_BUILTIN_DRIVER=true' was pretty smooth (this was repeatable).

And as I finish this message... yet another:
15:05:36:755 -> 
15:05:36:755 -> FastLED.Show() has jammed after 25.06 minutes since boot!
15:05:36:804 -> Current displayPatternNumberAll: 11.
15:05:38:119 -> Forcing semaphore reset try.
15:05:38:122 -> 

This makes the time of day thing look more likely since I didn't have any of these when running the system this morning (and three in a row is rare).  So my not terribly well-informed speculation is guess now is: hardware glitches causing interrupts to fail to fire caused by RF, wifi, or mains power supply glitches.

I'll see if I can get some better timings to show the difference between 'FASTLED_RMT_BUILTIN_DRIVER=false' and 'FASTLED_RMT_BUILTIN_DRIVER=true'.

davidlmorris, 2023-11-15:
I should add to aid my debugging, that I am sending a moderate amount of serial data, so that would also be loading the MCU.

davidlmorris, 2023-11-15:
One further comment... I note 'gWait.wait();' is blocking.  I know it is microseconds so is probably not material here (even if every cycle is precious!).

davidlmorris, 2023-11-15:
At the risk of spamming here:  ignore my comments about speed differences above.  Both drivers ('FASTLED_RMT_BUILTIN_DRIVER=false' and 'FASTLED_RMT_BUILTIN_DRIVER=true') are very close, well under a percent difference (closer to 1 in 1000), though in the two I tested 'FASTLED_RMT_BUILTIN_DRIVER=true' was (just) faster.  There are lots of variables here, so I wouldn't read much into this except that there is not a material speed improvement.

(Higher is better = average loops per second after a few minutes.)
Pattern 11: true 761, false 763.
Pattern 16: true 1005, false 1004

samguyer, 2023-11-15:
The built-in driver is probably faster because it does all of the data translation up front (that is, it converts all of the LED data bits into RMT wave patterns before it starts sending the data), which is probably more efficient. BUT it requires a lot more memory because each bit in the LED data turns into a 32-bit word in the RMT.

samguyer, 2023-11-15:
The gWait.wait() call just ensures that we don't try to send another frame before the previous one has latched in the LED strip.

davidlmorris, 2023-11-15:
I had figured gWait.wait() was doing the gap between frames, right?  And I assume blocking because the cost of a state transfer isn't worth it for 50 microseconds?  But I hadn't looked at the memory size issues.  (I read somewhere that MS in the early days of compiler optimisation had found that size reduction usually outperformed going just after pure speed.)  In any event, I think I've demonstrated that it probably isn't the code that directly causes the 'lost?' interrupt.  But I do think we need a 'work-around' for this issue.   I've implemented:

```
#if DEBUG_USE_PORT_MAX_DELAY_FOR_gTX_sem
        xSemaphoreTake(gTX_sem, portMAX_DELAY);
#else
        xSemaphoreTake(gTX_sem, 2000 / portTICK_PERIOD_MS);
#endif
```
And set DEBUG_USE_PORT_MAX_DELAY_FOR_gTX_sem to false.  So it should only wait 2 seconds and then carry on.  However, after 4 hours, there have not been any more 'hangs'!  (And I won't be able to test at the same time of day for several days now).  Hopefully, we will get one overnight.

samguyer, 2023-11-15:
OK, great! So maybe that is the temporary solution. I've been thinking about a redesign of this driver to make it less vulnerable to higher priority interrupts. I just need a block of time to focus on it!

Re gWait.wait(): all it does is block until AT LEAST 50ms has passed since the last show, so if more that 50ms has already passed, it doesn't do anything. The gWait().mark() call below it is what "starts" that timer (really, it just records the current time).

davidlmorris, 2023-11-15:
Well, disappointingly overnight there were Zero hangs recorded.

```
...
05:14:42:131 -> Running continuously for 10.00 hour(s) after boot.
05:29:42:531 -> Running continuously for 615.00 minutes(s) after boot.
05:44:41:987 -> Running continuously for 630.00 minutes(s) after boot.
05:59:42:692 -> Running continuously for 645.00 minutes(s) after boot.
06:14:42:135 -> Running continuously for 11.00 hour(s) after boot.
06:29:42:368 -> Running continuously for 675.00 minutes(s) after boot.
06:44:41:712 -> Running continuously for 690.00 minutes(s) after boot.
```
As I had reduced the serial output to just what you see above, I assume the MCU wasn't busy enough to lose an interrupt.  I'll try to keep it running as long as possible, and perhaps there was something going on with the solar power system or the mains in the early afternoon that might trigger it again (if I am able to/windows allows/and I don't need the computer).

davidlmorris, 2023-11-15:
(I should add that I left the code for detecting a hang after one second in place - so the event should be recorded, it is just that the semaphore reset is not sent.  If the time-out fails the MCU reboots after 15 secs.)

cor3kl, 2023-11-16:
Which way do you detect if the code hangs?

davidlmorris, 2023-11-16:
Showing my work...

Basically, there is a task running (on core 1 same as FastLed and the loop code) that wraps the  FastLED.show between global bools NotShowing = false; and NotShowing = true;  There are lots of yields so other tasks get a chance to run since FastLED.show might have kept the system busy.

```
/// @brief FastLED.show task.  Trigger with xTaskNotifyGive(FastLedShowHandlerTaskSignal)
/// @param  param unused.
void IRAM_ATTR fastLedShowHandlerTask(void* param)
    {
# if DEBUG_SERIAL_OUT_IS_ON
    xSemaphoreTake(xBinarySemaphoreSerialPort, pdMS_TO_TICKS(500));
    Serial.print("fastLedShowHandlerTask running on core ");
    Serial.print(xPortGetCoreID());
    Serial.println(".");
    xSemaphoreGive(xBinarySemaphoreSerialPort);
    vTaskDelay(pdMS_TO_TICKS(100));
# endif
    bFastLedInitialised = true;
    bFastLedReady = true;
    while (true)
        {
        // Sleep until the ISR gives us something to do, or for our 
        // timeout period in ticks (which is basically forever).
        ulTaskNotifyTake(pdFALSE, portMAX_DELAY);
        NotShowing = false;
        yield();
        while (bFastLEDShowWait)
            {
            yield();
            }
        FastLED.show(uiBrightness);
        yield();
        NotShowing = true;
        }
    }

```
This is triggered once per loop with the following code, every n milliseconds where n is the slowest framerate of a FastLED pin rounded down if NotShowing is true (which avoids tripping over FastLED.Show() calling it more often than the relevant framerate, and also allows us to spend more time on other tasks).

Otherwise, after a second if it is still blocked (NotShowing =false:- meaning FastLED.Show() has jammed) it will display a message.  After another second, the change to the RMT driver from the message above will activate, and if that fails in another 13 seconds it will reboot the Esp32.

```
/// @brief Used as a replacement for FastLED.Show() to minimise hangs and crashes
// and to not call FastLED.Show() more often than it can do an update.
/// @param  
void FastLEDshow(void)
    {
    static uint32_t restartCount = 0;
    static uint64_t restartNextUs = esp_timer_get_time() + 1000000;
    // This is an attempt to reduce contention issues with FastLED and
    // the rest of the code for the random rare hangs...  

    if (NotShowing)
        {
   // An attempt to resolve Esp32/FastLED timing issues
   // basically no point in calling this more than the frame rate
   // that the slowest FastLED channel can deliver.
   // I'm not entirely sure this actually has a noticeable effect, but 
   // my very subjective opinion is it slightly reduced some stuttering, that
   // I could only really see in scrolling text.

#if FASTLED_STUTTER_REDUCTION
        EVERY_N_MILLISECONDS(frameRateInMilliseconds)
#endif
            {
            xTaskNotifyGive(FastLedShowHandlerTaskSignal);
            }
        restartCount = 0;
        }
    else
[... 48 lines cut ...]
                Serial.print("FastLED.Show() has jammed after ");
                Serial.print((esp_timer_get_time() / 1000000.0) / 60.0);
                Serial.print(" minutes since boot!");
                Serial.println("");
                vTaskDelay(tickATinyBit);
                Serial.print("Current displayPatternNumberAll: ");
                Serial.print(displayPatternNumberAll);
                Serial.println(".");
                vTaskDelay(tickATinyBit);
                Serial.println("Resetting Now!");
                vTaskDelay(tickATinyBit);
                Serial.println("");
                Serial.println("");
                Serial.println("");
                xSemaphoreGive(xBinarySemaphoreSerialPort);
                vTaskDelay(pdMS_TO_TICKS(500));
# endif            
                ESP.restart();
                }
            restartCount++;
            }
        }
    }

```

davidlmorris, 2023-11-16:
@samguyer The code ran for 19 hours plus, but I had to go out, and all those LEDs and power supplies was a bit much of a fire risk for me to be entirely comfortable keeping it running, so I stopped it.  Sadly, there were no jams despite have 3 in 20 minutes the other day.

I'll run it overnight tonight in the hope that something will happen (of course, my code shown above may not be doing what I think it is!).
```

12:59:40:580 -> Running continuously for 1065.00 minutes(s) after boot.
13:14:40:821 -> Running continuously for 18.00 hour(s) after boot.
13:29:40:206 -> Running continuously for 1095.00 minutes(s) after boot.
13:44:40:521 -> Running continuously for 1110.00 minutes(s) after boot.
13:59:40:720 -> Running continuously for 1125.00 minutes(s) after boot.
14:14:39:616 -> Running continuously for 19.00 hour(s) after boot.
14:29:39:546 -> Running continuously for 1155.00 minutes(s) after boot.
```

restarted:

```
dlm_all_leds-: Hello!
Starting comms 3.76 seconds after boot.
Main App running on core 1 at 240 MHz.
fastLedShowHandlerTask running on core 1.
FASTLED_MATRIX_LEFT WS2812B with 256 Leds, max framerate 130.21 fps, dither = 0.
FASTLED_MATRIX_RIGHT WS2812B with 256 Leds, max framerate 130.21 fps, dither = 0.
FASTLED_STRAND_LEFT WS2812B with 470 Leds, max framerate 70.92 fps, dither = 0.
FASTLED_STRAND_RIGHT WS2812B with 470 Leds, max framerate 70.92 fps, dither = 0.
FastLED Lowest frame rate in use is 70 (14 ms per frame).
writeDmxHandlerTask running on core 0.
dmxDisplaySetup (DMX output) SUCCESS.
writeOutputAnaLedsHandlerTask running on core 0.
analogDisplaySetup (analog led output) SUCCESS.
Starting fastReadAnalogueSetup with 1 fast analog pin(s).
Using voltage ref stored in eFuse.
fastReadAnalogueSetup complete.
tM1637HandlerTask running on core 0.
modeBtnHandlerTask running on core 0.
rotEncHandlerTask running on core 0.
SPIreadTask running on core 0.
Initialisation Complete 11.77 seconds after boot.
```

samguyer, 2023-11-16:
Interesting. I wonder why it's not happening now. Thanks for all your work on investigating this issue!

davidlmorris, 2023-11-16:
It is quite random and possibly environmental, and I can prove it! (Yay, a result!).  Twice overnight, though I forgot to set the 'time of day' on the logs, but the start was around 21:30.

The semaphore timeout (as expected) worked.  However, what the value should actually be is another matter entirely, and I would caution against pushing the envelope there.

```
Initialisation Complete 11.77 seconds after boot.
Running continuously for 15.00 minutes(s) after boot.
Running continuously for 30.00 minutes(s) after boot.
Running continuously for 45.00 minutes(s) after boot.
Running continuously for 1.00 hour(s) after boot.
Running continuously for 75.00 minutes(s) after boot.
Running continuously for 90.00 minutes(s) after boot.
Running continuously for 105.00 minutes(s) after boot.

FastLED.Show() has jammed after 108.88 minutes since boot!
Current displayPatternNumberAll: 26.
Waiting another second while gTX_sem times out.

Running continuously for 2.00 hour(s) after boot.
Running continuously for 135.00 minutes(s) after boot.
Running continuously for 150.00 minutes(s) after boot.
Running continuously for 165.00 minutes(s) after boot.
Running continuously for 3.00 hour(s) after boot.
Running continuously for 195.00 minutes(s) after boot.
Running continuously for 210.00 minutes(s) after boot.

FastLED.Show() has jammed after 210.72 minutes since boot!
Current displayPatternNumberAll: 25.
Waiting another second while gTX_sem times out.

Running continuously for 225.00 minutes(s) after boot.
Running continuously for 4.00 hour(s) after boot.
Running continuously for 255.00 minutes(s) after boot.
Running continuously for 270.00 minutes(s) after boot.
Running continuously for 285.00 minutes(s) after boot.
Running continuously for 5.00 hour(s) after boot.
Running continuously for 315.00 minutes(s) after boot.
Running continuously for 330.00 minutes(s) after boot.
Running continuously for 345.00 minutes(s) after boot.
Running continuously for 6.00 hour(s) after boot.
Running continuously for 375.00 minutes(s) after boot.
Running continuously for 390.00 minutes(s) after boot.
Running continuously for 405.00 minutes(s) after boot.
Running continuously for 7.00 hour(s) after boot.

```

davidlmorris, 2023-11-17:
I would caution against doing 'tight bounds', as it might mask other problems.   I would suggest it should be a definable value or variable, rather than hard coded.  And, I would also suggest that my modified code to reset the semaphore also be included so that you could trigger it 'manually' so future us/someone might track down the cause of the 'hangs' (insertion of debug code et al).  I chose two seconds because, in an hour-long plus light show, one or two seconds will not really be that noticeable compared to a hang!  Though that may well not be long enough.  I've heard of some addressable LED types that can be ponderously slow.

In case it wasn't clear the logs above show the hangs when 'FASTLED_RMT_BUILTIN_DRIVER=true'.  And previously, when 'FASTLED_RMT_BUILTIN_DRIVER=false' which to me hints at a more general problem with Esp32 and lost interrupts.

niiiksh, 2024-03-22:
Okay, I have the same problem with show freezes. Is there a fix in a near perspective?

niiiksh, 2024-03-27:
What about a fix, is it planned? I've noticed FastLED.show() freezes when using Pacifica sketch with ESP32-S3 board. SO it should be easy to reproduce I think.

davidlmorris, 2024-08-01:
I've come back to this after a year or so, to upgrade the project to Platformio and (perhaps) to eventually publish this on Github and even make a YouTube clip.

@samguyer I ran some tests again (just short ones to make sure the code was running), and discovered (after three days of head scratching) that when 'FASTLED_RMT_BUILTIN_DRIVER=true' the code usually somewhere deep inside clockless_rmt_esp32 crashes with a Store Prohibited error.

As far as I can tell, this means either something has silently updated, or what I said above about testing with FASTLED_RMT_BUILTIN_DRIVER define set was rubbish. 

I confess to being at a complete loss as to what is what here (especially when the code seems fine Prima facie when FASTLED_RMT_BUILTIN_DRIVER  is not defined).  It is possible I was placing the 'defines' in the wrong place, and they were not being picked up by FastLED.h, though this seems unlikely.  Plus a Guru Meditation seems like something that would have been noticed by others, right?

```
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4010f62a  PS      : 0x00060730  A0      : 0x800d5b74  A1      : 0x3ffb4c20
A2      : 0x3ffc2600  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3ffb4c50  
A6      : 0x00000003  A7      : 0x00060123  A8      : 0x00000000  A9      : 0x0028800a
A10     : 0x00000000  A11     : 0x3ffc2608  A12     : 0x80084da6  A13     : 0x3ffb4c20  
A14     : 0x00000001  A15     : 0x3ffb2340  SAR     : 0x0000000f  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x40085521  LEND    : 0x40085529  LCOUNT  : 0x00000027  

Backtrace: 0x4010f627:0x3ffb4c20 0x400d5b71:0x3ffb4c40 0x400d5ad2:0x3ffb4c60 0x400d2ae1:0x3ffb4cb0 0x400817c1:0x3ffb4cf0
  #0  0x4010f627 in ESP32RMTController::convertByte(unsigned int) at .pio/libdeps/esp32doit-devkit-v1/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:551 (discriminator 4)
  #1  0x400d5b71 in ClocklessController<32, 60, 150, 90, (EOrder)66, 0, false, 5>::convertAllPixelData(PixelController<(EOrder)66, 1, 
4294967295u>&) at .pio/libdeps/esp32doit-devkit-v1/FastLED/src/platforms/esp/32/clockless_rmt_esp32.h:423
      (inlined by) ClocklessController<32, 60, 150, 90, (EOrder)66, 0, false, 5>::showPixels(PixelController<(EOrder)66, 1, 4294967295u>&) at .pio/libdeps/esp32doit-devkit-v1/FastLED/src/platforms/esp/32/clockless_rmt_esp32.h:400
  #2  0x400d5ad2 in CPixelLEDController<(EOrder)66, 1, 4294967295u>::show(CRGB const*, int, CRGB) at .pio/libdeps/esp32doit-devkit-v1/FastLED/src/controller.h:641
  #3  0x400d2ae1 in CLEDController::showLeds(unsigned char) at .pio/libdeps/esp32doit-devkit-v1/FastLED/src/controller.h:127
      (inlined by) CFastLED::show(unsigned char) at .pio/libdeps/esp32doit-devkit-v1/FastLED/src/FastLED.cpp:68
  #4  0x400817c1 in fastLedShowHandlerTask(void*) at src/displayFastLedCommon.cpp:273
```

samguyer, 2024-08-01:
@davidlmorris That's a very weird place to crash. All it is doing there is taking the CRGB data from the pixel array and converting it to data to send on the line (applying the global brightness and color adjustments, and fixing the color order for the particular type of strip).

I have not done extensive testing on the built-in RMT driver. It's not really our code at all. BUT the crash is not happening in that code, which is interesting. I need to look more closely at it.

samguyer, 2024-08-01:
@davidlmorris Actually, I thought of another possible problem. Do you have a large number of pixels in this program?

davidlmorris, 2024-08-01:
Funnily enough, I was thinking along those lines from a hint you gave above. (Memory issues.  However, my original thoughts were it was it wasn't correctly finding the end of the buffer because I had multiple controllers not using contiguous memory... but buffer guards didn't stop the crash).

So, there are 4 strands of LEDs, 2 x 256 matrix, and 2 x strands of 'sticks' each totaling 470 for a total of 1452 pixels.  Moreover, my code is doing a lot... DMX lights and '6 x Analogue' strands (with a sort of PWM), displaying around 30 different patterns, plus SPI receiving audio data, and a button, rotary encoder and (non-FastLED) SPI LED output.  (And still getting 1000 loops a second.)

This says I still have some capacity!
```
RAM:   [=         ]  11.2% (used 36840 bytes from 327680 bytes)
Flash: [===       ]  31.4% (used 411557 bytes from 1310720 bytes)
```
If I get a chance today I'll put an assert after the calloc in ```void ESP32RMTController::initPulseBuffer(int size_in_bytes)``` (line 539)  as my next suspect.

davidlmorris, 2024-08-01:
Well now...
(I'm not sure I understand this since 3 x 1452 x uint32_t = 17.4k, right? - in any case after adding a FAST_LED debug (pixels.size() * 3) = 1410 which would be the first controller with a strand with 470 pixels.)
```
FASTLED: initPulseBuffer size_in_bytes: 1410

assert failed: void ESP32RMTController::initPulseBuffer(int) clockless_rmt_esp32.cpp:541 (mBuffer != 0)

Backtrace: 0x40084551:0x3ffb4ab0 0x400894b1:0x3ffb4ad0 0x4008f145:0x3ffb4af0 0x400d2dbb:0x3ffb4c20 0x400d4eda:0x3ffb4c40 0x400d5951:0x3ffb4c60 0x400d2a45:0x3ffb4cb0 0x4008177a:0x3ffb4cf0
  #0  0x40084551 in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:408
  #1  0x400894b1 in esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c:137
  #2  0x4008f145 in __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c:85
  #3  0x400d2dbb in ESP32RMTController::initPulseBuffer(int) at .pio/libdeps/esp32doit-devkit-v1/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:541 (discriminator 1)
  #4  0x400d4eda in ClocklessController<32, 60, 150, 90, (EOrder)66, 0, false, 5>::convertAllPixelData(PixelController<(EOrder)66, 1, 4294967295u>&) at .pio/libdeps/esp32doit-devkit-v1/FastLED/src/platforms/esp/32/clockless_rmt_esp32.h:416
      (inlined by) ClocklessController<32, 60, 150, 90, (EOrder)66, 0, false, 5>::showPixels(PixelController<(EOrder)66, 1, 4294967295u>&) at .pio/libdeps/esp32doit-devkit-v1/FastLED/src/platforms/esp/32/clockless_rmt_esp32.h:400
  #5  0x400d5951 in CPixelLEDController<(EOrder)66, 1, 4294967295u>::show(CRGB const*, int, CRGB) at .pio/libdeps/esp32doit-devkit-v1/FastLED/src/controller.h:641
  #6  0x400d2a45 in CLEDController::showLeds(unsigned char) at .pio/libdeps/esp32doit-devkit-v1/FastLED/src/controller.h:127
      (inlined by) CFastLED::show(unsigned char) at .pio/libdeps/esp32doit-devkit-v1/FastLED/src/FastLED.cpp:68
  #7  0x4008177a in fastLedShowHandlerTask(void*) at src/displayFastLedCommon.cpp:273

```

davidlmorris, 2024-08-01:
Oops missed:  mBufferSize = size_in_bytes * 8 * 4;
So we should be allocating:  [470 (pixels) * 3]bytes * 8 * 4  = 45120 (uint32_t) or 180,480 bytes.

davidlmorris, 2024-08-02:
Some Googling later...

This [stackoverflow](https://stackoverflow.com/questions/71085927/how-to-extend-esp32-heap-size) discussion pointed to Esp32 Docs [Memory Types](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/memory-types.html#dram-data-ram) which says:

> Note
> 
> There is 520 KB of available SRAM (320 KB of DRAM and 200 KB of IRAM) on the ESP32. However, due to a technical limitation, the maximum statically allocated DRAM usage is 160 KB. The remaining 160 KB (for a total of 320 KB of DRAM) can only be allocated at runtime as heap.
> 

And 180,480 > 160 KB, so I guess that is the answer.  This means either the code following needs to be broken down more, or we are limited to around ~415 pixels in a controller.

niiiksh, 2024-08-02:
@davidlmorris I have only 364 pixels
 [364(pixels) * 3]bytes * 8 * 4 = 34944 (uint32_t) (139,776 bytes)
and experiencing something like described problem with show freeze with running standard example of FastLED library called Pacific. I have ESP32S3.
Workaround with #define FASTLED_RMT_MAX_CHANNELS 1 seems like fixed problem but one time it appeared again, it is rare case now with this workaround. I also experience something like white flicker once in a period of time (like 1 time in a hour or less) just for milliseconds and then returning to normal. Maybe it is related to this issue somehow too.

[9 more replies]

closed 2024-08-08 by zackees