FastLED/FastLED #1439: LED setup

ChrisvV1, 2022-10-15:
Hey community, 
I started a project where i made an app myself, but now i want to set up my LED strip (ws2812b 5050). 
I was thinking of using the app to let the user adjust the lighting until the first and last LED are on. 
Then a whant to use this value as NUM_LEDS instead of hardcodeing this.
Has anyone tried something like this before or has someone advice for me?

is this even possible?

marmilicious, 2022-10-16:
The easiest solution if it would work for your case is to pick a maximum number of leds your app would allow and make that the size of the CRGB leds array under the hood. Then have the user set their number of leds up to that maximum.

ChrisvV1, 2022-10-16:
> marmilicious

Thank you.
I had the same idea today. it's not perfect, but it works for now.

samguyer, 2022-11-25:
@ChrisvV1 You'll get a lot of great answers to questions like this one over on the FastLED subrededit: https://www.reddit.com/r/FastLED/

closed 2022-11-25 by samguyer