FastLED/FastLED #1441: Support for TM1908

cheemsHe, 2022-10-31:
Can you add support for TM1908? It is a single line high brightness Three-Channel LED Constant Current Driver Chip?

smurphyUSI, 2022-11-01:
Do you have a datasheet for this?

All I could find online is what appears to be a cloned WS2815, but using an external controller chip, instead of that is integrated.

If it is just a clone, then you should have four contacts; Ground, Data, Backup/Failover, 12V.
In which case, you can the WS2812B definition in FastLed to drive the pixels.
I believe that @marmilicious once pointed out that such clones would likely still use 5V on data, but need a 12V on power.

If your LEDs aren't like this at all, please provide as much reference material for them, or pictures, as you can.

cheemsHe, 2022-11-03:
I have datasheet for this；but I‘am a rookie；

cheemsHe, 2022-11-03:
I have a datasheet for this; but I'am a rookie. I don't how to write the code for it  and will only use the existing library for now. If possible ,could you plesase help me add it? I will send the datasheet to you. It might help other people, too.

cheemsHe, 2022-11-03:
> Do you have a datasheet for this?
>
> All I could find online is what appears to be a cloned WS2815, but using an external controller chip, instead of that is integrated.
>
> If it is just a clone, then [...]

I have a datasheet for this; but I'am a rookie. I don't how to write the code for it  and will only use the existing library for now. If possible ,could you plesase help me add it? I will send the datasheet to you. It might help other people, too.

marmilicious, 2022-11-04:
Were you going to share a link to the data sheet?

cheemsHe, 2022-11-07:
This link contains data about the chip.
https://github.com/cheemsHe/LEDdatasheet

zackees, 2024-09-20:
The T1, T2, T3 timings in our code are a little confusing and not the format used in data sheets, but is actually a more optimal representation of clockless settings.

If anyone is still interested in this then i should mention that in chipset.h there is a python script embedded in the comments that will generate T1...T3 settings from the datasets T1H...T0L timing formats.

closed 2026-09-02 by zackees