FastLED/FastLED #1442: D1mini and 8 led ring

tjodork1960, 2022-11-05:
I'm struggling with reliability ...i have a switch input to a D1mini that tells the LED ring to toggle from off to on and vice versa.
If I don't change the color it doesn't have any problems. I can toggle off and on and the ring obeys
The thing is I have a color ring...I'd like be able to set random colors.
When I toggle the switch AND change color sometimes I get NO lights on.
I dropped the voltage to the ring to 4.3 (w a diode) so I think the 3.3 should be high enough to drive the LED ring as (.7 x 4.3)=3v
So I don't think its a voltage thing.
I tried adding the delay as someone had mentioned after the Show but that didn't help.
1. Any suggestions?  
2. Any code I'm missing?
3. Is there any way to sense that the lights are not on (in FastLED code) so I could retry turning on lights?
Love FastLED ...not loving my current design : )

marmilicious, 2022-11-05:
Hello @tjodork1960 
We use github "issues" here mostly just for bugs or improvements in the library itself.  For all general discussions, help, and how-to, please use the FastLED Reddit group here: https://www.reddit.com/r/FastLED
And please provide a **link** to your code on pastebin.com or gist.github.com, otherwise there's no way for us to know what might  be missing?

tjodork1960, 2022-11-05:
I added code...not sure it was done correctly...I thought it would end up in a scrollable box....
Thanks for the link to Reddit

marmilicious, 2022-11-05:
Thank you for posting in the reddit group. Feel free to close this issue.

tjodork1960, 2022-11-05:
got answer on reddit..thanks to all

closed 2022-11-05 by tjodork1960