FastLED/FastLED #1443: Too slow data rate for dotstar

debojitk, 2022-11-05:
Hi I am trying to build a POV with dot star strip that is made of SK9822on esp8266 platform. With default settings (80 mhz) for a strip of 72 leds FastLED.show takes around 2000 us. Is this normal? Strangely on Arduino I get the transfer rate as 980 us. I guess FastLED can now use hspi, and the speed should ideally be way faster. It looks like SPI clock rate is around 1 mhz. I used the following calculation to get the data date - 
72 leds * 4 byte per led = 288 bytes +  8 bytes leading and trailing = 296 * 8 = 2368 bits. If the SPI speed is 40 mhz the should be 2368/40 mhz = 60 us. Where I am getting wrong.
On the other hand I tried using adafruit dotstar lib that gave me 3426 us to render the same strip. I am awestruck looking at the results.
Please help me out here. 
Thanks,
Debojit

yesyesuk, 2022-11-05:
The SK and WS LEDs have a fixed clock/data rate because they don't have a separate clock input. So you can't just send the data faster. (though, I don't know why the Arduino would need less than half the time).

For higher data rates you would need to use LEDs with clock input like the APA10x LEDs.

marmilicious, 2022-11-05:
@yesyesuk SK9822 is similar to/a clone of APA102C and uses a clock line.

@debojitk Have you experimented with changing the data rate? It can be set in the addLeds line, such as:
`FastLED.addLeds<APA102, dataPin, clockPin, RGB, DATA_RATE_MHZ(12)>(leds, NUM_LEDS);`

For more general discussion, help, and how-to, please use the FastLED Reddit group here: https://www.reddit.com/r/FastLED 
We use github "issues" here mostly just for bugs or improvements in the library itself.

debojitk, 2022-11-06:
@marmilicious Thanks for your reply. I ran some tests and found some results as given below:
[image]
[image]
Looks like changing the data rate does not have proportional speed change.

yesyesuk, 2022-11-06:
> @yesyesuk SK9822 is similar to/a clone of APA102C and uses a clock line.

Oh, sorry, I confused them with the SK**6**822 that I used before.

samguyer, 2022-11-25:
@debojitk Do you want us to look into this issue? Is there more to do here?

zackees, 2026-10-31:
@debojitk this should now be fixed on `master`, as long as the strip is wired to the ESP8266 hardware SPI pads: **data on GPIO13 (D7) and clock on GPIO14 (D5)**.

- #4933 and #4951 send APA102/SK9822 on those pads through the native HSPI peripheral, so `FASTLED_ALL_PINS_HARDWARE_SPI` is no longer needed. Before that, FastLED bit-banged those pads too, at about 1.4 Mbit/s whatever `DATA_RATE_MHZ` asked for. That rate is the ~2000 µs you measured.
- #4936 fixed the HSPI clock encoding and added on-device benches. They time the frames, read back the clock divider the driver programmed, and check clock and data on the pins with a loopback.

For your strip: 72 LEDs × 4 bytes plus start/end frames is ≈296 bytes, or 2368 bits. That is about **395 µs on the wire at the default 6 MHz**, and ~60 µs at `DATA_RATE_MHZ(40)`. FastLED adds roughly 250 µs of fixed cost per `show()`. Expect about 0.3–0.65 ms instead of 2 ms.

If your strip is on **other pins**, it still uses the software bit-bang. That is now tracked as an implementation issue in #4966, with a measured baseline.

Closing this. If you still see ~2 ms on GPIO13/14 with current `master`, comment with your `addLeds` line and it will be reopened.

closed 2026-10-31 by zackees