FastLED/FastLED #1444: Hardware SPI on ESP32-S3

Hokyjack, 2022-11-13:
Hello is it possible to compile FastLED on ESP32-S3 board, with Hardware SPI support? I know that the SPI can be mapped to any pin, but I don't know how to specify it.

I used `#define FASTLED_ALL_PINS_HARDWARE_SPI`
and `#include <SPI.h>`
before including <FastLED.h>

This is the error of compilation:
```
In file included from lib/FastLED/src/platforms/esp/32/fastled_esp32.h:6,
                 from lib/FastLED/src/platforms.h:36,
                 from lib/FastLED/src/FastLED.h:54,
                 from src/LedArray.h:13,
                 from src/LedArray.cpp:5:
lib/FastLED/src/platforms/esp/32/fastspi_esp32.h:2:17: note: #pragma message: ESP32 Hardware SPI support added
 #pragma message "ESP32 Hardware SPI support added"
                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
lib/FastLED/src/platforms/esp/32/fastspi_esp32.h:49:35: error: 'VSPI' was not declared in this scope
     #define FASTLED_ESP32_SPI_BUS VSPI
                                   ^~~~
lib/FastLED/src/platforms/esp/32/fastspi_esp32.h:52:17: note: in expansion of macro 'FASTLED_ESP32_SPI_BUS'
 SPIClass ledSPI(FASTLED_ESP32_SPI_BUS);
                 ^~~~~~~~~~~~~~~~~~~~~
lib/FastLED/src/platforms/esp/32/fastspi_esp32.h:49:35: note: suggested alternative: 'SPI'
     #define FASTLED_ESP32_SPI_BUS VSPI
                                   ^~~~
lib/FastLED/src/platforms/esp/32/fastspi_esp32.h:52:17: note: in expansion of macro 'FASTLED_ESP32_SPI_BUS'
 SPIClass ledSPI(FASTLED_ESP32_SPI_BUS);
                 ^~~~~~~~~~~~~~~~~~~~~
lib/FastLED/src/platforms/esp/32/fastspi_esp32.h: In instantiation of 'class ESP32SPIOutput<18, 23, 24000000>':
lib/FastLED/src/fastspi.h:59:7:   required from 'class SPIOutput<18, 23, 24000000>'
lib/FastLED/src/chipsets.h:158:6:   required from 'class LPD6803Controller<18, 23, (EOrder)136, 24000000>'
lib/FastLED/src/FastLED.h:227:92:   required from 'CLEDController& CFastLED::addLeds(CRGB*, int, int) [with ESPIChipsets CHIPSET = (ESPIChipsets)6; unsigned char DATA_PIN = 18; unsigned char CLOCK_PIN = 23; EOrder RGB_ORDER = (EOrder)136; unsigned int SPI_DATA_RATE = 24000000]'
src/LedArray.cpp:20:92:   required from here
lib/FastLED/src/platforms/esp/32/fastspi_esp32.h:73:44: error: static assertion failed: Invalid clock pin specified
  static_assert(FastPin<CLOCK_PIN>::validpin(), "Invalid clock pin specified");
                ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
src/Logger.cpp: In member function 'const char* Logger::getLog()':
src/Logger.cpp:18:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^
In file included from lib/FastLED/src/FastLED.h:67,
                 from src/Show.h:15,
                 from src/Show.cpp:5:
lib/FastLED/src/fastspi.h:145:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
*** [.pio\build\um_tinys3\src\LedArray.cpp.o] Error 1
src/Show.cpp: In member function 'Effect* Show::parseEffect(String)':
src/Show.cpp:33:62: warning: control reaches end of non-void function [-Wreturn-type]
     std::vector<int> v = Helper::extract_numbers(effectString);
                                                              ^
src/WebsocketClient.cpp:69:31: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
 char * FSHelper::fileSystem = "LittleFS";
                               ^~~~~~~~~~
src/WebsocketClient.cpp: In member function 'bool WebsocketClient::removeFile(const char*)':
src/WebsocketClient.cpp:272:44: warning: control reaches end of non-void function [-Wreturn-type]
     String path = String("/") + String(name);
                                            ^
src/WebsocketClient.cpp: In member function 'bool WebsocketClient::fileExists(const char*)':
src/WebsocketClient.cpp:281:44: warning: control reaches end of non-void function [-Wreturn-type]
     String path = String("/") + String(name);
                                            ^
In file included from lib/FastLED/src/FastLED.h:67,
                 from src/WiFiClient.cpp:39:
lib/FastLED/src/fastspi.h:145:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
```

Thank you!

Hokyjack, 2022-11-14:
I have no problem running it on normal ESP32 boards with hardware SPI support, but I'm talking about S3 version here. Yesterday I was able to compile it, but I cannot run it nowhere on the same speed at ani Pin on while using ESP32-S3 board. I understand the S3 is using a FSPI, which should be able to use Hardware SPI support on any pin, but still no luck to get the desired SPI speed.

Hokyjack, 2022-11-14:
I did some small benchmark:
I'm trying to do 12000 full pattern changes of 160 LED strip withing 10 seconds.
On normal ESP32 I'm able to make 9746 changes -> 1 change per 1.231 ms
On ESP32-S3 I'm able to make 6971 changes -> 1 change per 1.721 ms
Both using hardware SPI.
With bit-banging I get around 4500 changes. 
If I disable the FastLED.show() both devices are able to get full 12000 changes. So the performance issue is not with computing and loading the pattern/effect.

The changes are run in this loop:
```
if (micros() - effect_time >= 1000000L/speed) {
    effect_time = micros()
    generate_effect_and_set_to_led()
    FastLED.show();
}
```
The code with which I made ESP32-S3 is this pull request: #1411 

Any ideas how can I obtain same hardware SPI speed as with the older ESP32?
Thanks!

smurphyUSI, 2022-11-14:
Unfortunately, I don't have an S3 here to test with, so this is likely out of my ability to help.
You may have to take this to the Reddit forum for help.

However, I know that FastLED on the original ESP32 is using the Master/Slave Mode transfer for communication to LEDs.
This means using the MOSI, MISO, CLK, CS pins.
This can run at the 80MHz clock.

The new S3 has different SPI modes, which affect which pins are used for each mode, and at what speed.

Looking at the datasheet, it appears the the PIN numbers for hardware SPI vs GPIO SPI are different.
https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf
(Pages 34,35, and then 12).

I suspect you are in a mode which is using 60MHz clock for GPIO mode instead of hardware SPI.
Hardware SPI clocks on the S3 start at pin 30, not 18.

But... this is where I would need to test.
There are configurations that can affect the mode, so I don't know if simply changing the pin numbers in fastspi_esp32.h will fix the problem.

Hokyjack, 2022-11-15:
@smurphyUSI 
Still no luck switching to pins above > 30, no matter what I set the speed stays the same. I made a clean benchmark, where I just render one pallete once, and then I call FastLED.show() in the loop and measure FPS once in a 100 cycles.
Normal ESP32: 896 FPS
S3-ESP32: 722 FPS

If you'd like to help me to solve this problem, I could send you one or two TinyS3 board for free. I would really like to solve this problem, otherwise, I'll have to switch back to previous ESP32 boards, which I really don't want to.

Hokyjack, 2022-11-15:
Ha! Interesting thing just happened: I have hardcoded the clock value of SPISettings to be 32000000L inside `fastapi_esp32h.` : `ledSPI.beginTransaction(SPISettings(32000000L, MSBFIRST, SPI_MODE0));`
and I got the exactly same FPS with S3 as I got with ESP32 without this modification (FPS=896). 
But I don't really understand this. It is forcing the APA102 to go with 32Mhz correct? Which would be above the limit, I don't know if this is the desired solution. Anyone who understand why this happened can explain?

smurphyUSI, 2022-11-17:
This makes sense.
As I mentioned before, the base datarate of the S3 is 60MHz and not 80MHz.   And, it should be possible to push it to 120MHz given the correct scenario.

Unfortunately, FastLED hasn't been updated to automatically use the new ESP32 revisions.

But, it was designed to let you override things.

What is supposed to happen; is that the FastLED template for the APA102 passes the expected data rate for the LEDs to the chipset, which then calculates the timings based on the chipset clock.    This is great, because it's SPI is usually much faster than LEDs can accept, so you don't want to drive and overdrive them (much).
This doesn't work so well if the chipset clocks aren't what you expect, or your LEDs aren't compliant with the advertised timings (i.e. cheap no-name brand LEDs)

Your override should really be when you use the APA102 template, which is coming from chipsets.h
(That is, until the EPS32-S3 is added to FastLED)

I haven't tried this personally, but you could just use the .addLeds which has the parameter for providing you own data rate
.addLeds<CHIPSET, SPI_DATA, SPI_CLOCK, RGB_ORDER, SPI_DATA_RATE>(data, nLedsOrOffset, nLedsIfOffset);

samguyer, 2022-11-25:
@Hokyjack and @smurphyUSI It sounds like we need to make a small change to support ESP32-S3 -- just a conditional compilation around the SPI clock speed? Is that right?

lmancini, 2023-04-16:
@Hokyjack @smurphyUSI I'm following this too, is there any news on this issue? Thanks!