FastLED/FastLED #1445: Seed  nRF52 mbed-enabled board - compiling error

Loaded-Dice, 2022-11-16:
Hi everyone,
I got one of the XIAO BLE Sense - nRF52840 Boards and got compiling issues with fastLED.
My version is the mbed-enabled version ( Seeed  nRF52 mbed-enabled board > Seeed XIAO BLE Sense - nRF52840)
Has anyone got this issue before?
I also tried the neopixel library, which works fine with this board.
I already looked through all nRF52 related issues here but had no luck.
Can I create a workaround where I "trick" fastLED into compiling without FastLED using any Hardware outputs?
I already got a lot of code from another project with a different platform/controller, and I want to use my old fastLED based code to fill the CRGB array and instead of using FastLED for the output, I copy the data to the neopixel structure.
Is this possible? And if so, ho to implement this?
Any help would be appreciated.
Best Regards

kakapo5140cc0, 2024-07-26:
did you get this to work? im having the same issues and really need fastled

Loaded-Dice, 2024-07-26:
Hey there,

Unfortunately, I had to use a very sketchy solution to use FastLed features with the nRF52

  If you just want to control addressable RGB LEDs in  an easy way and could go without the advantages of the FastLED 
  library (like fast math-/ timing- / noise - / color - functions), then you can simply go with the
  Adafruit_NeoPixel library -> https://github.com/adafruit/Adafruit_NeoPixel

  But if you want to use those mentioned functions in particular, then you may have to do it with this 
 workaround:
Use this core:  Seeed Studio nRF52 Boards v1.1.1  (not mbed)

include the FastLED library

And create an CRGB leds[NUM_LEDS]; array for your LEDs
 and don't initialize the hardware ( = avoid FastLED.addLeds... ) otherwise the build will fail.

Also include Adafruit_NeoPixel.h
and here you initialize 
Adafruit_NeoPixel strip = Adafruit_NeoPixel(NUM_LEDS, LED_PIN, NEO_GRB + NEO_KHZ800);
Or whatever LED controller you might have.

Then you can use the FastLED color functions on your CRGB array and copy the data from the virtual CRGB array to the Neopixel strip.

void fastToNeo(){for(int i = 0; i < NUM_LEDS; i++){ strip.setPixelColor(i, leds[i].r, leds[i].g, leds[i].b); }}

you can find my working nRF52 + FastLED  & Neopixel  project here

https://github.com/Loaded-Dice/Nightstick/tree/main/Nightstick-Arduino

https://github.com/Loaded-Dice/Nightstick/blob/b333b01ab9f8833c7c347652aabb02401f6bbb04/Nightstick-Arduino/Nightstick.h#L238C5-L238C6

good luck & regards

kakapo5140cc0, 2024-07-26:
awesome, thanks! the idea is understandable, but the execution is rather complicated. Would it be possible for you to write a blink sketch for me? it would be highly appreciated!

kakapo5140cc0, 2024-07-26:
i am reading through the whole project.. its awesome! definitely going to take some snippets of your code later... just trying to digest it as good as possible

kakapo5140cc0, 2024-07-26:
i got it working!! thanks so much

Loaded-Dice, 2024-07-26:
Great job!  :) 

But writing down a simple example to a workaround, that I found years ago, might be a great Idea anyway.
Just in case anyone else stumbles across this problem:
Here is a simple workaround to get FastLed working with the XIAO nRF52:

 https://github.com/Loaded-Dice/XIAO_nrf52_FastLED_Workaround_example

zackees, 2024-09-06:
Whats your sketch?

I'm about to push a fix that will make the compiler errors not be 20 pages long for this chipset. The problem is that we only define valid pins. Invalid pins go through a default FastPin template. For the NRF52 series I'm introducing an invalid pin specification that binds to a specialization of the FastPin<> type that will static_assert instead of going through the default template which then static asserts in a weird way.

I suspect that you used one of these invalid pins and that this was causing your issue and it's not obvious what the fix is.

Loaded-Dice, 2024-09-06:
Hi,
the sketch that caused the issue in the first place is this one:
https://github.com/Loaded-Dice/Nightstick/tree/main/Nightstick-Arduino
Which boils down to a Xiao nRF52840 Sense Board, where I used pin D1 ( P0.03_AIN1_LOW_A1_D1 )
Which didn't work, apparently.

I'm looking forward to your updates.
Best regards

zackees, 2024-09-06:
It's an invalid pin because it's used for UART RX. It probably works but kills external Serial output? Or maybe it just doesn't work at all. Hard to tell. Anyway, master now has this fix and it will compile and attempt to run. No idea if it will generate a core panic, but if it does that, you'll get a nice message on the output console that the pin isn't supported and a 2 second delay before it attempts to run.

Good luck!

zackees, 2024-09-13:
Comments here on this bug report.

Target IDE: Arduino Latest

board package json: https://files.seeedstudio.com/arduino/package_seeeduino_boards_index.json

[image]

zackees, 2024-09-13:
I'm missing something because it still won't compile.

Btw, you should convert your project to platformio, which features a real build system and a real IDE. If your project was platformio then I could just open it up and everything would just download and install. No selection of board, no selection of packages, no finding out which port I need. Everything just works. The entire Arduino package ecosystem is also available.

It works with *.ino files and should just be 100% compatible with your source code, though you might have to move all your files except the root *.ino file to the src/ directory.

Anyway, here's the error code. To continue working on this I need to know either what the ArduinoIDE steps are or have you do a quick conversion to PlatformIO.

Compile error log

```bash
In file included from C:\Users\niteris\dev\Nightstick\Nightstick-Arduino\Nightstick-Arduino.ino:180:
C:\Users\niteris\dev\Nightstick\Nightstick-Arduino\Nightstick.h:9:21: error: 'D1' was not declared in this scope; did you mean 'A1'?
    9 | #define LED_PIN     D1
      |                     ^~
C:\Users\niteris\dev\Nightstick\Nightstick-Arduino\Nightstick.h:252:55: note: in expansion of macro 'LED_PIN'
  252 | Adafruit_NeoPixel strip = Adafruit_NeoPixel(NUM_LEDS, LED_PIN, NEO_GRB + NEO_KHZ800);//254
      |                                                       ^~~~~~~
In file included from C:\Users\niteris\dev\Nightstick\Nightstick-Arduino\Nightstick.h:254,
                 from C:\Users\niteris\dev\Nightstick\Nightstick-Arduino\Nightstick-Arduino.ino:180:
C:\Users\niteris\dev\Nightstick\Nightstick-Arduino\Palettes.h:136:3: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  136 |   "rainbow1",
      |   ^~~~~~~~~~
C:\Users\niteris\dev\Nightstick\Nightstick-Arduino\Palettes.h:137:3: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  137 |   "fancy1",
      |   ^~~~~~~~
C:\Users\niteris\dev\Nightstick\Nightstick-Arduino\Palettes.h:138:3: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  138 |   "smooth1",
      |   ^~~~~~~~~
C:\Users\niteris\dev\Nightstick\Nightstick-Arduino\Palettes.h:139:3: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  139 |   "smooth2",
      |   ^~~~~~~~~
C:\Users\niteris\dev\Nightstick\Nightstick-Arduino\Palettes.h:140:3: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  140 |   "sunrise1",
      |   ^~~~~~~~~~
C:\Users\niteris\dev\Nightstick\Nightstick-Arduino\Palettes.h:141:3: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  141 |   "sunrise2",
      |   ^~~~~~~~~~
C:\Users\niteris\dev\Nightstick\Nightstick-Arduino\Palettes.h:142:3: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  142 |   "other1",
      |   ^~~~~~~~
C:\Users\niteris\dev\Nightstick\Nightstick-Arduino\Palettes.h:143:3: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  143 |   "other2",
      |   ^~~~~~~~
C:\Users\niteris\dev\Nightstick\Nightstick-Arduino\Palettes.h:144:3: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  144 |   "other3",
      |   ^~~~~~~~
C:\Users\niteris\dev\Nightstick\Nightstick-Arduino\Palettes.h:145:3: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  145 |   "other4",
      |   ^~~~~~~~
C:\Users\niteris\dev\Nightstick\Nightstick-Arduino\Palettes.h:146:3: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  146 |   "other5",
      |   ^~~~~~~~
C:\Users\niteris\dev\Nightstick\Nightstick-Arduino\Palettes.h:147:3: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  147 |   "firelike1",
      |   ^~~~~~~~~~~
C:\Users\niteris\dev\Nightstick\Nightstick-Arduino\Palettes.h:148:3: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  148 |   "firelike2",
      |   ^~~~~~~~~~~
C:\Users\niteris\dev\Nightstick\Nightstick-Arduino\Palettes.h:149:3: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  149 |   "ocean1",
      |   ^~~~~~~~
C:\Users\niteris\dev\Nightstick\Nightstick-Arduino\Palettes.h:150:3: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  150 |   "ocean2",
      |   ^~~~~~~~
In file included from C:\Users\niteris\dev\Nightstick\Nightstick-Arduino\Nightstick-Arduino.ino:180:
C:\Users\niteris\dev\Nightstick\Nightstick-Arduino\Nightstick.h:330:1: error: 'LSM6DS3' does not name a type; did you mean 'IMU_LSM6DS3'?
  330 | LSM6DS3 myIMU(I2C_MODE, 0x6A);    //I2C device address 0x6A
      | ^~~~~~~
      | IMU_LSM6DS3
[... 228 lines cut ...]
In file included from C:\Users\niteris\dev\Nightstick\Nightstick-Arduino\Nightstick-Arduino.ino:180:
C:\Users\niteris\dev\Nightstick\Nightstick-Arduino\Nightstick.h:76:19: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   76 | #define MAIN_PATH "/Nightstick"  // Main Folder on SD where BMPs Folder is in
      |                   ^
C:\Users\niteris\dev\Nightstick\Nightstick-Arduino\Nightstick.h:82:26: note: in expansion of macro 'MAIN_PATH'
   82 | #define FULLPATH_STATIC (MAIN_PATH SUBFLD_BMP SUBFLD_STATIC)
      |                          ^~~~~~~~~
[... cut ...]
      |           HUE_GREEN
C:\Users\niteris\dev\Nightstick\Nightstick-Arduino\System.ino: In function 'void setBoardLed(char, bool)':
C:\Users\niteris\dev\Nightstick\Nightstick-Arduino\System.ino:26:28: error: 'LED_GREEN' was not declared in this scope; did you mean 'HUE_GREEN'?
   26 |   else if(ch=='G'){chPin = LED_GREEN;}
      |                            ^~~~~~~~~
      |                            HUE_GREEN

exit status 1

Compilation error: 'D1' was not declared in this scope; did you mean 'A1'?
```

Loaded-Dice, 2024-09-13:
Good morning zackees,
I will compile a simple pio project for you to have a look at, right when I' back from Work. 
Thanks for your efforts

Loaded-Dice, 2024-09-13:
Hi @zackees !
I uploaded the PIO project for you
As you might suspect, I'm not too familiar with PIO, so please excuse any mistakes

Here is the PIO project that recreates the Fasted compiling error

  https://github.com/Loaded-Dice/FastLED_compile_error

I might inform you that PIO itself lacks sufficient support for the specific Board I used 
(XIAO nRF52 BLE Sense) 
But I managed to find an open pull request that would fix this issue and used the mentioned platformio.ini settings.

[env]
platform = https://github.com/maxgerhardt/platform-nordicnrf52
framework = arduino
[env:xiaoblesense_adafruit_nrf52]
board = xiaoblesense_adafruit
 
The above setting would equal my Arduino IDE settings that had the issues in the first place

PR: PlatformIO - Add Xiao nRF52840 (Sense) board support

https://github.com/platformio/platform-nordicnrf52/pull/151

If there is anything else that might be helpful,
please let me know.

Best regards

zackees, 2024-09-13:
Thanks for putting this all in a repo and allowing me to reproduce the issue exactly.

Yes, this is an issue with FastLED and non of the pins work. We have successfully reproduced this build issue on our side and have generated the following build badge to monitor the incompatibility:

[image]

zackees, 2024-09-13:
Okay for some reason there are some defines missing for this platform.

The pin definitions are in this file:
[src\platforms\arm\nrf52\clockless_arm_nrf52.h](src\platforms\arm\nrf52\clockless_arm_nrf52.h)

However when I do an

```C++
#ifndef NRF52_SERIES
#error
#endif
```

The expected #error is not being hit.

zackees, 2024-09-13:
Okay I think I found the cause:

[src\platforms\arm\nrf52\fastpin_arm_nrf52_variants.h](https://github.com/FastLED/FastLED/blob/master/src/platforms/arm/nrf52/fastpin_arm_nrf52_variants.h)

zackees, 2024-09-13:
Yes, confirmed.

I inserted `#error` on every board definition in the above file and it wasn't hit. No wonder you are hitting this issue.

zackees, 2024-09-13:
I see that you are using the nRF52840. I see that we do indeed have support for this compile type under the `NRF52840_FEATHER` definition.

I went ahead and put this in your platformio.ini and it compiled fine:

```
build_flags=-DARDUINO_NRF52840_FEATHER
```

zackees, 2024-09-13:
Okay, the proper fix is now submitted and your board should compile now with no further changes on your part IF you point your FastLED version to github master branch. This is easy to do with PlatformIO, I'm not sure how Arduino IDE would do this though.

The public fix for this board will go out with the next release on monday.

closed 2024-09-13 by zackees

Loaded-Dice, 2024-09-13:
Absolutely amazing ! I really appreciate your work and additionally this may serve as motivation to port my project to PIO :)
Thx for making my day.

zackees, 2024-09-13:
Thanks @Loaded-Dice. Your project looks interesting. Good luck with it!

bu5hm4nn, 2024-11-05:
@zackees Can you support the mbed framework version as well? It's the same repository, just a different board name. Here is a `platformio.ini` for the mbed enabled stack:

```
[env]
platform = https://github.com/maxgerhardt/platform-nordicnrf52
framework = arduino
[env:xiaoblesense_nrf52]
board = xiaoblesense
```

zackees, 2024-11-06:
Okay the support process has started.

How close is this to the adafruit xiaoblesense_nrf52840? I see that we have that board building.

[image]

zackees, 2024-11-06:
I've gotten about as far as I could go. However, it looks like both arduino.h and this nrf board is trying to pull in their own version of mbed and therefore there are #define redefines and other warnings. Ultimatly is doesn't compile and it looks like I need to use a forked version of the arduino framework. However, all the ones that I tried result is an error message saying this board isn't supported by the framework.

I'm not sure where to go from here. You can always just clone our repo and run `./compile xiaoblesense` which will set everything up and give you the exact error messages that's I'm seeing. 

We have our own compiler front end for platformio. Our board definition code can be found here:

https://github.com/FastLED/FastLED/blob/master/ci/ci/boards.py

Compiler log

```
TARGET_NRF5x/nrf5x_lf_clk_helper.h:43:6: warning: #warning No configuration for LF clock source. Xtal source will be used as a default configuration. [-Wcpp]
     #warning No configuration for LF clock source. Xtal source will be used as a default configuration.    
      ^~~~~~~
In file included from C:\Users\niteris\.platformio\packages\framework-arduino-mbed-seeed\cores\arduino/mbed/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52840/config/sdk_config.h:11967,
                 from C:\Users\niteris\.platformio\packages\framework-arduino-mbed-seeed\cores\arduino/mbed/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_SDK_15_0/integration/nrfx/nrfx_config.h:45,
                 from C:\Users\niteris\.platformio\packages\framework-arduino-mbed-seeed\cores\arduino/mbed/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_SDK_15_0/modules/nrfx/nrfx.h:44,
                 from C:\Users\niteris\.platformio\packages\framework-arduino-mbed-seeed\cores\arduino/mbed/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_SDK_15_0/modules/nrfx/hal/nrf_spim.h:44,
                 from src\platforms/arm/nrf52/led_sysdefs_arm_nrf52.h:46,
                 from src\led_sysdefs.h:14,
                 from src\FastLED.h:59,
                 from src\FastLED.cpp:2:
C:\Users\niteris\.platformio\packages\framework-arduino-mbed-seeed\cores\arduino/mbed/targets/TARGET_NORDIC/TARGET_NRF5x/nrf5x_lf_clk_helper.h:43:6: warning: #warning No configuration for LF clock source. Xtal source will be used as a default configuration. [-Wcpp]
     #warning No configuration for LF clock source. Xtal source will be used as a default configuration.    
      ^~~~~~~
In file included from C:\Users\niteris\.platformio\packages\framework-arduino-mbed-seeed\cores\arduino/mbed/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52840/config/sdk_config.h:11967,
                 from C:\Users\niteris\.platformio\packages\framework-arduino-mbed-seeed\cores\arduino/mbed/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_SDK_15_0/integration/nrfx/nrfx_config.h:45,
                 from C:\Users\niteris\.platformio\packages\framework-arduino-mbed-seeed\cores\arduino/mbed/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_SDK_15_0/modules/nrfx/nrfx.h:44,
                 from C:\Users\niteris\.platformio\packages\framework-arduino-mbed-seeed\cores\arduino/mbed/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_SDK_15_0/modules/nrfx/hal/nrf_spim.h:44,
                 from src\platforms/arm/nrf52/led_sysdefs_arm_nrf52.h:46,
                 from src\led_sysdefs.h:14,
                 from src\FastLED.h:59,
                 from src\five_bit_hd_gamma.cpp:5:
C:\Users\niteris\.platformio\packages\framework-arduino-mbed-seeed\cores\arduino/mbed/targets/TARGET_NORDIC/TARGET_NRF5x/nrf5x_lf_clk_helper.h:43:6: warning: #warning No configuration for LF clock source. Xtal source will be used as a default configuration. [-Wcpp]
     #warning No configuration for LF clock source. Xtal source will be used as a default configuration.    
      ^~~~~~~
Compiling .pio\build\xiaoblesense\lib1cd\src\fx\video\stream_buffered.cpp.o
Compiling .pio\build\xiaoblesense\lib1cd\src\hsv2rgb.cpp.o
In file included from C:\Users\niteris\.platformio\packages\framework-arduino-mbed-seeed\cores\arduino/mbed/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52840/config/sdk_config.h:11967,
                 from C:\Users\niteris\.platformio\packages\framework-arduino-mbed-seeed\cores\arduino/mbed/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_SDK_15_0/integration/nrfx/nrfx_config.h:45,
                 from C:\Users\niteris\.platformio\packages\framework-arduino-mbed-seeed\cores\arduino/mbed/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_SDK_15_0/modules/nrfx/nrfx.h:44,
                 from C:\Users\niteris\.platformio\packages\framework-arduino-mbed-seeed\cores\arduino/mbed/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_SDK_15_0/modules/nrfx/hal/nrf_spim.h:44,
                 from src\platforms/arm/nrf52/led_sysdefs_arm_nrf52.h:46,
                 from src\led_sysdefs.h:14,
                 from src\FastLED.h:59,
                 from src\colorutils.cpp:11:
C:\Users\niteris\.platformio\packages\framework-arduino-mbed-seeed\cores\arduino/mbed/targets/TARGET_NORDIC/TARGET_NRF5x/nrf5x_lf_clk_helper.h:43:6: warning: #warning No configuration for LF clock source. Xtal source will be used as a default configuration. [-Wcpp]
     #warning No configuration for LF clock source. Xtal source will be used as a default configuration.    
      ^~~~~~~
Compiling .pio\build\xiaoblesense\lib1cd\src\lib8tion.cpp.o
[... cut ...]
 #define MBED_CONF_NORDIC_NRF_LF_CLOCK_SRC                                       NRF_LF_SRC_XTAL
                                                                      // set by target:MCU_NRF52840

In file included from C:\Users\niteris\.platformio\packages\framework-arduino-mbed-seeed\cores\arduino/mbed/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52840/config/sdk_config.h:11967,
                 from C:\Users\niteris\.platformio\packages\framewo
Compilation finished in 00m:24s.
```

bu5hm4nn, 2024-11-08:
Thanks. Yeah their library fork is half hearted at best. I will see if I can fix their library up a bit.

zackees, 2024-11-13:
Do let me know any updates