FastLED/FastLED #1447: Sudden validpin() compilation error without changes.

TheBloodyScreen, 2022-11-20:
Yesterday I was playing around with my ESP8266-12F and a WS2812B Led matrix.
I went to bed having everything working exactly as I wanted it. This morning, just for OCD reasons, I tried to rebuild and upload the exact same sketch to the controller and suddenly I get the error below.
I have no idea how that can even be since not even the USB port nor the cable I use has changed since yesterday. 

`In file included from C:\Users\Dooms\Documents\Arduino\BloodyFrame\A_BloodyFrame.ino\A_BloodyFrame.ino.ino:1:
c:\Users\Dooms\Documents\Arduino\libraries\FastLED\src/FastLED.h:14:21: note: '#pragma message: FastLED version 3.004.000'
   14 | #    pragma message "FastLED version 3.004.000"
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from c:\Users\Dooms\Documents\Arduino\libraries\FastLED\src/FastLED.h:65,
                 from C:\Users\Dooms\Documents\Arduino\BloodyFrame\A_BloodyFrame.ino\A_BloodyFrame.ino.ino:1:
c:\Users\Dooms\Documents\Arduino\libraries\FastLED\src/fastspi.h:135:23: note: '#pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output'
  135 | #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from c:\Users\Dooms\Documents\Arduino\libraries\FastLED\src/FastLED.h:48,
                 from C:\Users\Dooms\Documents\Arduino\BloodyFrame\A_BloodyFrame.ino\A_BloodyFrame.ino.ino:1:
c:\Users\Dooms\Documents\Arduino\libraries\FastLED\src/fastpin.h: In instantiation of 'class FastPin<7>':
c:\Users\Dooms\Documents\Arduino\libraries\FastLED\src/platforms/esp/8266/clockless_esp8266.h:21:49:   required from 'class ClocklessController<7, 20, 50, 30, GRB, 0, false, 50>'
c:\Users\Dooms\Documents\Arduino\libraries\FastLED\src/chipsets.h:578:7:   required from 'class WS2812Controller800Khz<7, GRB>'
c:\Users\Dooms\Documents\Arduino\libraries\FastLED\src/FastLED.h:103:52:   required from 'class WS2812<7, GRB>'
c:\Users\Dooms\Documents\Arduino\libraries\FastLED\src/FastLED.h:302:39:   required from 'static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with CHIPSET = WS2812; unsigned char DATA_PIN = 7; EOrder RGB_ORDER = GRB]'
C:\Users\Dooms\Documents\Arduino\BloodyFrame\A_BloodyFrame.ino\A_BloodyFrame.ino.ino:48:55:   required from here
c:\Users\Dooms\Documents\Arduino\libraries\FastLED\src/fastpin.h:210:24: error: static assertion failed: Invalid pin specified
  210 |  static_assert(validpin(), "Invalid pin specified");
      |                ~~~~~~~~^~

exit status 1

Compilation error: exit status 1`

Loaded-Dice, 2022-11-20:
Have you tried declaring your pin as D7 instead of 7 ?
#define DATA_PIN D7
It may be possible that FastLed tries to find  "GPIO 7"  when you only use 7 as pin no.

TheBloodyScreen, 2022-11-20:
Oh my god! That really does seem to have fixed the issue.
Thank you! 

Though I still don't understand why it worked flawlessly one day without that D and suddenly I get an error without anything having changed.

Loaded-Dice, 2022-11-20:
Have you chosen a slightly different ESP8266 from the board menu last time? This may be the reason. Some ESP versions are pretty much identical, except for pin no. , definitions and related constants.

TheBloodyScreen, 2022-11-20:
No, that's just it. As I said above, nothing has changed. I didn't even shut down my PC. That issue just popped up between one compile and the next.

closed 2022-11-20 by TheBloodyScreen