FastLED/FastLED #1448: CRGB::Red and CRGB::Green are inverted

diegochencoconut, 2022-11-24:
To those whom may concern:
When we're using this repository by ATTiny, we've noticed that CRGB::Red and CRGB::Green are inverted.

When we use CRGB::Red, the light shows Green.
Otherwise, when we use CRGB::Green, the light shows Red.

We're using WS2812 & Attiny.

Thank you.

zackees, 2022-11-24:
That's your LED strip. Change the RGB order in your FastLED template instantiation and it will work fine.

Loaded-Dice, 2022-11-24:
Zackees is right.
To make it a little bit easier to get what he means:

Find  this line in your code

FastLED.addLeds<WS2812, DATA_PIN, RGB>(leds, NUM_LEDS); 

and change it into this line:

FastLED.addLeds<WS2812, DATA_PIN, GRB >(leds, NUM_LEDS); 

If the color order has swapped but still not correct
Adjust the three letters you changed from RGB to GRB to another order until your color order is correct.

samguyer, 2022-11-25:
Thanks @zackees and @Loaded-Dice for helping out! @diegochencoconut, did this fix your problem? Can we close this issue?

diegochencoconut, 2022-11-26:
@samguyer  @zackees @Loaded-Dice 

Thanks for your reply!
It fixed my problem.

zackees, 2022-11-27:
Glad it worked! Let's close this issue then.

closed 2022-11-27 by diegochencoconut