FastLED/FastLED #1452: Does not compile for esp32 with latest arduino appimage and arduino updated libs etc

emilfihlman, 2022-11-27:
```cpp
In file included from /home/emil/Arduino/libraries/FastLED/src/FastLED.h:67,
                 from /home/emil/Arduino/valokasvot/valokasvot.ino:2:
/home/emil/Arduino/libraries/FastLED/src/fastspi.h:145:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

```cpp
/home/emil/Arduino/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp: In static member function 'static void ESP32RMTController::init(gpio_num_t)':
/home/emil/Arduino/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:111:15: error: variable 'espErr' set but not used [-Werror=unused-but-set-variable]
     esp_err_t espErr = ESP_OK;
               ^~~~~~
```

```cpp
/home/emil/Arduino/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp: In member function 'void ESP32RMTController::startOnChannel(int)':
/home/emil/Arduino/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:239:15: error: variable 'espErr' set but not used [-Werror=unused-but-set-variable]
     esp_err_t espErr = ESP_OK;
               ^~~~~~
cc1plus: some warnings being treated as errors
```

Using library FastLED at version 3.5.0 in folder: /home/emil/Arduino/libraries/FastLED 
exit status 1

Compilation error: exit status 1

jgromes, 2023-02-04:
This is still a problem using latest ESP32 and FastLED (3.5.0), the error I get is exactly the same when installing the library from Arduino library manager.

It was fixed in a PR, https://github.com/FastLED/FastLED/pull/1399, but there were no releases since, so maybe it's time to do so? cc @kriegsman

zackees, 2024-09-19:
fixed

closed 2024-09-19 by zackees