FastLED/FastLED #1455: Potential issue with dithering support

dotorg, 2022-12-08:
I'm not sure if this behavior is as-intended or not. The current dithering implementation has an issue if/when a strip (or, in a more obvious case, a matrix) has an even number of pixels.  Any time the strip length is a multiple of the number of dither steps, the incrementing of the dithering step per-pixel is causing the same pixels to always end up with the same dithering level.   Depending on the orientation of a matrix, the end result is the dithering is resulting in stripes, where alternate rows or columns are always getting the brighter or darker value.  The problem isn't super obvious at brighter levels, but its plain as day when the LED brightness starts to get low (like below 8-12). 

Everywhere in the code where the dithering is being applied, stepDithering() is being called in the pixel output while loop, which results in each update having the same pixels always getting the same dither adjustment level in the case where everything comes out even. 

I've patched it locally, which corrects the problem, by moving the stepDithering() call out of the while loop, so the step advances per frame, not per-pixel. 

I'd be happy to adjust it everywhere and do a pull request, but I wanted to verify the repeating behavior isn't deliberate for some reason I'm missing.

dotorg, 2022-12-11:
[image]
Its not a great pic, but you can see it here. This is on a 32x8 panel, so 256 LEDs. Now, I didn't dig super deep into the code, but it sure looked like it was a 6-multiple, which doesn't divide out. However, I have a second strip of 14, and 270 *is* evenly divisible by 6. So, its possible it doesn't show up with most panels because they're in common powers-of-two. (8x8, 8x32, 32x32 and 64x64 seeming the most common.) If my reading was right and its a 6-step cycle, that may be why no one has noticed it. (I didn't think to now to see what happened if I made the 14-pixel strip 15 as a test.)

So, it might be easy to reproduce by telling FastLED that a panel is slightly larger. Like a 32x32 is 1026 pixels, not 1024.

kriegsman, 2022-12-11:
Thank you for the visual example, and your notes.   I spent a little time reviewing the code; I know that there's _supposed_ to be a mechanism that changes the starting dither state at the start of each frame _as well as_ once per pixel, to prevent this exact thing.  I'm going to see where that's falling through the cracks and if I can make it work as intended.  The per-pixel dither step is definitely desirable -- but it's not supposed to get stuck in lockstep cycle with the number of pixels, the way it is.

dotorg, 2022-12-12:
Well, for completeness' sake, I'm running with an ESP32-S2, RMI controller because I2S doesn't work on the S2. The hanging problem with the RMI controller means I'm running with the number set to 1, with two defined strips -- a 256 element WS2812B and a 14 element WS2812B. So they're running sequentially, not in parallel.  So, its possible the problem isn't universal and is specific to that controller, as well. Once I identified the "shift-every-pixel" logic and guessed it was causing, essentially, a resonance in the output, I didn't dig much deeper. It was faster to fork and move that line of code out of the while loop.

closed 2026-08-31 by zackees