FastLED/FastLED #1456: Complie error platforms/arm/stm32/fastpin_arm_stm32.h:58:88: error: incomplete type '__gen_struct_GPIOC' used in nested name specifier

Thermaltake, 2022-12-14:
Arduino IDE
STM32F103C8

In file included from C:\Users\slavi\Documents\Arduino\libraries\FastLED-master\src/platforms/arm/stm32/fastled_arm_stm32.h:5:0,
                 from C:\Users\slavi\Documents\Arduino\libraries\FastLED-master\src/platforms.h:31,
                 from C:\Users\slavi\Documents\Arduino\libraries\FastLED-master\src/FastLED.h:62,
                 from C:\Users\slavi\Documents\Arduino\libraries\FastLED-master\examples\DemoReel100\DemoReel100.ino:5:
C:\Users\slavi\Documents\Arduino\libraries\FastLED-master\src/platforms/arm/stm32/fastpin_arm_stm32.h: In instantiation of 'static volatile uint32_t* _ARMPIN<PIN, _BIT, _MASK, _GPIO>::port() [with unsigned char PIN = 3u; unsigned char _BIT = 14u; long unsigned int _MASK = 16384ul; _GPIO = __gen_struct_GPIOC; _ARMPIN<PIN, _BIT, _MASK, _GPIO>::port_ptr_t = volatile long unsigned int*; uint32_t = long unsigned int]':
C:\Users\slavi\Documents\Arduino\libraries\FastLED-master\src/platforms/arm/stm32/clockless_arm_stm32.h:30:41:   required from 'void ClocklessController<DATA_PIN, T1, T2, T3, RGB_ORDER, XTRA0, FLIP, WAIT_TIME>::init() [with int DATA_PIN = 3; int T1 = 24; int T2 = 24; int T3 = 47; EOrder RGB_ORDER = (EOrder)66u; int XTRA0 = 0; bool FLIP = false; int WAIT_TIME = 50]'
C:\Users\slavi\Documents\Arduino\libraries\FastLED-master\examples\DemoReel100\DemoReel100.ino:126:1:   required from here
C:\Users\slavi\Documents\Arduino\libraries\FastLED-master\src/platforms/arm/stm32/fastpin_arm_stm32.h:58:88: error: incomplete type '__gen_struct_GPIOC' used in nested name specifier
     inline static port_ptr_t port() __attribute__ ((always_inline)) { return &_GPIO::r()->ODR; }
                                                                                        ^
exit status 1
Ошибка компиляции для платы Generic STM32F103C series.

zackees, 2022-12-14:
what's your ide? Arduino or PlatformIO? Please provide more detail

Thermaltake, 2022-12-14:
Arduino IDE 1.8.16

zackees, 2022-12-14:
From looking at the stack trace here my guess is that one of the types in the inline function `port_ptr_t port();` has been forward declared but not defined.

The likely culprits are that port_ptr_t type, or the type of _GPIO. My advice would be to find the headers that have these class definitions and include them BEFORE you `#include <FastLED.h>`, that way you'll force the compiler to see the full type definition of the problematic class/struct and it should fix the solution. If this fixes it, please let us know what header needed to be included.

alex-eri, 2023-10-30:
Same issue on platformio

```
[env:bluepill_f103c8]
platform = ststm32
board = bluepill_f103c8
framework = arduino
board_build.core = maple
board_build.ldscript = jtag_c8.ld
```

closed 2025-01-13 by zackees