FastLED/FastLED #1457: ISO C++17 does not allow 'register' storage class specifier anymore

nielsnl68, 2022-12-29:
Hi, with the latest versions of espressif8266 using the most recent version of PlatformIO i get the warning that ISO C++17 does not allow 'register' storage class specifier anymore

here is my compiler header message:
```
processing lum-iot25 (board: esp01_1m; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------
Library Manager: Installing ottowinter/ESPAsyncTCP-esphome @ 1.2.3
INFO Installing ottowinter/ESPAsyncTCP-esphome @ 1.2.3
Unpacking  [####################################]  100%
Library Manager: ESPAsyncTCP-esphome@1.2.3 has been installed!
INFO ESPAsyncTCP-esphome@1.2.3 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
INFO Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@2.1.0 has been installed!
INFO ESPAsyncWebServer-esphome@2.1.0 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing fastled/FastLED @ 3.5.0
INFO Installing fastled/FastLED @ 3.5.0
Unpacking  [####################################]  100%
Library Manager: FastLED@3.5.0 has been installed!
INFO FastLED@3.5.0 has been installed!
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- ESPAsyncTCP-esphome @ 1.2.3
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- ESPAsyncTCP-esphome @ 1.2.3
|   |-- Hash @ 1.0
|   |-- ESP8266WiFi @ 1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- FastLED @ 3.5.0 
```

and this part of the compiler log:
```
/data/lum-iot25/.piolibdeps/lum-iot25/FastLED/src/fastpin.h:61:37: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
   61 |  inline void hi(register port_ptr_t port) __attribute__ ((always_inline)) { *port |= mPinMask; }
      |                                     ^~~~
/data/lum-iot25/.piolibdeps/lum-iot25/FastLED/src/fastpin.h:62:37: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
   62 |  inline void lo(register port_ptr_t port) __attribute__ ((always_inline)) { *port &= ~mPinMask; }
      |                                     ^~~~
/data/lum-iot25/.piolibdeps/lum-iot25/FastLED/src/fastpin.h:63:34: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
   63 |  inline void set(register port_t val) __attribute__ ((always_inline)) { *mPort = val; }
      |                                  ^~~
/data/lum-iot25/.piolibdeps/lum-iot25/FastLED/src/fastpin.h:65:42: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
   65 |  inline void fastset(register port_ptr_t port, register port_t val) __attribute__ ((always_inline)) { *port  = val; }
      |                                          ^~~~
/data/lum-iot25/.piolibdeps/lum-iot25/FastLED/src/fastpin.h:65:64: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
   65 |  inline void fastset(register port_ptr_t port, register port_t val) __attribute__ ((always_inline)) { *port  = val; }
      |                                                                ^~~
 <[etc.]>
 ```
 I hope someone can suggest how to replace `register`.

zackees, 2022-12-29:
At the very least the register storage qualifier needs to be replaced with a conditional macro expansion like REGISTER that defined to nothing when C++17 is detected

zackees, 2023-01-02:
My advice is to clone the repo and remove the register qualifier for your project. The register keyword is not needed.

zackees, 2023-01-02:
So diving deeper, it looks like there was already a renaming of register -> REGISTER, but it was incomplete. I'm issuing a fix because this is very straightforward.

nielsnl68, 2023-01-02:
I have found that out as well and working on a PR for it already.
After i resolved the issue above, there are a couple of warnings left in `clockless_esp8266.h` around line 56 and 57 :
```cpp
  while((__clock_cycles() - last_mark) < (T1+T2+T3));
  last_mark = __clock_cycles();
```
with errors:
```
warning: this 'while' clause does not guard... [-Wmisleading-indentation]
note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'while'
```

I understand what is going on here, but I am not good fully known Cpp programmer to know how i should fix this.

zackees, 2023-01-02:
This indentation is an error?!

If so, then do this:

```C++
  while((__clock_cycles() - last_mark) < (T1+T2+T3)) {
    ;
  }
  last_mark = __clock_cycles();
```

nielsnl68, 2023-01-02:
thanks i will try that.

closed 2023-01-15 by kriegsman

zackees, 2023-01-16:
> thanks i will try that. 

My PR fix for the register has been merged. I recommend trying to recompile now and see if this fixes your issue.

Guenni75, 2023-03-10:
Still the same Problem with FastLed 3.5.0.
Can't do anything with FastLed and an ESP8266.
Every error line has a problem with "register".

nielsnl68, 2023-03-10:
I created an PR for this but it seems there is no activation for some time now. Sadly.

m4rc77, 2023-04-07:
@kriegsman FASTLed release 3.5.0 was released on Jan 2, 2022. Today, April 7th, 2023 this is still the latest Release available. 
PR  #1460 was merged into master Branch on January 15th 2023.

When will be an official release (e.g. 3.5.1) available with this patch included?

m4rc77, 2023-04-07:
If you are using [PlatformIO](https://platformio.org/) you can add in your `platformio.ini` file the following lines, in order to "silence" the warnings from FastLED until a patch is available:

```
; fix in order to get rid of 'warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]' from FastLED Library <= 3.5.0
; fix in order to get rid of 'warning: this 'while' clause does not guard... [-Wmisleading-indentation]' from FastLED Library <= 3.5.0
build_flags =
    -Wno-register
    -Wno-misleading-indentation
```
**But be warned, this will also hide similar issues in your own code!**