FastLED/FastLED #1458: ESP8266 + FastLED 3.5.0 + WS2812 = 13us gap after first LED

jalmeroth, 2022-12-29:
Hi :wave:,

I am trying to drive some WS2812 LEDs and everytime I reboot the ESP8266 and make the first call to `FastLED.showColor()` (or show) I see a gap after the first LED's data:
[image]

This breaks the timing and leads to various effects on the following LEDs.

Subsequent color changes are fine then:
[image]

Is there anything I can do about this?

samguyer, 2023-01-22:
But it doesn't happen in subsequent calls to show()? I.e., is it a one-time thing after a reboot?

jalmeroth, 2023-01-22:
> But it doesn't happen in subsequent calls to show()? I.e., is it a one-time thing after a reboot?

After a reboot or re-initialization, yes

zackees, 2026-04-19:
The driver has been reworked since this was filed. The specific mechanism that caused the post-first-LED gap on cold start is now mitigated in two places:

1. **`showRGBInternal` is now IRAM-resident** ([`clockless_esp8266.h:108`](https://github.com/FastLED/FastLED/blob/master/src/platforms/esp/8266/clockless_esp8266.h#L108)): the `FL_IRAM` attribute moves the hot loop out of flash, so the first call doesn't pay for an iCache miss mid-frame.
2. **Per-bit deadline check** ([lines 100–102](https://github.com/FastLED/FastLED/blob/master/src/platforms/esp/8266/clockless_esp8266.h#L100-L102)): if an NMI (or cache stall) stretches a bit past `T1+T2+T3`, the frame aborts cleanly and `showPixels` retries, rather than emitting the partial gap you scoped.

If you can re-test on master with the same setup (D1 mini + WS2812) and the same scope config, I'd love to know whether that first-frame gap is gone. If it's still there, please attach the new scope capture and I'll dig further — but I suspect this is fixed.