FastLED/FastLED #1459: [Question] Do I have to worry about using "delete"?

Macintosh-Fan, 2022-12-30:
_I'm posting the question here since there is no discussion tab for this repository._

My Arduino only has 2 KB of memory and 32 KB of flash. When I am changing the colour of the lights, should I `delete` my other instances of `CRGB` to make sure memory usage is minimal?

Using `delete` for the objects in my `CRGB led_lights[LED_COUNT]` array.

marmilicious, 2022-12-30:
Feel free to join our subreddit for discussion too. 

https://www.reddit.com/r/FastLED

zackees, 2022-12-30:
No.

And on a larger note please don't use new/delete and malloc/free unless you have to. Prefer static allocation and an init function. If you have to allocate do it once and try not to free memory.

This is how C programs are written.

Why?

Malloc/new and free/delete if called with large large allocation will fragment your small memory space and possibly cause the unit to crash when it fails to allocate a new memory block and returns null instead.

Macintosh-Fan, 2022-12-30:
I see. Thanks @zackees!

closed 2022-12-30 by Macintosh-Fan