FastLED/FastLED #1462: Serial input error when using fast LED

Camo2000, 2023-01-04:
Arduino IDE V2 Serial input while using fast led fails if you use FastLED.delay(); but not under FastLED.show();

See code below: may not be formatted 100% correct as it is only for my use, do not need leds on A0 pin to test

#include <FastLED.h>
#define NUM_LEDS 144
#define LED_TYPE WS2811
#define COLOR_ORDER GRB  //RGB//RBG// GRB = Circles
#define DATA_PIN A0
CRGBArray<NUM_LEDS> leds;
String readString, data1, data2, data3;

void setup() {
  Serial.begin(19200);
  FastLED.addLeds<LED_TYPE, DATA_PIN, COLOR_ORDER>(leds, NUM_LEDS);
}

void loop() {
leds=CHSV(random(255),255,128);

FastLED.delay(1000); //makes serial input corrupt somehow even if you input the same text it changes the pointers

/* this code works OK and does the same.
FastLED.show();
delay(1000);
*/

  getSerial();//H023S255V128   <<try inputting this string
}

void getSerial() {
  //H023S255V128   <<try inputting this string
  
  while (Serial.available() > 0) {

    // if (Serial.available() >0) {
    // char c = Serial.read();  //gets one byte from serial buffer
    // readString += c;         //makes the string readString
    // }

    readString = Serial.readString();
    readString.trim();
    delay(200);
  }
  if (readString.length() > 0) {
    Serial.print("Original txt:");
    Serial.print(readString);  //see what was received

    // expect a string 12 characters long, then split in to 3 three data parts
    data1 = readString.substring(0, 4);   //get the first 4 characters
    data2 = readString.substring(4, 8);   //get the next 4 characters
    data3 = readString.substring(8, 12);  //get the next 4 characters
    Serial.print("  Data1:");
    Serial.print(data1);  //print to serial monitor to see results
    Serial.print("  Data2:");
    Serial.print(data2);
    Serial.print("  Data3:");
    Serial.println(data3);

    readString = "";
    data1 = "";
    data2 = "";
    data3 = "";
    delay(200);
  }
}

Camo2000, 2023-01-04:
an update 
so using FastLED.show() serial output is correct as follows
Original txt:H023S255V128  Data1:H023  Data2:S255  Data3:V128

under fastLED.delay(1000); it fails and does not collect the string returned correctly and is always different
Text entered is always H023S255V128 
Original txt:H02S25128  Data1:H02S  Data2:2512  Data3:
Original txt:H0225128  Data1:H022  Data2:  Data3:
it is different every time even with the same text entered
please note if there are not 8 or 12 characters it does not get them for data2 and data3

dmadison, 2023-01-05:
This is expected behavior, and caused by FastLED disabling interrupts when writing to WS2812B strips.

With `FastLED.show()` the LED data is only written once. With `FastLED.delay()` the LED data is written multiple times for as long as the delay period in order to achieve dithering. This blocks the serial interrupt from firing.

Camo2000, 2023-01-05:
Thanks for your reply.
This does not seem right to me. If the delay is there why keep writing if no other code is executed or calculated?
Where would this dithering coming from and why?

Sorry for the additional question just trying to understand what the delay is really for in the fast environment.
Am I misunderstanding FastLED.delay()  I thought it behaved as the show and delay together. Am I to understand it stops and starts at random as this means the while loop for collecting is stopped from FastLED and that the code is not linear.
thx

dmadison, 2023-01-05:
>Am I misunderstanding FastLED.delay() I thought it behaved as the show and delay together

Not quite. The purpose of the function is to wait (just like plain `delay()`), but let the LEDs keep refreshing so dithering does not stop. [You can see the current code for the function here](https://github.com/FastLED/FastLED/blob/4b6d3bb57e8419d792f35039996fd716c235313f/src/FastLED.cpp#L132). I'll copy it for convenience:

```cpp
unsigned long start = millis();
do {
	::delay(1);
	show();
	yield();
}
while((millis()-start) < ms);
```

If you don't need the dithering behavior, you can use the plain `FastLED.show()` with `delay()`. But beware that serial communication while using WS2812B LEDs can be unreliable due to the interrupt issues. Clocked chipsets like APA102 do not have these issues. [See this wiki page for more information](https://github.com/FastLED/FastLED/wiki/Interrupt-problems).

Camo2000, 2023-01-05:
Thanks
I was using serial input to change values for HSV, Global Brightness, D delays, P print off. 
inputting a string then chopping it up H255S255V255  Or B255 brightness D255 etc:
The "HSVDBP<>" is a Char[] and I use it to look up the menu position by searching the char[] position number then go to case select number. So "< "= (7) previous effect -1 so i know where in the menu to access from the number position of the char, only from serial while developing, saves compiling every time. and building a development board.
I changed the code to use show instead when I spotted the difference.

Glad it is not a bug.
Many thanks for the reply and HNY.

samguyer, 2023-01-22:
@dmadison Thanks for your response! @Camo2000 can we close this issue?

samguyer, 2023-11-27:
Thank you all for helpful discussion!

closed 2023-11-27 by samguyer