FastLED/FastLED #1469: CLEDController::size isn't working properly

kristof01124, 2023-01-30:
Hi!

I am currently using an esp32 to control a led strip. I am trying to use the CLEDController class returned by the FastLED.addLeds function call to modify the led values, but when i call the size() function on the reference it returns the actual size multiplied by 256.

Example:
        
        CLEDController& controller = FastLED.addLeds<LED_TYPE, LED_PIN>(new CRGB[10], 10);
        Serial.println(controller .size()); // This prints out 2560

I am using version 3.5.0 and Arduino IDE 2.0.3.
My microcontroller is a NodeMCU ESP-32S.

zackees, 2023-02-01:
You are doing it wrong.

You pass the led array to FastLED. Then you modify the leds in your own code, you don't have to pass them in again, but when you want to update the led strip you call show().

dmadison, 2023-02-01:
I can't replicate this behavior.

Using an ESP32 and 3.5.0, testing with both Arduino 1.8.19 and Arduino 2.0.1:

```cpp
#include <FastLED.h>

void setup() {
    Serial.begin(115200);
    CLEDController& controller = FastLED.addLeds<WS2812B, 2>(new CRGB[10], 10);
    Serial.println(controller.size());
}

void loop() {}
```

Result:
```
10
```

closed 2024-09-20 by zackees