FastLED/FastLED #1470: Increasing the delay when reaching "n" the LED

Evgen0922, 2023-02-01:
Hello! I have this problem:
### WS2812 (90 leds), ESP8266.
I have a code that the LED shows "I", then makes a delay(current_delay) or Fast LAD.delay(current_delay) (tried both) and "I+1"
When I run this code (delay = 10), the first ~26 LEDs are lit with a delay of 10-13 milliseconds, and after 26 LEDs the delay increases to 17-20.
I tried to run the LED strip from the beginning and from the end.
I attach the code and the output of the arduino IDE.

My code:

```
#define FASTLED_ESP8266_D1_PIN_ORDER
#include <FastLED.h>
#define LED_PIN      D6
#define NUM_LEDS     90
#define BRIGHTNESS   64
#define LED_TYPE WS2812B
#define COLOR_ORDER GRB
CRGB leds[NUM_LEDS];
#include <EncButton2.h>
#include <StopWatch.h>
StopWatch sw_millis;
int one = 1;                 
int ThisPixel = 0; 
...........
void setup()
{
  FastLED.addLeds<LED_TYPE, LED_PIN, COLOR_ORDER>(leds, NUM_LEDS).setCorrection( TypicalLEDStrip );
  FastLED.setBrightness( BRIGHTNESS );
}
void loop(){
......
 SLL_usual();
 enc.tick();
...
}
void SLL_usual() { 
if (enc.click())
{
 sw_millis.start();
 btnclick = 1;
}
while (btnclick == 1)
{
  ThisPixel += one;
  leds[ThisPixel] = CRGB::Red;
  FastLED.delay(current_delay_millis);

  ThisPixel_last = ThisPixel;

  Serial.print(countn);     // number led
  Serial.print("--");
  Serial.print(sw_millis.elapsed());    //delay
  countn = countn + 1;
}
```

And arduino IDE output:
```
---16--167 led
---17--177 led
---18--187 led
---19--197 led
---20--207 led
---21--217 led
---22--234 led
---23--251 led
---24--267 led
---25--284 led
---26--301 led
---27--317 led
```

Please help me!

dmadison, 2023-02-01:
My guess is that you're filling the serial buffer and the microcontroller is waiting for it to clear before processing the `print()` statements. Have you tried upping the serial baud rate?

Evgen0922, 2023-02-02:
Oh! Thanks! I've been trying to solve this problem for a long time, and **it helped**!
I think it can be added to the wiki, for beginners.

closed 2023-02-02 by Evgen0922